espressif/ESP8266_NONOS_SDK #127: How to use native AES Encrypt implemented on hardware?

augustocadini, 2018-05-21:
I did read on datasheet that ESP8266 have a hardware implementation of AES Encrypt. But, I only found examples using Arduino and C third librarys to make this. So, how I use the native hardware implementation of AES Encrypt? Where I found example, documentation and header files?

eriksl, 2018-09-07:
What are the function names (see PROVIDEs in loader script). It's probably the same way as with MD5 and SHA1, functions having almost the same name of those from openssl and the interface is the same then.

But indeed, it's a shame they're not just documented (or even mentioned as function prototype). And if it's a software or hardware implementation.

I have it working for SHA1 and MD5. It seems to work quite fast, so it might a hardware implementation.

xcguang, 2018-09-10:
Hi,
There is no hardware AES Encrypt in ESP8266. And ESP32 has it.

closed 2018-09-21 by xcguang