espressif/ESP8266_NONOS_SDK #133: Please add PAP for authentication phase 2 (along with MSCHAPV2)

d-a-v, 2018-06-11:
Hello,

I have the chance to have a full access (with logs) to a local server of the widely used (and now reaching asia) 'EDUcation ROAMing' [eduroam](https://www.eduroam.org/what-is-eduroam/) network to which a bunch of esp8266 users would like  to connect to. It's generally a wpa2-enterprise network.

It does not work here and I think for a simple reason: the ESP is using `MSCHAPV2`,
But this network **can require the `PAP` phase 2 authentication method**.
That explains why some user can connect to that network, and why some others not.
The required EAP method is `TTLS` which the ESP honours from user application and  according to the esp-nonos-sdk logs below:
```
11:41:45.038 -> SDK:2.2.1(cfd48f3)
[...]
11:42:13.606 -> reconnect
11:42:13.606 -> state: 2 -> 0 (0)
11:42:13.706 -> scandone
11:42:13.706 -> state: 0 -> 2 (b0)
11:42:13.706 -> state: 2 -> 3 (0)
11:42:13.706 -> state: 3 -> 5 (10)
11:42:13.706 -> add 0
11:42:13.706 -> aid 5
11:42:13.739 -> cnt 
11:42:13.739 -> EAP-TTLS: Start (server ver=0, own ver=0)
11:42:17.482 -> (user log:) Status: 1 - Arduino status: 6 - Local IP:0.0.0.0
11:42:17.515 -> EAP-TTLS: TLS done, proceed to Phase 2
11:42:17.548 -> EAP-TTLS: received 0 bytes encrypted data for Phase 2
11:42:17.548 -> EAP-TTLS: empty data in beginning of Phase 2 - use fake EAP-Request Identity
11:42:17.548 -> EAP-TTLS: Phase 2 MSCHAPV2 Request
11:42:17.548 -> (user log:) Status: 1 - Arduino status: 6 - Local IP:0.0.0.0
11:42:18.608 -> state: 5 -> 2 (2a0)
11:42:18.608 -> rm 0
11:42:18.608 -> wifi evt: 1
11:42:18.608 -> STA disconnect: 2
11:42:19.342 -> state: 2 -> 3 (0)
11:42:19.541 -> (user log:) Status: 1 - Arduino status: 6 - Local IP:0.0.0.0
11:42:20.369 -> state: 3 -> 0 (4)
11:42:20.370 -> reconnect
```

FWIW,
* Here are the Phase 2 authentication available methods proposed by my android phone:
   None PAP MSCHAP MSCHAPV2 GTC.
The ESP uses MSCHAPV2 and to my knowledge the nonos-sdk API does not propose to select something here.
* After forgetting the network and setting it up again, my android phone does not automatically propose the right Phase 2 authentication method, it must be configured by the user. I don't know if they can be deduced from initial handhake. This eduroam wifi network was historically not supported by some android phone at the beginning, especially because of the unavailable `PAP` phase 2 method.

Here are the laconic logs of the radius server of my local network. Note that my user name is correctly transmitted from the esp, but not the password. The mac address shown is the esp's one:
```
Jun 11 11:42:17 servername radiusd[6804]:   [ldap] Attribute "User-Password" is required for authentication.
Jun 11 11:42:17 servername radiusd[6804]: Login incorrect: [my-working-login] (from client some-client port 0 via TLS tunnel)
Jun 11 11:42:17 servername radiusd[6804]: Login incorrect: [my-working-login] (from client come-client port 12293 cli 5C-CF-7F-C3-AD-51)
```
Relevant part of the source code:
```
[...]

#define SSID          "eduroam"
#define PASSWORD      ""
#define WPA2_USERNAME "my-working-login"
#define WPA2_IDENTITY WPA2_USERNAME
#define WPA2_PASSWORD "my-working-password"

[...]

// the following is a slightly modified copy-paste of the wpa2-enterprise nonos-sdk's example

void setup() {
  wifi_station_disconnect();

  Serial.begin(115200);
  Serial.setDebugOutput(true);

  Serial.print("Trying to connect to ");
  Serial.println(SSID);

  {
    char ssid[32] = SSID;
    char password[64] = PASSWORD;
    struct station_config sta_conf;// = { 0 };

    os_memset(&sta_conf, 0, sizeof(sta_conf));
    os_memcpy(sta_conf.ssid, ssid, 32);
    os_memcpy(sta_conf.password, password, 64);
    wifi_station_set_config(&sta_conf);
  }

  {
    typedef enum {
      EAP_TLS,
      EAP_PEAP,
      EAP_TTLS,
    } eap_method_t;

    eap_method_t method = EAP_TTLS;
    const char *identity = WPA2_IDENTITY;
    const char *username = WPA2_USERNAME;
    const char *password = WPA2_PASSWORD;

    wifi_station_set_wpa2_enterprise_auth(1);

    wifi_station_set_enterprise_identity((u8*)(void*)identity, os_strlen(identity));

    if (method == EAP_TLS) {
      Serial.println("error");
      //wifi_station_set_enterprise_cert_key(client_cert, os_strlen(client_cert) + 1, client_key, os_strlen(client_key) + 1, NULL, 1);
      //wifi_station_set_enterprise_username(username, os_strlen(username));//This is an option for EAP_PEAP and EAP_TLS.
    }
    else if (method == EAP_PEAP || method == EAP_TTLS) {
      wifi_station_set_enterprise_username((u8*)(void*)username, os_strlen(username));
      wifi_station_set_enterprise_password((u8*)(void*)password, os_strlen(password));
      //wifi_station_set_enterprise_ca_cert(ca, os_strlen(ca)+1);//This is an option for EAP_PEAP and EAP_TTLS.
    }
  }

  wifi_station_connect();

  // Wait for connection AND IP address from DHCP
  while (true)
  {
    Serial.print("Status: ");
    Serial.print(wifi_station_get_connect_status());
    
    Serial.print(" - Arduino status: ");
    Serial.print(WiFi.status());
    Serial.print(" - Local IP:");
    Serial.println(WiFi.localIP());
    delay(2000);
  }
} // setup
```

Thus, the question is:

Would you be able to propose an API to select at least the Phase2 authentication method ?

Thanks for your support

d-a-v, 2018-06-13:
@FayeY, in this statement of yours: ([link to thread](https://bbs.espressif.com/viewtopic.php?f=66&t=5962&p=15165#p15433))

> We will add an API to set auth mode in the future release

Is it the phase 2 authentication that you are referring to ?

kapyaar, 2018-06-16:
@d-a-v Thanks for opening this issue. I hope someone from espressif will address this soon. As for your PR about SDK3.0.0, did you mean that this version addresses the wpa2 issue?

kapyaar, 2018-06-21:
@d-a-v @FayeY  

Seems like things are quiet. Here is one thing I wonder. 

I came across https://github.com/JeroenBeemster/ESP32-WPA2-enterprise.

Now, I have not tested this, but I wonder if ESP32 is working, would it be possible to find the relavent files, and port it to ESP32? Is this even a possibility or are we talking apples and oranges here?

victorclaessen, 2018-06-21:
I have tried that example with my esp32 and the eduroam network. Didn't
work. Stuck waiting for wifi to connect.

d-a-v, 2018-06-21:
@kapyaar Maybe there exists some common code shared between esp32 and esp8266 regarding the physical layer firmware and specifically wpa2-enterprise. In that case, solving this issue for one would solve it for the other. 
Only espressif can say, because this part of the firmware is closed-source for both chips.

victorclaessen, 2018-06-25:
An update: apparently they're working on it: https://bbs.espressif.com/viewtopic.php?f=66&t=5962&start=10#p20892

d-a-v, 2018-06-26:
Latest commits (as of 89920dcc40aee30da11899c832f311e381312b38) do not solve this TTLS/PAP issue:
```
18:29:06.302 -> SDK:3.0.0-dev(c0f7b44)
18:29:06.302 -> Trying to connect to eduroam
18:29:06.368 -> WPA2 ENTERPRISE VERSION: [v2.0] enable
18:29:07.163 -> wifi evt: 2
18:29:09.250 -> scandone
18:29:10.212 -> state: 0 -> 2 (b0)
18:29:10.212 -> state: 2 -> 3 (0)
18:29:10.212 -> state: 3 -> 5 (10)
18:29:10.212 -> add 0
18:29:10.212 -> aid 15
18:29:10.212 -> cnt 
18:29:10.244 -> EAP-TTLS: Start (server ver=0, own ver=0)
18:29:13.793 -> EAP-TTLS: TLS done, proceed to Phase 2
18:29:13.793 -> EAP-TTLS: received 0 bytes encrypted data for Phase 2
18:29:13.793 -> EAP-TTLS: empty data in beginning of Phase 2 - use fake EAP-Request Identity
18:29:13.825 -> EAP-TTLS: Phase 2 MSCHAPV2 Request
18:29:14.852 -> state: 5 -> 2 (2a0)
18:29:14.852 -> rm 0
18:29:14.852 -> wifi evt: 1
18:29:14.885 -> STA disconnect: 2
```

kapyaar, 2018-07-05:
Are we stuck on this? I do have a wpa2 enterprise network (and also eduroam access), and going to test it out this weekend. Do I just need to copy the lib files from espressif repo to the arduino sdk folder, compile and run? I read on [here](https://github.com/esp8266/Arduino/tree/master/tools/sdk/lib) that I have to 

- Copy .a files from SDK lib directory to this directory
- Run fix_sdk_libs.sh

 I tried this on my windows 10, but  sh fix_sdk_libs.sh on terminal does not seem to work (in the correct directory offcourse). 

Am I doing it wrong?

d-a-v, 2018-07-05:
>  Do I just need to copy the lib files from espressif repo to the arduino sdk folder, 

This is an arduino specific discussion, if you wish to use the lastest nonos-sdk with arduino, check the arduino's [relevant PR](https://github.com/esp8266/Arduino/pull/4853).