espressif/ESP8266_NONOS_SDK #149: SSL Connections with AT-Commands doesn't support SNI

Boodorgad, 2018-07-12:
# SSL Connections with AT-Commands doesn't support SNI
     
We have a Server with multiple Virtual Hosts on it, listening on Port 443.
 
One of these virtual Hosts is the one we wan't to connect to, assume it's name is: "myserver.test.com"(example name)

## Problem
If our Virtual-Host is not the first Virtual-Host Item on the Server, all
Connection-Attempts with the Command: 
```c 
AT+CIPSTART="SSL","myserver.test.com",443,1200
```
just results in **ERROR** response without any additional Information and the Connection's closed.

Coincidentally, one of our Test-setup with multiple Virtual-Host results in a (unexpected) valid Connection
and we received the following important Data:

>403 Forbidden
>
>You don't have permission to access /api/xy.txt. on this server.
>
>Reason: The client software did not provide a hostname using Server Name Indication (SNI), which is required to access this server.

Used Versions 
- SDK 2.2.1(6ab97e9)
- AT: 1.6.2.0(Apr 13 2018 11:10:59)

MaJerle, 2018-07-26:
SNI is part of HTTP header, nothing to do with connection.

Did you add `Host: myserver.test.com\r\n` in HTTP request header?

MaJerle, 2018-07-26:
I'm deleting comment now as I mixed things. Sorry.

Boodorgad, 2018-11-06:
Ok, 5 Months without any comment, dear Espressif-Team why are you ignoring this issue?

We've tested it again with the current Versions and it's still not working!
Used Versions 
- SDK 3.1.0-dev(ea7d451)
- AT: 1.7.0.0(Aug 16 2018 00:57:04)

This User reported exact the same Problem, even without any response:
https://www.esp8266.com/viewtopic.php?f=12&t=13916

The Arduino-Team seems also to know the SNI-Problem and they do care about it  
https://github.com/esp8266/Arduino/issues/4469

Dear Espressif-Team, do you see any chance you may fix this issue in the near future?

marcelstoer, 2019-04-29:
@xcguang et.al. I know this SDK isn't getting much love from Espressif anymore these days. However, don't you think that SNI has become so prevalent that adding support for it is almost required if you offer TLS?

FayeY, 2019-04-29:
Hi,

Sorry that we may not add any new feature for the old ESP8266 non OS SDK. 
We are working on the new [ESP8266 RTOS SDK](https://github.com/espressif/ESP8266_RTOS_SDK) (IDF style) now, and there will be a new AT firmware based on it. Those will be the firmware we mainly focus on in the future. 
And for the SNI feature request, we will evaluate it after the new AT firmware released.
So sorry for the inconvenience.

Thanks.

bialabs, 2019-07-23:
i fixed in the past buying a dedicated IP address on my hosting but suddently this morning my sketch has stopped working. Now i get

```
>> AT+CIPSTART=3,"SSL","www.myhosting.it",443

ERROR
---------------------------------------------- > 1

Unable to connect to server
```

nothing is changed, the hosting still has a dedicated IP address and the sketch is working with google address . Anyone can help me?

OzEdri, 2021-07-21:
@Boodorgad Did you manage to solve this issue?
I know about a similar issue where trying to connect to Firebase (also by Google) with the AT+CIPSTART="SSL" command fails.
That said, connecting to other SSL sites with that command, including google.com, is working as expected.
I still didn't manage to check if using the IP instead of the DNS will help. I'll look into it.

@bialabs You wrote that you solved this issue by using a dedicated IP on your hosting, then wrote that it suddenly stopped working. What did you mean by dedicated IP and not shared one? And did you try connecting by using IP instead of DNS?