espressif/ESP8266_NONOS_SDK #153: wifi_station_get_connect_status() returns 255

uzi18, 2018-07-24:
Hi all!
Please explain what does it mean if wifi_station_get_connect_status() returns 255 ?
It is not documented anywhere.
I use SDK 2.0.0 

@wujiangang @FayeY please help me :)

xcguang, 2018-08-23:
Hi @uzi18 ,
If wifi mode is SOFTAP_MODE or NULL_MODE, or the connect is not established, the value wifi_station_get_connect_status() returns is 0xff.

xcguang, 2018-08-29:
Hi @uzi18 ,
Can this issue be closed? @FayeY

FayeY, 2018-08-29:
Closing as this question appears to be answered, please feel free to reopen if this is still an issue. Thanks.

closed 2018-08-29 by FayeY

uzi18, 2018-08-29:
ok thanks for info, please add this value to SDK :)

kriegste, 2018-08-29:
Stuff like this should be closed only after being properly documented in the pdf.

Inq720, 2021-11-09:
Oldie, but a goodie...  
In my code, I double checked my mode and status because of thread using this statement...
    os_printf("connect - mode=%u, status=%u\n", wifi_get_opmode(), wifi_station_get_connect_status());
...and got this...
   connect - mode=3, status=255

I'm on SDK version: 2.1.0(7106d38).  I'm a long time developer using the Arduino version, but new (3 days) to this native development using VSCode/PlatformIO and I haven't figured out how to upgrade it to a newer one.