espressif/ESP8266_NONOS_SDK #155: calloc (pvPortCallocIram) fails when it fills IRAM

gilpinheiro, 2018-07-31:
I have been having crashes in the mbedtls connection routines, on the current MASTER branch and spent some time to diagnose what was going on.

It looks like calloc is failing - the calloc request asks for more space than is available in what remains of the 0x4010xxxx-0x40108000 IRAM block - it is supposed to realize that, and return another buffer of the requested size in the 0x3fffxxxx (DRAM) range instead.

What it does: returns the 0x4010xxxx ptr (with insufficient space available), and writes to that buffer are  truncated at 0x40108000. Subsequent callocs work fine. 

```c
    #define COUNT 20
    #define BUFSIZE 1024
    void *buffers[COUNT];

    os_printf("create buffers...\n");
    int i=0;
    for (i=0; i<COUNT; i++) {
        //        buffers[i] = os_malloc(BUFSIZE); // This works
        buffers[i] = pvPortCallocIram(BUFSIZE, 1, "", __LINE__);
        os_printf("building buffer %d (0x%p)\n", i, buffers[i]);
        if (buffers[i] == NULL) {
            error("failed to alloc");
        } else {
            os_printf(".. buffer is 0x%p\n", buffers[i]);
            // fill buffer with value that we can verify to make sure it hasn't been
            // overwritten by another alloc'd buffer
            os_memset(buffers[i], i, BUFSIZE);
        }

        // do a quick check to make sure memset has worked
        uint8_t *d = buffers[i];
        for (int j=0; j<BUFSIZE; j++) {
            uint8_t v = *d++;
            if (v != i) {
                os_printf("failed write verify %d : [%d] = (%d)\n", i, j, v);
                print_bytes(buffers[i], BUFSIZE);
                goto FAIL_WRITE_VERIFY;
            }
        }
        os_printf(".. write verify passed\n");
        FAIL_WRITE_VERIFY:
        continue;
    }
```

```
create buffers...
building buffer 0 (0x40106c50)
.. buffer is 0x40106c50
.. write verify passed
building buffer 1 (0x40107060)
.. buffer is 0x40107060
.. write verify passed
building buffer 2 (0x40107470)
.. buffer is 0x40107470
.. write verify passed
building buffer 3 (0x40107880)
.. buffer is 0x40107880
.. write verify passed
building buffer 4 (0x40107c90)
.. buffer is 0x40107c90
failed write verify 4 : [880] = (0)
....
0x40107fe0:   04 04 04 04 04 04 04 04 04 04 04 04 04 04 04 04       ................
0x40107ff0:   04 04 04 04 04 04 04 04 04 04 04 04 04 04 04 04       ................
0x40108000:   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00       ................
0x40108010:   00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00       ................
...

building buffer 5 (0x3fff22a8)
.. buffer is 0x3fff22a8
.. write verify passed
building buffer 6 (0x3fff26b8)
.. buffer is 0x3fff26b8
.. write verify passed
building buffer 7 (0x3fff2ac8)
.. buffer is 0x3fff2ac8
.. write verify passed
building buffer 8 (0x3fff2ed8)
.. buffer is 0x3fff2ed8
.. write verify passed
building buffer 9 (0x3fff32e8)
.. buffer is 0x3fff32e8
.. write verify passed
building buffer 10 (0x3fff36f8)
.. buffer is 0x3fff36f8
.. write verify passed
building buffer 11 (0x3fff3b08)
.. buffer is 0x3fff3b08
.. write verify passed
building buffer 12 (0x3fff3f18)
.. buffer is 0x3fff3f18
.. write verify passed
building buffer 13 (0x3fff4328)
.. buffer is 0x3fff4328
.. write verify passed
building buffer 14 (0x3fff4738)
.. buffer is 0x3fff4738
.. write verify passed
building buffer 15 (0x3fff4b48)
.. buffer is 0x3fff4b48
.. write verify passed
building buffer 16 (0x3fff4f58)
.. buffer is 0x3fff4f58
.. write verify passed
building buffer 17 (0x3fff5368)
.. buffer is 0x3fff5368
.. write verify passed
building buffer 18 (0x3fff5778)
.. buffer is 0x3fff5778
.. write verify passed
building buffer 19 (0x3fff5b88)
.. buffer is 0x3fff5b88
.. write verify passed
```

xcguang, 2018-08-01:
Hi @gilpinheiro ,
We will reproduce it, and debug it. And we will reply here if have any update,  thanks.

gilpinheiro, 2018-08-09:
Hi @xcguang 

I've updated to latest build @6dcfc86e186b6e37e1b5c22aaf19086fbc659e7c

pvPortCallocIram() is exhibiting different bad behaviour now.

I get failures just doing:

```
#define COUNT 50
#define BUFSIZE 4096
    void *buffers[COUNT];

    os_printf("create buffers...\n");
    int i=0;
    for (i=0; i<COUNT; i++) {
        buffers[i] = pvPortCallocIram(BUFSIZE, 1, "", __LINE__);
        if (buffers[i] == NULL) {
            os_printf("failed to alloc");
        } else {
            os_printf(".. buffer is 0x%p\n", buffers[i]);
        }
    }
    os_printf("done\n");
```
result:

```
create buffers...
.. buffer is 0x0x40107a30
.. buffer is 0x0x3ffeff78
.. buffer is 0x0x3fff0f88
.. buffer is 0x0x3fff1f98
.. buffer is 0x0x3fff2fa8
.. buffer is 0x0x3fff3fb8
.. buffer is 0x0x3fff4fc8
.. buffer is 0x0x3fff5fd8
.. buffer is 0x0x3fff6fe8
.. buffer is 0x0x3fff7ff8
.. buffer is 0x0x3fff9008
.. buffer is 0x0x3fffa018
.. buffer is 0x0x3fffb028
.. buffer is 0x
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
```

device sits at that last 0x for a long time before watchdog triggers. I'm not sure what the cause could be -- it seems like an odd place to get stuck - I thought it could be something unrelated in the print function, but removing all print statements still results in a watchdog failure (we never reach the end of the containing function).

Doesn't matter whether or not user_iram_memory_is_enabled() is true or false.

xcguang, 2018-08-13:
Hi,
Did you suspend the task schedule? Could you disable these log, and only print them when fail?

gilpinheiro, 2018-08-13:
I've tried it with no user-defined tasks running at all - previously it had been run during a timer callback after user_init. 

Now I've tried it there, and directly in user_init, with no change in behaviour. 

Enabling or disabling printing doesn't change the behaviour either.

My guess is that there is a exception occurring, and it is being caught by the new aligned load/store exception handler unexpectedly.

gilpinheiro, 2018-08-22:
Can anyone reproduce this with the current master? It has been driving me crazy, and I'd like some validation that this isn't a problem with my project.

This code will crash, but you should see that the allocations continue past the end of dram (instead of returning NULL). 

```
    while(true) {
        void *p = pvPortMalloc(1024, __FILE__, __LINE__, true);
        if (p == NULL)
            os_printf("NULL\n");
        else if ((p >= 0x3ffe8000) && (p <= 0x3fffbfff)) {
            os_printf("0x%p (DRAM)\n", p);
        } else if ((p >= 0x40100000) && (p <= 0x40107fff)) {
            os_printf("0x%p (IRAM)\n", p);
        } else {
            os_printf("0x%p (!!!)\n", p);
        }
    }
```

```
0x0x40106e80 (IRAM)
0x0x40107290 (IRAM)
0x0x401076a0 (IRAM)
0x0x40107ab0 (IRAM)
0x0x40107ec0 (IRAM)
0x0x3ffeec88 (DRAM)
0x0x3ffef098 (DRAM)
0x0x3ffef4a8 (DRAM)
0x0x3ffef8b8 (DRAM)
0x0x3ffefcc8 (DRAM)
0x0x3fff00d8 (DRAM)
0x0x3fff04e8 (DRAM)
0x0x3fff08f8 (DRAM)
0x0x3fff0d08 (DRAM)
0x0x3fff1118 (DRAM)
0x0x3fff1528 (DRAM)
0x0x3fff1938 (DRAM)
0x0x3fff1d48 (DRAM)
0x0x3fff2158 (DRAM)
0x0x3fff2568 (DRAM)
0x0x3fff2978 (DRAM)
0x0x3fff2d88 (DRAM)
0x0x3fff3198 (DRAM)
0x0x3fff35a8 (DRAM)
0x0x3fff39b8 (DRAM)
0x0x3fff3dc8 (DRAM)
0x0x3fff41d8 (DRAM)
0x0x3fff45e8 (DRAM)
0x0x3fff49f8 (DRAM)
0x0x3fff4e08 (DRAM)
0x0x3fff5218 (DRAM)
0x0x3fff5628 (DRAM)
0x0x3fff5a38 (DRAM)
0x0x3fff5e48 (DRAM)
0x0x3fff6258 (DRAM)
0x0x3fff6668 (DRAM)
0x0x3fff6a78 (DRAM)
0x0x3fff6e88 (DRAM)
0x0x3fff7298 (DRAM)
0x0x3fff76a8 (DRAM)
0x0x3fff7ab8 (DRAM)
0x0x3fff7ec8 (DRAM)
0x0x3fff82d8 (DRAM)
0x0x3fff86e8 (DRAM)
0x0x3fff8af8 (DRAM)
0x0x3fff8f08 (DRAM)
0x0x3fff9318 (DRAM)
0x0x3fff9728 (DRAM)
0x0x3fff9b38 (DRAM)
0x0x3fff9f48 (DRAM)
0x0x3fffa358 (DRAM)
0x0x3fffa768 (DRAM)
0x0x3fffab78 (DRAM)
0x0x3fffaf88 (DRAM)
0x0x3fffb398 (DRAM)
0x0x3fffb7a8 (DRAM)
0x0x3fffbbb8 (DRAM)
0x0x3fffbfc8 (DRAM)
0x0x3fffc3d8 (!!!)
0x0x3fffc7e8 (!!!)
0x0x3fffcbf8 (!!!)
0x0x3fffd008 (!!!)
0x0x3fffd418 (!!!)
0x0x3fffd828 (!!!)
0x0x3fffdc38 (!!!)
0x0x3fffe048 (!!!)
0x0x3fffe458 (!!!)
0x0x3fffe868 (!!!)
0x0x3fffec78 (!!!)
0x0x3ffff088 (!!!)
0x0x3ffff498 (!!!)
0x0x3ffff8a8 (!!!)
```
From what I can tell, all the malloc functions are allocating buffers into "ets system data" ram space ( >3FFFC000h) which eventually overwrites something important (probably the malloc related data structures themselves).

gilpinheiro, 2018-08-28:
Okay, I went through the process of using your VM and a clean copy of the sdk 3.0 - this issue seems to be reproducible, and not because of something I've done in my project.

@xcguang I think this is important.

Allocations continue into 0x3FFFC000h (which was previously marked as reserved system memory), and will overwrite system data causing crashes.

```c
    while(true) {
        void *p = (void *) pvPortMalloc(1024, __FILE__, __LINE__, false);        // true/false doesn't make a difference
        if (p == NULL)
            os_printf("NULL\n");
        else if (((uint32_t) p >= 0x3ffe8000) && ((uint32_t)p <= 0x3fffbfff)) {
            os_printf("%p (DRAM)\n", p);
        } else if (((uint32_t)p >= 0x40100000) && ((uint32_t)p <= 0x40107fff)) {
            os_printf("%p (IRAM)\n", p);
        } else {
            os_printf("%p (!!!)\n", p); // outside of expected heap memory range
        }
        os_memset(p, 0, 1024); // added memset - sometimes you get luck and the malloc datastructures don't overwrite anything vital: this ensures a crash for me
    }
```

xcguang, 2018-08-29:
Hi @FayeY,
Please create a teamwork to trace this issue.

xcguang, 2018-08-30:
Hi @gilpinheiro ,
Yes, you're right. Please test it using the attachment, and we will sync at GitHub  after code is reviewed.
[libmain.a.zip](https://github.com/espressif/ESP8266_NONOS_SDK/files/2335346/libmain.a.zip)

kriegste, 2018-08-30:
The new libmain is stable here, but free heap dropped by 16kB.
CONFIG_ENABLE_IRAM_MEMORY is implemented as described in the manual.

xcguang, 2018-08-31:
Hi @kriegste ,
The old lib misuses 0x3fffc000~0x4000000 as memory, and it should be used for other function.

xcguang, 2018-09-05:
Hi @kriegste @gilpinheiro  ,
Have fixed it in branch release/v3.0.0 (commit 86c46d0) and branch master (commit id d948d7e4).

closed 2018-09-06 by FayeY