espressif/ESP8266_NONOS_SDK #157: system_deep_sleep with max time sometimes doesn't wake up, possible race condition

devyte, 2018-08-03:
Reference discussion in Arduino core issue tracker [here](https://github.com/esp8266/Arduino/pull/4936).
CC @marcvtew
CC @5chufti

Hi,

Per [NONOS SDK api reference doc](https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf)  section 3.3.9:
> The theoretical maximum value of time_in_us can be calculated by formula:
(time_in_us	/	cali)	<<	12	=	2^31	-	1
> -  cali	=	system_rtc_clock_cali_proc(), the cali is the RTC clock period (in
us); bit11 ~ bit0 are decimal. For more details about the cali, please see the
API: system_rtc_clock_cali_proc

From the previous follows [this implementation](https://github.com/esp8266/Arduino/blob/75a6a3f0730a2c20af7b444f2e0044e39e77607c/cores/esp8266/Esp.cpp#L119) for calculating the max theoretical sleep time::
```
uint64_t deepSleepMax()
{
  //cali*(2^31-1)/(2^12)
  return (uint64_t)system_rtc_clock_cali_proc()*(0x80000000-1)/(0x1000);
}
```

However, calling this:
`    system_deep_sleep(deepSleepMax());
`
seems to have odd behavior. Sometimes there is an error message in the serial output saying sleep time too long, sometimes the ESP sleeps, and sometimes the ESP doesn't wake up.

The [NONOS SDK api reference doc](https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf) says in section 3.3.9, in Note, 3rd bullet:
> After configuration, the chip will not enter Deep-sleep mode immediately, but will wait for a while till the Wi-Fi core is closed safely.

I suspect that:
1. there is a race condition, because it is possible that the max theoretical value for sleep changes between the time the user calls system_deep_sleep() and the actual sleep begins
2. If system_deep_sleep() is called with a value larger than the max theoretical value, the ESP could never wake up again
3. If there is a check on the SDK code side that checks against some max value, that it suffers from the same race condition above, which means, which means that it is still possible for a value that is too big to reach the sleep code, and make the ESP never wake up.

devyte, 2018-08-03:
I should also note that I see 2 distinct use cases, neither of which is really addressed by the current SDK api:
1. sleep for a specific time, which should always be "safe"
2. sleep for the max time possible, whatever that time is, even if that max time changes over time

Point 1 hints at some way to calculate a max sleep time that will always be safe and can never fail.
Point 2 requires that either the current suspected race condition be fixed, or a new api be implemented that sleeps for the max possible time independently of the value of cali, e.g.:
`system_deep_sleep_max_time(); //sleep for max possible time (receives no arguments)`

marcvtew, 2018-08-04:
Function system_deep_sleep() is of type bool. I would suggest he api should return 'false' when max sleeptime is exceeded. Currently it does not.

xcguang, 2018-08-29:
@FayeY