espressif/ESP8266_NONOS_SDK #158: register the partition table in non-FOTA bin

liuyang-z, 2018-08-07:
How can I modify the partition table in non-FOTA example before registering the table? Please give some references.

gilpinheiro, 2018-08-09:
#159 might help.

I made the partition table something like:

```
static const partition_item_t at_partition_table[] = {
    {SYSTEM_PARTITION_RF_CAL, 0x400000 - 0x3000 - 0x1000 - 0x1000, 0x1000},
    {SYSTEM_PARTITION_PHY_DATA, 0x400000 - 0x3000 - 0x1000, 0x1000},
    {SYSTEM_PARTITION_SYSTEM_PARAMETER, 0x400000 - 0x3000, 0x3000}, 
};

```
0x400000 is the flash size of my device (4mb), and this puts the three configuration partitions beside each other at the end of flash.

ustccw, 2018-08-15:
@liuyang-z @gilpinheiro 
we have partition table in [at example](https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/examples/at/user/user_main.c) for user.
please take a look.

gilpinheiro, 2018-08-15:
We did, which is why we are here: those examples all assume we are using your bootloader and FOTA, but many projects do not. See #159.

ustccw, 2018-08-15:
@gilpinheiro 
user could choose not register partition table by not-calling `system_partition_table_regist`.

kriegste, 2018-08-15:
In the README it says

"you MUST call system_partition_table_regist() in user_pre_init"

https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/README.md

gilpinheiro, 2018-08-15:
And it isn't quite true. The system crashes during boot:

```
>> 0x0,0x0rf_cal[0] !=0x05,is 0xE9
```

I'm guessing that it is looking somewhere for the rf calibration, but it either isn't being set to the previous default or at least the user_rf_cal_sector_set() function isn't being invoked, which I ask about in #159, as this is the way that value was previously set.

ustccw, 2018-08-16:
@kriegste @gilpinheiro sorry for my mistake. 
it is true that "you MUST call system_partition_table_regist() in user_pre_init".

system get phy init parameter by partition table. so we must register. otherwises, system would be force to crashes at rf cal,  and print `>> 0x0,0x0rf_cal[0] !=0x05,is 0xE9`

ustccw, 2018-08-16:
Partition table is a security way to protect and save system info and user data. 
Before that, some customers could encounter strange questions and bugs because of misoperating the data in flash. And now, partition table can avoid it.

gilpinheiro, 2018-08-16:
Yes, I think this is a good feature. Thank you for working on it.

I hope you can use some of our suggestions to improve it further so it works well for everyone. 

Adding documentation about how to use it properly without OTA would be especially helpful. What I did 
in my first comment works, but is it the right way to do it?

ustccw, 2018-08-16:
@gilpinheiro 
thanks for your suggestions, we would mark it to inner.

as for non-OTA mode, you could define a partition table like the following:
```
static const partition_item_t at_partition_table[] = {
    { SYSTEM_PARTITION_BOOTLOADER, 	0x0, 0x1000},
    { SYSTEM_PARTITION_RF_CAL, SYSTEM_PARTITION_RF_CAL_ADDR, 0x1000},
    { SYSTEM_PARTITION_PHY_DATA, SYSTEM_PARTITION_PHY_DATA_ADDR, 0x1000},
    { SYSTEM_PARTITION_SYSTEM_PARAMETER,SYSTEM_PARTITION_SYSTEM_PARAMETER_ADDR, 0x3000},
    { SYSTEM_PARTITION_CUSTOMER_BEGIN, 0x2000, 0x8000},
};
```

`SYSTEM_PARTITION_CUSTOMER_BEGIN` to mark user code start, 
enter user code by call 
```
system_partition_get_item(SYSTEM_PARTITION_CUSTOMER_BEGIN, partition_item);
```

ustccw, 2018-08-20:
@FayeY please addition titile to #TW25584

closed 2018-09-21 by xcguang