espressif/ESP8266_NONOS_SDK #160: ESP8266 - I2C DS3231 Fail

minhhieuec, 2018-08-12:
Hi espressif!
i tried user_sensor example IoT_Demo for DS3231, I repaired 0x88 (MVH3004_Addr) to 0x68 (DS3231 Addr), after that i read 7 bytes from i2c bus. But, when call user_mvh3004_burst_read() function, it return false - (no ACK os_printf("addr not ack when tx write cmd \n");).

Could you help me to fix it?

davydnorris, 2018-08-24:
This device uses clock stretching on the I2C bus. The default I2C driver code in the SDK doesn't implement clock stretching. This is why it's failing.

There are several implementations out there that have clock stretch. Contact me if you want mine, which is based on Pete Scargill's code

minhhieuec, 2018-08-24:
@davydnorris  thanks for your information,
Please send to me your example code or document of this?
My Email: minhhieuec@gmail.com

best regard!

valkuc, 2018-08-24:
I'm using https://github.com/pasko-zh/brzo_i2c in several projects. It's support clock stretching and work stable at 400 Khz speed.

eriksl, 2018-09-07:
My I2C code also supports a wide variety of I2C "quirks". Every now and then I attach an oscilloscope to see if it really does what it should do. For 100 kHz standard mode on 80 MHz cpu and 400 kHz fast mode on 160 MHz, it takes care to only use proper timings, i.e. only change SDA when SCL is exactly in the middle of a cycle and prevent glitches of > 100 or > 400 kHz. It of course also supports clock stretching and faulty devices that tend to loose bits every now and then and not releasing the bus in time.

https://github.com/eriksl/esp8266-universal-io-bridge

The I2C code delivered as "example code" with the SDK is very poor.

devyte, 2018-09-07:
@eriksl your repo is licensed under GPL, and I understand that makes it incompatible with Espressif's sdk for any project that requires distribution. That means only personal projects could make use of it.
If you would like a wider audience, I suggest changing the license to LGPL, MIT-like, or similar.

eriksl, 2018-09-08:
I never understood the exact semantics of GPL vs. LGPL vs. MIT etc.

On the other hand, I do endorse the spirit of GPL where open source remains true open source. I fear that if I change the license to LGPL or MIT, my code can end up somewhere in a commercial project without that code being open. Ask the ffmpeg developers what happens... I don't want that. I do this in my spare time, so I think I may ask something in return (i.o.w. your code).

As long as that is guaranteed, I am open to changing the license.

eriksl, 2018-09-08:
What a strange conclusion?

marcelstoer, 2018-09-08:
This discussion is quickly getting OT. I'll keep it short.

> I fear that if I change the license to LGPL or MIT, my code can end up somewhere in a commercial project without that code being open.
> I never understood the exact semantics of GPL vs. LGPL vs. MIT etc.

Not being a subject matter expert myself it's hard to judge but I think you understand quite well. GPL is the right choice then for you.

> Erik, in this case it is better not to advertise your code.

Nonsense, what has one got to do with the other? Are you seriously suggesting [popular GPL projects](https://en.wikipedia.org/wiki/Category:Software_using_the_GPL_license) like WordPress, MariaDB, Ansible et.al. better kept their code to themselves?

closed 2019-03-14 by xcguang