espressif/ESP8266_NONOS_SDK #161: [AT] Reason for AT+CIPSTART failure

MaJerle, 2018-08-15:
Currently there are some *debug messages* for `AT+CIPSTART` command, such as `DNS fail`. Do you plan to release official response for failure, such as `+CIPSTARTFAILURE:x`?

MaJerle, 2018-08-29:
Hello,
Why is this request ignored by Espressif team?

xcguang, 2018-08-30:
Yes,  we will add some messages by AT+SYSMSG.

MaJerle, 2018-09-21:
@xcguang,

If I focus on ticket: https://github.com/espressif/ESP8266_NONOS_SDK/issues/65
Also add reason for failure like *this connection is already populated, please go with another one*, so that we know everything could be ok but we have to go with different connection.

MaJerle, 2019-03-15:
@xcguang how do we stand here, please?

Thanks.

xcguang, 2019-03-15:
Hi,
We will add none feature in the NonOS AT, is it OK adding it in the RTOS AT?

MaJerle, 2019-03-15:
@xcguang 
- Will be RTOS-AT compatible with Non-OS AT in terms of AT commands?
- When can we expect first RTOS-AT release?

Thanks.

xcguang, 2019-03-15:
Hi,
- ESP8266 RTOS AT is compatible with ESP32 AT. And we will add a document to explain the difference among ESP8266 NonOS AT, ESP8266 RTOS AT and ESP32 AT.
- ESP8266 RTOS AT will be release in Q2, the project has been merged into https://github.com/espressif/esp32-at, but we will do more work to optimize memory.
- For ESP8266 RTOS AT, it will depend on the memory limitation, if memory is affluent, we are willing to more function.

MaJerle, 2019-03-15:
Thanks @xcguang 

Ok, this sounds good. I'm willing to become beta tester or similar to adapt my library for new firmware approach.

If you have any binary I can flash now to ESP8266, I'm willing to do so. I see some commits on esp32-at repo are saying "supporting esp8266".

xcguang, 2019-03-15:
Yes, but currently you have to run `make` by yourself.
And you need modify https://github.com/espressif/esp32-at/blob/master/Makefile 
```
export ESP_AT_PROJECT_PLATFORM ?= PLATFORM_ESP32
export ESP_AT_MODULE_NAME ?= WROOM-32
```
into
```
export ESP_AT_PROJECT_PLATFORM ?= PLATFORM_ESP8266
export ESP_AT_MODULE_NAME ?= WROOM-02
```

You can select an `ESP_AT_PROJECT_PLATFORM ` and `ESP_AT_MODULE_NAME ` from https://github.com/espressif/esp32-at/blob/master/components/customized_partitions/raw_data/factory_param/factory_param_data.csv

xcguang, 2019-03-15:
Or you can try my firmware which just for test.

``` python /esp-at/esp-idf/components/esptool_py/esptool/esptool.py --chip esp8266 --port /dev/ttyUSB0 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size 4MB 0x0000 /esp-at/build/bootloader/bootloader.bin 0xBF000 /esp-at/build/at_customize.bin 0xC0000 /esp-at/build/customized_partitions/factory_param.bin 0xC1000 /esp-at/build/customized_partitions/server_cert.bin 0xC3000 /esp-at/build/customized_partitions/server_key.bin 0xC5000 /esp-at/build/customized_partitions/server_ca.bin 0xC7000 /esp-at/build/customized_partitions/client_cert.bin 0xC9000 /esp-at/build/customized_partitions/client_key.bin 0xCB000 /esp-at/build/customized_partitions/client_ca.bin 0x10000 /esp-at/build/esp-at.bin 0x8000 /esp-at/build/partitions_at.bin```

[build_test.zip](https://github.com/espressif/ESP8266_NONOS_SDK/files/2970372/build_test.zip)

MaJerle, 2019-03-15:
Thanks. So I focus on ESP32 AT commands document and I can run it on ESP8266?
Thanks. @xcguang

xcguang, 2019-03-15:
Not all command can work because of lack of memory.   
And we have not tested it, so maybe there are some bugs

closed 2020-12-17 by MaJerle