espressif/ESP8266_NONOS_SDK #162: SDK is getting fat (relief inside)

kriegste, 2018-08-18:
Like most other software the SDK is getting bigger and bigger and some people might already hit a limit with their tiny EEPROMs. We soon will.

So I tried some gcc optimization switches and found out that

-fno-guess-branch-probability
-freorder-blocks-and-partition
-fno-cse-follow-jumps

in the Makefile save me a few kilobytes. I even recompiled lwip and mbedtls to save more. These switches only affect the automatic optimization process in gcc. They do not alter the function of the code.
So my request would be that Espressif compiled the other libs with these switches to find out if more space can be saved.

gilpinheiro, 2018-08-20:
Thanks for sharing that. I gave it a try in my project (I'm also compiling in mbedtls) and I saved about 2.2k. Modest but nice.

fvpalha, 2018-08-20:
I saved 448 bytes. Thank you.

kriegste, 2018-08-20:
Remove -O2 and you should save more (it overwrites the preceding -Os which is better).

fvpalha, 2018-08-21:
Thank you @kriegste 
Now I saved 2032 bytes.

`CFLAGS  = -I. -w -Os -ggdb -std=c99 -g -Wpointer-arith -Wundef \
          -Wall -Werror -Wno-implicit -Wl,-EL -Wno-implicit-function-declaration \
          -fno-exceptions -fno-inline-functions -nostdlib -mlongcalls \
          -mtext-section-literals -ffunction-sections -fdata-sections \
          -fno-builtin-printf -fno-jump-tables -mno-serialize-volatile \
          -fno-guess-branch-probability -freorder-blocks-and-partition -fno-cse-follow-jumps \
          -D__ets__ -DICACHE_FLASH -DUSE_US_TIMER -DUSE_OPTIMIZE_PRINTF`

kriegste, 2018-08-23:
As a side note:
https://github.com/espressif/ESP8266_NONOS_SDK/commit/c172368f24d435b48bb1788adf4db27c7447fbde
may not be necessary (yet) if SDK and AT were compiled with those optimization parameters...

xcguang, 2018-08-23:
@kriegste 
Yes，because the bin is too large, and if want to use 515+512 map，you have to layout partition table and recompile it(the libmbedtls.a is larger 40KB+ than libssl.a, maybe you can try it). About this, we are writing documents.

By the way, your gcc optimization is very wonderful, and we will and them next version.

kriegste, 2018-08-25:
I personally do not use the AT firmware so I will not touch it.

But after a quick look it seems there still is plenty of space between 0x01000 and 0x79000 (480 kB) and 0x81000 and 0xFB000 (488 kB).

Editing the original ld file is necessary, I think. For example I am currently squeezing two (OTA) 492 kB images into a 1 MB flash without any problems.

```
#define SYSTEM_PARTITION_OTA_SIZE							0x6A000
#define SYSTEM_PARTITION_OTA_2_ADDR							0x81000
#define SYSTEM_PARTITION_RF_CAL_ADDR						0xfb000
#define SYSTEM_PARTITION_PHY_DATA_ADDR						0xfc000
#define SYSTEM_PARTITION_SYSTEM_PARAMETER_ADDR				0xfd000
#define SYSTEM_PARTITION_AT_PARAMETER_ADDR					0x7d000
#define SYSTEM_PARTITION_SSL_CLIENT_CERT_PRIVKEY_ADDR		0x7c000
#define SYSTEM_PARTITION_SSL_CLIENT_CA_ADDR					0x7b000
#define SYSTEM_PARTITION_WPA2_ENTERPRISE_CERT_PRIVKEY_ADDR	0x7a000
#define SYSTEM_PARTITION_WPA2_ENTERPRISE_CA_ADDR			0x79000

static const partition_item_t at_partition_table[] = {
	{SYSTEM_PARTITION_BOOTLOADER,					0x0, 												0x1000},
	{SYSTEM_PARTITION_OTA_1,						0x1000,												SYSTEM_PARTITION_OTA_SIZE},
	{SYSTEM_PARTITION_OTA_2,						SYSTEM_PARTITION_OTA_2_ADDR, 						SYSTEM_PARTITION_OTA_SIZE},
	{SYSTEM_PARTITION_RF_CAL,						SYSTEM_PARTITION_RF_CAL_ADDR, 						0x1000},
	{SYSTEM_PARTITION_PHY_DATA,						SYSTEM_PARTITION_PHY_DATA_ADDR, 					0x1000},
	{SYSTEM_PARTITION_SYSTEM_PARAMETER,				SYSTEM_PARTITION_SYSTEM_PARAMETER_ADDR, 			0x3000},
	{SYSTEM_PARTITION_AT_PARAMETER,					SYSTEM_PARTITION_AT_PARAMETER_ADDR, 				0x3000},
	{SYSTEM_PARTITION_SSL_CLIENT_CERT_PRIVKEY,		SYSTEM_PARTITION_SSL_CLIENT_CERT_PRIVKEY_ADDR, 		0x1000},
	{SYSTEM_PARTITION_SSL_CLIENT_CA,				SYSTEM_PARTITION_SSL_CLIENT_CA_ADDR, 				0x1000},
	#ifdef CONFIG_AT_WPA2_ENTERPRISE_COMMAND_ENABLE
		{SYSTEM_PARTITION_WPA2_ENTERPRISE_CERT_PRIVKEY,	SYSTEM_PARTITION_WPA2_ENTERPRISE_CERT_PRIVKEY_ADDR,	0x1000},
		{SYSTEM_PARTITION_WPA2_ENTERPRISE_CA,			SYSTEM_PARTITION_WPA2_ENTERPRISE_CA_ADDR,			0x1000},
	#endif
};
```

kriegste, 2018-08-27:
You can try setting

line 8 in
https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/ld/eagle.app.v6.new.1024.app1.ld
to
`  irom0_0_seg :                         org = 0x40201010, len = 0x78000

and

line 8 in
https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/ld/eagle.app.v6.new.1024.app2.ld
to
`  irom0_0_seg :                         org = 0x40281010, len = 0x78000

Then recompile and see if it fits now (it should).

xcguang, 2019-02-21:
Added the following flags in the latest master branch

`
-fno-guess-branch-probability
-freorder-blocks-and-partition
-fno-cse-follow-jumps
`

The internal lib is compiled by xt-xcc, and the xt-xcc does not support these flags. If compiling the internal lib using gcc with these flags, it seems more bigger, so I gave up.

kriegste, 2019-02-21:
Thanks much for trying!

closed 2019-03-14 by xcguang