espressif/ESP8266_NONOS_SDK #163: How to use rBoot with v3.0 ?

fvpalha, 2018-08-24:
Hi.

How I can to configure the partition table in FOTA with rBoot (https://github.com/raburton/rboot) ?

I thinked:

> #if ((SPI_FLASH_SIZE_MAP == 0) || (SPI_FLASH_SIZE_MAP == 1))
> #error "The flash map is not supported"
> #elif (SPI_FLASH_SIZE_MAP == 2)
> #define SYSTEM_PARTITION_OTA_SIZE							0x6A000
> #define SYSTEM_PARTITION_OTA_1_ADDR						0x02000
> #define SYSTEM_PARTITION_OTA_2_ADDR						0x82000
> #define SYSTEM_PARTITION_RF_CAL_ADDR						0xfb000
> #define SYSTEM_PARTITION_PHY_DATA_ADDR					0xfc000
> #define SYSTEM_PARTITION_SYSTEM_PARAMETER_ADDR			0xfd000
> #elif (SPI_FLASH_SIZE_MAP == 3)
> #define SYSTEM_PARTITION_OTA_SIZE							0x6A000
> #define SYSTEM_PARTITION_OTA_1_ADDR						0x02000
> #define SYSTEM_PARTITION_OTA_2_ADDR						0x82000
> #define SYSTEM_PARTITION_RF_CAL_ADDR						0x1fb000
> #define SYSTEM_PARTITION_PHY_DATA_ADDR					0x1fc000
> #define SYSTEM_PARTITION_SYSTEM_PARAMETER_ADDR			0x1fd000
> #elif (SPI_FLASH_SIZE_MAP == 4)
> #define SYSTEM_PARTITION_OTA_SIZE							0x6A000
> #define SYSTEM_PARTITION_OTA_1_ADDR						0x02000
> #define SYSTEM_PARTITION_OTA_2_ADDR						0x82000
> #define SYSTEM_PARTITION_RF_CAL_ADDR						0x3fb000
> #define SYSTEM_PARTITION_PHY_DATA_ADDR					0x3fc000
> #define SYSTEM_PARTITION_SYSTEM_PARAMETER_ADDR			0x3fd000
> #elif (SPI_FLASH_SIZE_MAP == 5)
> #define SYSTEM_PARTITION_OTA_SIZE							0x6A000
> #define SYSTEM_PARTITION_OTA_1_ADDR						0x02000
> #define SYSTEM_PARTITION_OTA_2_ADDR						0x102000
> #define SYSTEM_PARTITION_RF_CAL_ADDR						0x1fb000
> #define SYSTEM_PARTITION_PHY_DATA_ADDR					0x1fc000
> #define SYSTEM_PARTITION_SYSTEM_PARAMETER_ADDR			0x1fd000
> #elif (SPI_FLASH_SIZE_MAP == 6)
> #define SYSTEM_PARTITION_OTA_SIZE							0x6A000
> #define SYSTEM_PARTITION_OTA_1_ADDR						0x02000
> #define SYSTEM_PARTITION_OTA_2_ADDR						0x102000
> #define SYSTEM_PARTITION_RF_CAL_ADDR						0x3fb000
> #define SYSTEM_PARTITION_PHY_DATA_ADDR					0x3fc000
> #define SYSTEM_PARTITION_SYSTEM_PARAMETER_ADDR			0x3fd000
> #else
> #error "The flash map is not supported"
> #endif

@raburton, do you have a tip?

Thank you.

xcguang, 2018-08-24:
Hi,
Maybe https://github.com/espressif/ESP8266_NONOS_SDK/blob/release/v3.0.0/documents/EN/%20Partition%20Table.md can help you.

fvpalha, 2018-08-24:
Hi @xcguang 

My doubt is about the "MUST", in rBoot the addresses are 0x2000 and 0x82000 or 0x102000.

> user1.bin MUST be at address 0x1000.
user2.bin:
For 512+512 map, MUST be at address 0x81000.
For 1024+1024 map, MUST be at address 0x101000.

xcguang, 2018-08-24:
Hi @fvpalha ,
I guess the following code would work well for rBoot, but I'm not sure. Next Monday I will try rBoot,if it doesn't work, we will provide a solution. Sorry for the inconvenience.
```
#define RBOOT_OTA1				（SYSTEM_PARTITION_CUSTOMER_BEGIN + 1）
#define RBOOT_OTA2 			    （SYSTEM_PARTITION_CUSTOMER_BEGIN + 2）

static const partition_item_t partition_table[] = {
    { SYSTEM_PARTITION_BOOTLOADER, 	0x0000, 0x2000},
    { RBOOT_OTA1, 	0x2000, 0x60000},
    { RBOOT_OTA2, 0x82000, 0x60000},
    { SYSTEM_PARTITION_RF_CAL, SYSTEM_PARTITION_RF_CAL_ADDR, 0x1000},
    { SYSTEM_PARTITION_PHY_DATA, SYSTEM_PARTITION_PHY_DATA_ADDR, 0x1000},
    { SYSTEM_PARTITION_SYSTEM_PARAMETER,SYSTEM_PARTITION_SYSTEM_PARAMETER_ADDR, 0x3000},
};

void ICACHE_FLASH_ATTR user_pre_init(void)
{
    if(!system_partition_table_regist(partition_table, sizeof(partition_table)/sizeof(partition_table[0]),SPI_FLASH_SIZE_MAP)) {
		os_printf("system_partition_table_regist fail\r\n");
		while(1);
	}
}
```

fvpalha, 2018-08-24:
Thank you @xcguang 

I go to wait the end solution.

Congratulation by the SDK.

ustccw, 2018-08-27:
@fvpalha would you mind providing the bin file of `rboot` for us to test it.

ustccw, 2018-08-27:
@fvpalha 
if `rboot` works well with sdk which not including partition table.
the following partition table also can work.
```
#define RBOOT_OTA1				（SYSTEM_PARTITION_CUSTOMER_BEGIN + 1）
#define RBOOT_OTA2 			    （SYSTEM_PARTITION_CUSTOMER_BEGIN + 2）

static const partition_item_t partition_table[] = {
    { SYSTEM_PARTITION_BOOTLOADER, 	0x0000, 0x2000},
    { RBOOT_OTA1, 	0x2000, 0x60000},
    { RBOOT_OTA2, 0x82000, 0x60000},
    { SYSTEM_PARTITION_RF_CAL, SYSTEM_PARTITION_RF_CAL_ADDR, 0x1000},
    { SYSTEM_PARTITION_PHY_DATA, SYSTEM_PARTITION_PHY_DATA_ADDR, 0x1000},
    { SYSTEM_PARTITION_SYSTEM_PARAMETER,SYSTEM_PARTITION_SYSTEM_PARAMETER_ADDR, 0x3000},
};

void ICACHE_FLASH_ATTR user_pre_init(void)
{
    if(!system_partition_table_regist(partition_table, sizeof(partition_table)/sizeof(partition_table[0]),SPI_FLASH_SIZE_MAP)) {
		os_printf("system_partition_table_regist fail\r\n");
		while(1);
	}
}
```

fvpalha, 2018-08-27:
@ustccw the binary can be founded in https://github.com/fvpalha/rboot

ustccw, 2018-08-28:
@fvpalha hi, rboot cannot work, error always happened like the following, could you give me some advice: 
[image]

fvpalha, 2018-08-28:
Hi @ustccw .
I will wait the @raburton solution.
At moment, I use NON-OS SDK v2.2.1

someburner, 2018-09-04:
I have not tried yet, but if this is true, this represents a breaking change in the SDK. At least give us the option of not using the partition table. The math is really not that hard to figure out and many, many projects have been happily using rBoot for years now.

fvpalha, 2018-09-05:
Hi @xcguang ,

Right, I will try again with your suggestion.

Thank you.

eriksl, 2018-09-06:
I think (famous last words...) that it is going to work if you don't mention OTA at all. Don't use the SDK OTA partitions, use "user defined" partitions there. Rboot OTA is not the same as SDK OTA.

I will report.

eriksl, 2018-09-06:
It works :-)

The only hoop I had to jump through was that my flash is 16 Mb, 1024/1024 and rboot (in fact esptool2) doesn't understand that layout. For 16 Mb it will select 16 Mb, 512/512 which is code 3, while I am using 1024/1024 which is code 5. This is a mismatch according to the SDK and it refuses to but. I patched up rboot and esptool2 and now it works.

stat_debug_1 = 1, so the function gets called
stat_debug_2 = 1, so the regist function returns OK
it does complain over missing boot and ota partitions though, on the UART but apparently it can live with it.

```
static const partition_item_t partition_items[] =
{
	{	SYSTEM_PARTITION_RF_CAL, 				RFCAL_OFFSET,				RFCAL_SIZE,				},
	{	SYSTEM_PARTITION_PHY_DATA,				PHYDATA_OFFSET,				PHYDATA_SIZE,			},	
	{	SYSTEM_PARTITION_SYSTEM_PARAMETER,		SYSTEM_CONFIG_OFFSET,		SYSTEM_CONFIG_SIZE,		},	
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 0,	USER_CONFIG_OFFSET,			USER_CONFIG_SIZE,		},
#if IMAGE_OTA == 0
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 1,	OFFSET_IRAM_PLAIN,			SIZE_IRAM_PLAIN,		},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 2,	OFFSET_IROM_PLAIN,			SIZE_IROM_PLAIN,		}
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 3,	SEQUENCER_FLASH_OFFSET,		SEQUENCER_FLASH_SIZE,	},
#else
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 1,	OFFSET_OTA_BOOT,			SIZE_OTA_BOOT,			},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 2,	OFFSET_OTA_RBOOT_CFG,		SIZE_OTA_RBOOT_CFG,		},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 3,	OFFSET_OTA_IMG_0,			SIZE_OTA_IMG,			},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 4,	OFFSET_OTA_IMG_1,			SIZE_OTA_IMG,			},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 5,	SEQUENCER_FLASH_OFFSET_0,	SEQUENCER_FLASH_SIZE,	},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 6,	SEQUENCER_FLASH_OFFSET_1,	SEQUENCER_FLASH_SIZE,	},
#endif
};

void user_pre_init(void);
iram void user_pre_init(void)
{
	stat_debug_1 = 1;
	stat_debug_2 = system_partition_table_regist(partition_items, sizeof(partition_items) / sizeof(*partition_items), FLASH_SIZE_16M_MAP_1024_1024);
}

```

someburner, 2018-09-06:
@eriksl  - that is good news, thanks for looking into it!

@xcguang 

> it does complain over missing boot and ota partitions though

It would be nice to know if this is indeed just a warning, or is it possible this will result in some sort of instability or crash?

xcguang, 2018-09-07:
Hi @someburner ,
It's just a warning. `SYSTEM_PARTITION_RF_CAL`,`SYSTEM_PARTITION_PHY_DATA` and `SYSTEM_PARTITION_SYSTEM_PARAMETER` is necessary. Others are optional, but you must ensure it works well.

xcguang, 2018-09-12:
How about the issue? Is it resolved?

eriksl, 2018-09-12:
Yes, there is no issue.

The original poster did not realise that rboot is a "normal" application as the SDK is concerned. Once rboot is started, it will take care of loading the further images. So everyone using rboot should NOT mark the image slots as "OTA" but as "USER" partitions and then it works.

E.g. like this

```
> flash partition table:
>   map: 16 Mb map 1024/1024
>     RF calibration    start: 0x0fb000 size: 01000
>     PHY config        start: 0x1fc000 size: 01000
>     SYSTEM config     start: 0x1fd000 size: 03000
>     USER config       start: 0x0fa000 size: 01000
>     RBOOT OTA boot    start: 0x000000 size: 01000
>     RBOOT OTA config  start: 0x001000 size: 01000
>     OTA image slot 0  start: 0x002000 size: f4000
>     OTA image slot 1  start: 0x102000 size: f4000
>     sequencer slot 0  start: 0x0f6000 size: 04000
>     sequencer slot 1  start: 0x1f6000 size: 04000
```

xcguang, 2018-09-21:
Hi All,
I will close this issue, if you have any question, please create a new issue, thanks.

closed 2018-09-21 by xcguang

scargill, 2018-11-11:
Hi

I use RBOOT.

I copied the partition info from the comment above - and the user pre-init from 3 comments up - and I put the lot into my user_main.c page. I changed from SDK 2.2 to DSK 3.1 

snd this lot won't compile. I tried adding integer definitions for stat_debug_1 and stat_debug_2 - still a boatl;oad of errors.

My code, with RBOOT works a TREAT with DSK 2.1 and previous... what am I missing?

Errors include "RFCAL_OFFSET undeclared here" and similar

and also

user/user_main.c:153:6: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'

I just did a copy and paste from above so I've not introduced errors.

Regards

Pete

scargill, 2018-11-11:
As to the solution proposed by ERIKDL - I can't even SAVE that, even with NOTEPAD++ on my PC - as strange characters get introduced into the file which I can't seem to remove...

eriksl, 2018-11-11:
I never said you could 1-to-1 copy and paste this code. For a start it contains references to debugging variables that are declared elsewhere, the stat* variables. Just remove them.

Furthermore, I think your editor expects the dreadful MS/Windows line ends, consisting of both line feed AND carriage return. This is not common practice in the open source world, where Unix style line ends are used -- just the line feed is enough.

As far as I know there are no non-ASCII characters, it would be weird to have them in a C program without user interface. So maybe github added them in their user interface. They're most likely UTF-8 and not a Windows codepage...

scargill, 2018-11-11:
Hi Erik

Right..

This with debug vars removed...

`static const partition_item_t partition_items[] =
{
	{	SYSTEM_PARTITION_RF_CAL, 				RFCAL_OFFSET,				RFCAL_SIZE,				},
	{	SYSTEM_PARTITION_PHY_DATA,				PHYDATA_OFFSET,				PHYDATA_SIZE,			},
	{	SYSTEM_PARTITION_SYSTEM_PARAMETER,		SYSTEM_CONFIG_OFFSET,		SYSTEM_CONFIG_SIZE,		},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 0,	USER_CONFIG_OFFSET,			USER_CONFIG_SIZE,		},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 1,	OFFSET_OTA_BOOT,			SIZE_OTA_BOOT,			},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 2,	OFFSET_OTA_RBOOT_CFG,		SIZE_OTA_RBOOT_CFG,		},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 3,	OFFSET_OTA_IMG_0,			SIZE_OTA_IMG,			},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 4,	OFFSET_OTA_IMG_1,			SIZE_OTA_IMG,			},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 5,	SEQUENCER_FLASH_OFFSET_0,	SEQUENCER_FLASH_SIZE,	},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 6,	SEQUENCER_FLASH_OFFSET_1,	SEQUENCER_FLASH_SIZE,	},
};

void user_pre_init(void);
iram void user_pre_init(void)
{
	system_partition_table_regist(partition_items, sizeof(partition_items) / sizeof(*partition_items), FLASH_SIZE_16M_MAP_1024_1024);
}
`
That fails as everything is undeclared.... then at the end..
`user/user_main.c:151:6: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'`

This one has samew issues after sorting the Linux Line endings (my fault)

`static const partition_item_t partition_items[] =
{
	{	SYSTEM_PARTITION_RF_CAL, 				RFCAL_OFFSET,				RFCAL_SIZE,				},
	{	SYSTEM_PARTITION_PHY_DATA,				PHYDATA_OFFSET,				PHYDATA_SIZE,			},	
	{	SYSTEM_PARTITION_SYSTEM_PARAMETER,		SYSTEM_CONFIG_OFFSET,		SYSTEM_CONFIG_SIZE,		},	
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 0,	USER_CONFIG_OFFSET,			USER_CONFIG_SIZE,		},
#if IMAGE_OTA == 0
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 1,	OFFSET_IRAM_PLAIN,			SIZE_IRAM_PLAIN,		},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 2,	OFFSET_IROM_PLAIN,			SIZE_IROM_PLAIN,		}
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 3,	SEQUENCER_FLASH_OFFSET,		SEQUENCER_FLASH_SIZE,	},
#else
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 1,	OFFSET_OTA_BOOT,			SIZE_OTA_BOOT,			},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 2,	OFFSET_OTA_RBOOT_CFG,		SIZE_OTA_RBOOT_CFG,		},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 3,	OFFSET_OTA_IMG_0,			SIZE_OTA_IMG,			},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 4,	OFFSET_OTA_IMG_1,			SIZE_OTA_IMG,			},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 5,	SEQUENCER_FLASH_OFFSET_0,	SEQUENCER_FLASH_SIZE,	},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 6,	SEQUENCER_FLASH_OFFSET_1,	SEQUENCER_FLASH_SIZE,	},
#endif
};
`
Here is the output

`CC user/user_main.c
user/user_main.c:140:33: error: 'RFCAL_OFFSET' undeclared here (not in a function)
  { SYSTEM_PARTITION_RF_CAL,     RFCAL_OFFSET,    RFCAL_SIZE,    },
                                 ^
user/user_main.c:140:50: error: 'RFCAL_SIZE' undeclared here (not in a function)
  { SYSTEM_PARTITION_RF_CAL,     RFCAL_OFFSET,    RFCAL_SIZE,    },
                                                  ^
user/user_main.c:141:34: error: 'PHYDATA_OFFSET' undeclared here (not in a function)
  { SYSTEM_PARTITION_PHY_DATA,    PHYDATA_OFFSET,    PHYDATA_SIZE,   },
                                  ^
user/user_main.c:141:53: error: 'PHYDATA_SIZE' undeclared here (not in a function)
  { SYSTEM_PARTITION_PHY_DATA,    PHYDATA_OFFSET,    PHYDATA_SIZE,   },
                                                     ^
user/user_main.c:142:40: error: 'SYSTEM_CONFIG_OFFSET' undeclared here (not in a function)
  { SYSTEM_PARTITION_SYSTEM_PARAMETER,  SYSTEM_CONFIG_OFFSET,  SYSTEM_CONFIG_SIZE,  },
                                        ^
user/user_main.c:142:63: error: 'SYSTEM_CONFIG_SIZE' undeclared here (not in a function)
[... 21 lines cut ...]
  { SYSTEM_PARTITION_CUSTOMER_BEGIN + 3, OFFSET_OTA_IMG_0,   SIZE_OTA_IMG,   },
                                         ^
user/user_main.c:146:61: error: 'SIZE_OTA_IMG' undeclared here (not in a function)
  { SYSTEM_PARTITION_CUSTOMER_BEGIN + 3, OFFSET_OTA_IMG_0,   SIZE_OTA_IMG,   },
                                                             ^
user/user_main.c:147:41: error: 'OFFSET_OTA_IMG_1' undeclared here (not in a function)
  { SYSTEM_PARTITION_CUSTOMER_BEGIN + 4, OFFSET_OTA_IMG_1,   SIZE_OTA_IMG,   },
                                         ^
user/user_main.c:148:41: error: 'SEQUENCER_FLASH_OFFSET_0' undeclared here (not in a function)
  { SYSTEM_PARTITION_CUSTOMER_BEGIN + 5, SEQUENCER_FLASH_OFFSET_0, SEQUENCER_FLASH_SIZE, },
                                         ^
user/user_main.c:148:67: error: 'SEQUENCER_FLASH_SIZE' undeclared here (not in a function)
  { SYSTEM_PARTITION_CUSTOMER_BEGIN + 5, SEQUENCER_FLASH_OFFSET_0, SEQUENCER_FLASH_SIZE, },
                                                                   ^
user/user_main.c:149:41: error: 'SEQUENCER_FLASH_OFFSET_1' undeclared here (not in a function)
  { SYSTEM_PARTITION_CUSTOMER_BEGIN + 6, SEQUENCER_FLASH_OFFSET_1, SEQUENCER_FLASH_SIZE, },
                                         ^
user/user_main.c:151:2: error: expected identifier or '(' before '/' token
 */
  ^
user/user_main.c:178:6: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'void'
 iram void user_pre_init(void)
      ^
mingw32-make.exe: *** [build/user/user_main.o] Error 1
`

scargill, 2018-11-11:
Above I am in user_main.c   and clearly it knows nothing about those definitions...

eriksl, 2018-11-11:
As said, this serves as an example, not as a ready-to-copy-and-paste piece of code.

Most of the partitions mentioned you will not have in your code. Also the second column consists of actual memory addresses. I have put them into an enum for readability, which is declared in some include. For your project you must find out what adresses it uses and fill them in.

The partitions from "SYSTEM_PARTITION_CUSTOMER_BEGIN + 0" on are probably not required or useful for your project. That's why they're called "customer" partitions. For the others you will have to find out what to use as actual address.

scargill, 2018-11-11:
Right - getting nearer..

So I need this..
`	{	SYSTEM_PARTITION_RF_CAL, 				RFCAL_OFFSET,				RFCAL_SIZE,				},
	{	SYSTEM_PARTITION_PHY_DATA,				PHYDATA_OFFSET,				PHYDATA_SIZE,			},
	{	SYSTEM_PARTITION_SYSTEM_PARAMETER,		SYSTEM_CONFIG_OFFSET,		SYSTEM_CONFIG_SIZE,		},
`

Having removed the wor irsm from the pre-init function, that error went away.

Errors now are:

`CC user/user_main.c
user/user_main.c:137:1: error: unknown type name 'partition_item_t'
 static const partition_item_t partition_items[] =
 ^
user/user_main.c:139:4: error: 'SYSTEM_PARTITION_RF_CAL' undeclared here (not in a function)
  { SYSTEM_PARTITION_RF_CAL,     RFCAL_OFFSET,    RFCAL_SIZE,    },
    ^
user/user_main.c:139:33: error: 'RFCAL_OFFSET' undeclared here (not in a function)
  { SYSTEM_PARTITION_RF_CAL,     RFCAL_OFFSET,    RFCAL_SIZE,    },
                                 ^
user/user_main.c:139:50: error: 'RFCAL_SIZE' undeclared here (not in a function)
  { SYSTEM_PARTITION_RF_CAL,     RFCAL_OFFSET,    RFCAL_SIZE,    },
                                                  ^
user/user_main.c:140:4: error: 'SYSTEM_PARTITION_PHY_DATA' undeclared here (not in a function)
  { SYSTEM_PARTITION_PHY_DATA,    PHYDATA_OFFSET,    PHYDATA_SIZE,   },
    ^
user/user_main.c:140:34: error: 'PHYDATA_OFFSET' undeclared here (not in a function)
  { SYSTEM_PARTITION_PHY_DATA,    PHYDATA_OFFSET,    PHYDATA_SIZE,   },
                                  ^
user/user_main.c:140:53: error: 'PHYDATA_SIZE' undeclared here (not in a function)
  { SYSTEM_PARTITION_PHY_DATA,    PHYDATA_OFFSET,    PHYDATA_SIZE,   },
                                                     ^
user/user_main.c:141:4: error: 'SYSTEM_PARTITION_SYSTEM_PARAMETER' undeclared here (not in a function)
  { SYSTEM_PARTITION_SYSTEM_PARAMETER,  SYSTEM_CONFIG_OFFSET,  SYSTEM_CONFIG_SIZE,  },
    ^
user/user_main.c:141:40: error: 'SYSTEM_CONFIG_OFFSET' undeclared here (not in a function)
  { SYSTEM_PARTITION_SYSTEM_PARAMETER,  SYSTEM_CONFIG_OFFSET,  SYSTEM_CONFIG_SIZE,  },
                                        ^
user/user_main.c:141:63: error: 'SYSTEM_CONFIG_SIZE' undeclared here (not in a function)
  { SYSTEM_PARTITION_SYSTEM_PARAMETER,  SYSTEM_CONFIG_OFFSET,  SYSTEM_CONFIG_SIZE,  },
                                                               ^
mingw32-make.exe: *** [build/user/user_main.o] Error 1`

In other words, just those 9 definitions. I'm afraid they mean nothing to me. Can you suggest any starting points?

kriegste, 2018-11-11:
Looks like you copied the struct but forgot the defines. You do not have to use defines. Just put in the addresses and sizes by hand (depending on your flash map and config).

eriksl, 2018-11-11:
That's what I said already.

Minus the #define plus enum. #define is evil.

scargill, 2018-11-11:
Sorry I didn't see the defines. I put them in, picked the various options in turn and all the error messages went away... however... back to square 1.

c:/espressif/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: build/app_0.out section `.text' will not fit in region `iram1_0_seg'

scargill, 2018-11-11:
The point being - on SDK 2.1 my code works perfectly and I have room to spare.. I thought it would be snsile to "upgrade" to the latest SDK as I always did until 2.2 told me not enough room just as here. I am using ICACHE_FLASH_ATTR on virually everything. I'll try those options thanks.

kriegste, 2018-11-11:
Well, like I already said elsewhere, the SDK is getting fatter and fatter. :P

eriksl, 2018-11-12:
I don't agree. The only thing that matters is the IRAM usage, that is hard limited to 32 kbyte. A lot of the space is taken up bij the SDK code BUT the exact amount has been varying between releases and currently it's quite not the most, as Espressif has taken of lot of effort reducing the SDK IRAM footprint. One of the efforts to achieve that was to create linkage units for every separate function instead of having a linkage unit per source file. That means that if you only need one function from a source file, the linker won't draw in all the other functions from the same source file. This has made a huge difference. BUT for that to work, you need to change your linker script, Espressif has shown how to do that. If you fail to do that, you will end up with LESS instead of MORE free IRAM space. I have said that before in this thread. You can hardly blame Espressif for that.

96 kbytes of DRAM is quite a huge pile, for embedded systems. If you use it wisely, it is not going to be a bottleneck. E.g. make sure it's not wasted by text strings that are needlessly copied to DRAM on startup.

And for code size, that's virtually unlimited, using a 4 Mbyte or even 16 Mbyte flash (the max).

If you feel you are too limited by these numbers (i.e. you don't have the skills to make sufficiently light code), maybe the more user-friendly ESP32 is more appropriate, or something like Raspberry etc.

For me the ESP8266 is still a huge step forward from the ATTiny's I used before, with 2 Kb flash memory and 512 bytes DRAM....

kriegste, 2018-11-12:
Maybe @scargill can post his Makefile and we can have a look at it.

scargill, 2018-11-12:
The ESP8266 is truly marvellous, no doubt,, but Espressif seem to make a lot of assumptions about people’s skill. I am VERY good with C coding, but not THAT familiar with Linux and the details of the compiler and Linker. I’ve used from SDK 15 to 2.1 and would LOVE to keep up to date but I have other duties which stop me spending all day researching Espressif info and don’t really understand where everything goes. I know RBOOT goes to 0, my code starts at 2000h, near the top of the 1 meg space is the user data. I tried that code yesterday and the errors seemed to go away but then – no room – just as before. I don’t know how to progress other than sticking with SDK 2.1 unless someone can help.

 

I’ll respond to Kriegste

 

 

Pete

scargill, 2018-11-12:
Here as requested is my MAKE file (runs in Windows with the onofficial 8266 dev kit and Eclipse. Mskefile as requested...

`#############################################################
#
# Root Level Makefile 
#
# (c) by CHERTS <sleuthhound@gmail.com>
#
# esp_mqtt_dev2
#
#############################################################

# *** if you change RBOOT_BIG_FLASH you must 'make clean' and build again
RBOOT_BIG_FLASH ?= 1

BUILD_BASE	= build
FW_BASE		= firmware

PYTHON  ?= C:\Python27\python.exe
ROM1MEG ?= C:\espressif\utils\blank_1MB.bin

UTILROMS ?= C:\Users\User\documents\esp-go\firmware
VERFILE  ?= C:\Users\User\documents\esp-go\include\user_config.h

#WEBTOOL ?= c:\Espressif\utils\web_page_convert.py
WEBTOOL ?= web_page_convert.py
FIND    ?= find
FLASH_OPTS ?= -fs 32m -ff 80m -fm qio
FLASH_DIO ?= -fs 32m -ff 80m -fm dio

# Base directory for the compiler
XTENSA_TOOLS_ROOT ?= c:/Espressif/xtensa-lx106-elf/bin

NEWSTUFF ?= c:/Espressif/esp8266_sdk/bin

# base directory of the ESP8266 SDK package, absolute
SDK_BASE	?= c:/Espressif/ESP8266_SDK
SDK_TOOLS	?= c:/Espressif/utils

#Extra Tensilica includes from the ESS VM
SDK_EXTRA_INCLUDES ?= c:/Espressif/ESP8266_SDK/include
SDK_EXTRA_LIBS ?= c:/Espressif/ESP8266_SDK/lib

# esptool path and port
ESPTOOL ?=c:/espressif/utils/esptool.exe
ESPPORT ?=COM8
ESPBAUD ?=230400
ESPTOOL2 ?= C:/esptools/esptool2.exe

#Position and maximum length of espfs in flash memory
ESPFS_POS  = 0x70000
ESPFS_SIZE = 0x08000

#Static gzipping is disabled by default.
GZIP_COMPRESSION ?= no

#If USE_HEATSHRINK is set to "yes" then the espfs files will be compressed with Heatshrink and decompressed
#on the fly while reading the file. Because the decompression is done in the esp8266, it does not require
#any support in the browser.
USE_HEATSHRINK ?= yes

# name for the target project
TARGET = app

# which modules (subdirectories) of the project to include in compiling
MODULES		= driver mqtt user softuart easygpio rboot/appcode

EXTRA_INCDIR	= include $(SDK_BASE)/../include \
		. \
		lib/heatshrink/ \
[... 221 lines cut ...]

ota_store: 
	$(PYTHON) $(SDK_TOOLS)/ftp_upload.py $(FW_BASE)/rboot.bin
	$(PYTHON) $(SDK_TOOLS)/ftp_upload.py $(FW_BASE)/rom.bin	
	
ota_store_webutu: 
	$(PYTHON) $(SDK_TOOLS)/ftp_upload_webutu.py $(FW_BASE)/rboot.bin
	$(PYTHON) $(SDK_TOOLS)/ftp_upload_webutu.py $(FW_BASE)/rom.bin
	grep SYSTEM_VER include/user_config.h | cut -d\  -f3 | tr -d '"' > $(FW_BASE)/version.html
	echo " " >> $(FW_BASE)/version.html
	grep SYSTEM_COMMENT include/user_config.h | cut -d\" -f2 >> $(FW_BASE)/version.html
	$(PYTHON) $(SDK_TOOLS)/ftp_upload_webutu.py $(FW_BASE)/version.html
	
ota_all: ota_store ota_store_webutu

clean:
	@echo "RM $(BUILD_BASE)"
	@rm -rf $(BUILD_BASE)
	@echo "RM $(FW_BASE)"
	@rm -rf $(FW_BASE)

$(foreach bdir,$(BUILD_DIR),$(eval $(call compile-objects,$(bdir))))

rebuild: $(info "***** REBUILDING ALL  *****") clean all
`

kriegste, 2018-11-12:
As a starting point, try adding -Wl,-gc-sections to your LDFLAGS.

scargill, 2018-11-12:
Apparently there is no such thing as -W1 (I already have -Wl (ie lower case L) and -gc and -gc-sections both come up as invalid.

`# linker flags used to generate the main object file
LDFLAGS = -nostdlib -Wl,--no-check-sections -u Cache_Read_Enable_New, -u call_user_start -Wl,-static -Wl,-Map,foo.map`

kriegste, 2018-11-12:
Try copy&paste, just append it (with a space before it; no space between , and -).

scargill, 2018-11-12:
Well, if nothing else, adding that in gives me more space in SDK 2.1 - from 7950 down to 7774. That's good.

scargill, 2018-11-12:
> Have you tried going back to the original boot loader?
> When I google rboot, I find two year old code with three year old linker files. And you seem to use non-standard linker files. Maybe this is the cause. Too much has changed since then.

Possible of course, but I’ve no idea what’s needed… Make files are not exactly my key skill.

 

Pete

eriksl, 2018-11-12:
Using RBOOT doesn't add any code to the IRAM section, unless you're using the BIGFLASH option, then it adds about 150 bytes, not worth mentioning.

Rboot hasn't change much over the last few years and that's because it just works. Just a few extra options have been added. For instance it can now fetch from RTC memory what slot needs to be booted, making it possible to implement a fail-safe boot. Very convenient when you're OTA flashing and you can't easily reach the device.

kriegste, 2018-11-12:
Then maybe the problems lie in his linker files (ld files)?

scargill, 2018-11-12:
What I don’t understand is, what can be so wrong with the files I use… They’ve worked reliably from SDK 1.odd to 2.1 without issue on many upgrades to my software and many different ESP boards.

 

Stills, happy to look at whatever needs doing to let me move to the latest SDK.

 

Pete

eriksl, 2018-11-12:
This, I quote myself from a few posts back:

> A lot of the space is taken up bij the SDK code BUT the exact amount has been varying between releases and currently it's quite not the most, as Espressif has taken of lot of effort reducing the SDK [...]

scargill, 2018-11-12:
I am using BigFlash. Never fully understood why but I am. I use 1 meg sections – the first and second megs – for the two OTA versions. I use the top end of meg 4, don’t use the third meg at all. RBOOT DOES work and I’d rather keep using it if possible.

 

In recent versions of my software I’ve reliably OTA’s many times.

 

Pete

scargill, 2018-11-12:
> In the RBOOT documentation you can see BIGFLASH is required to have just one image for all slots. If you're not using BIGFLASH, you must create separate images for every slots, with different loader scripts so they're running at different addresses.
>
> It isn't too difficult to crash the ESP8266 with a bug. If your code has a bug that crashes the ESP8266, you won't be able to flash it OTA, the ESP8266 will keep trying to boot and crash again. That's when you're glad you have the fallback scheme where the older slot is automatically booted whenever the newly slot can't complete booting.

As well as making source available I make BIN files available, just one RBOOT file (at 0), my code at 2000h and that’s it, that gets uploaded to either the first meg slot or the second – the same file.

 

If the OTA fails it a does go back to the previous version. And all of that works quite well.

 

Regards

 

Pete

scargill, 2018-11-12:
Though I don’t follow the details it does make sense that they would be linking in smaller chunks of object code rather than entire files?

scargill, 2018-11-12:
Kind of assumes a lot. Do you mean linker code in my Makefile or a separate file somewhere?

 

And that begs the question, WHERE have Espressif shown how to do this? I don’t think I know what a “linkage unit” is..

 

Pete

scargill, 2018-11-12:
I can see from above that the relevant file to grab is eagle.app.v6.ld - and I kno where that will need to go. Not entirely sure I follow all the changes that need to be made to the file, however.

This is the current eagle.app.6.ld that sits in the rboot directory. Do I tske it you're suggesting taking a clean one from the Espressif Github and make hte relevat mods whatever they are to THAT file instead?

kriegste, 2018-11-12:
Apparently only one line needs to be changed. The one beginning with "irom0_0_seg". It's the address of the OTA image. And this is the only difference in all the ld files anyway.
You need two copies of the ld file (one for each OTA image, with their respective addresses). Maybe just take these addresses from your old files... should not be too hard.

scargill, 2018-11-12:
I tried taking the Espressif eagle.app.v6.ld file and putting that into the ROOT directory, replacing the original - then editing the one line as R Burton suggests. That netted me a response:  eagle.rom.addr.v6.ld: no such file or directory. I've no idea why it says that.

kriegste, 2018-11-12:
In the last line, there is an include command. Maybe just correct the path and try again.

scargill, 2018-11-12:
Yes i do because Eichard also changed to a local version of that file. As you say also update the ROM file in RBOOT. Fixed that - and now I get...

`c:/espressif/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: build/app_0.out section `.text' will not fit in region `iram1_0_seg'`

scargill, 2018-11-12:
So near yet so far.

scargill, 2019-02-21:
Can someone help. I use Richard Burton's ROOT and almost all of my functions are in Flash. I've not been able to upgrade the OS since NONOS 2.1 due to IRAM not enough space problems. Code works perfectly in SDK 2.1 and earlier - I note today latest updates to SDK v 3.10 - but I've never had to mess with partitions and hence don't know how. My code (using RBOOT) is used with ESP-12 and I use RBOOT for OTA. Works a treat with 2.1 but I'm running short of RSM for more functions and would like to upgrade to latest SDK for future expansion.   Will the latest SDK work with RBOOT (at 0) and my code at 2000h... and can anyone advise on pre-init and partition table info needed, I compile on a PC using the unofficial CHERTS development kit and  edit in Eclipse.

fvpalha, 2019-02-21:
@scargill, I am searching an alternative to RBOOT.
At moment I am studing the project [esp_nano_httpd_fota](https://github.com/QB4-dev/esp_nano_httpd_fota).
My develop environment is Linux, with [esp-open-sdk](https://github.com/pfalcon/esp-open-sdk) upgrade the [NONOS SDK 2.2.x](https://github.com/espressif/ESP8266_NONOS_SDK/commit/34f1abd9c4196e02180fe41c42a4027d2433fbb2).

raburton, 2019-02-21:
@scargill you don't need to mess with the partition stuff: https://github.com/espressif/ESP8266_NONOS_SDK/issues/159#issuecomment-419338547

scargill, 2019-02-21:
Richard, so glad you are about. Others have said you need to understand the partitioning and some are not very understanding of those of us who don’t eat, sleep and dream about Espressif. I’m good with my code but know nothing about the finer workings of Espressif designer’s thinking. RBOOT just works so I don’t mess with it.

 

SO what steps/changes do I need to go through to move from SDK 2.1 use to current? All attempts up to now have led to being out of space for functions – iRAM won’t fit or similar. Won’t compile. It would make my day to be able to move on. I’m not out of space yet but quite close which stops me getting ambitious – even though I have loads of ram for normal use.

 

Pete

scargill, 2019-03-31:
I finally got RBOOT and my code running on SDK 3.1 - I hsve noticed sometimes while doing the OTA, what I assume are memory errors... E.M.xxxx - where xxxx is a number... and the OTA fails... no harm done and trying again usually gets it. I have no idea why I see these errors - I'm assuming some kind of timeout issue  - anyone any idea what they are for and how to stop that happening?

kriegste, 2019-03-31:
That is an out of memory (ram) error.

scargill, 2019-03-31:
I just checked - I have a heap of 17k before WIFI connects - then 7688 bytes pretty consistently in normal use... don't understand why 5800 for an OTA would be an issue and why only sometimes... ideas anyone?

[46 more replies]