espressif/ESP8266_NONOS_SDK #166: please update link for ESP8266_Non_OS_SDK_API_Reference__EN.pdf v3.0

MaBecker, 2018-08-27:
should be:

https://www.espressif.com/sites/default/files/documentation/2C-ESP8266_Non_OS_SDK_API_Reference__EN.pdf

FayeY, 2018-08-28:
Hi @MaBecker, Could you explain more about this issue? Which link should be replaced by the one you wrote? Thanks.

MaBecker, 2018-08-28:
In file Readme.md

MaBecker, 2018-08-30:
Hi,

there are two links pointing to the same pdf

 ESP8266 SDK Getting Started Guide

    https://www.espressif.com/sites/default/files/documentation/2a-esp8266-sdk_getting_started_guide_en.pdf
 

ESP8266 SDK 入门指南

    https://www.espressif.com/sites/default/files/documentation/2a-esp8266-sdk_getting_started_guide_cn.pdf

Asking to replace 

    https://www.espressif.com/sites/default/files/documentation/2a-esp8266-sdk_getting_started_guide_cn.pdf 

with 

    https://www.espressif.com/sites/default/files/documentation/2C-ESP8266_Non_OS_SDK_API_Reference__EN.pdf

for ESP8266 SDK 入门指南

MaBecker, 2018-08-30:
Sorry everything is fine; I mixed up things

closed 2018-08-30 by MaBecker