espressif/ESP8266_NONOS_SDK #168: ESP8266_AT_Bin_V1.7 Not Working on ESP-WROOM-02

Quaqmire, 2018-08-29:
I cannot get the AT V1.7 binaries to work on my ESP-WROOM-02 (4MB version).

I followed the notes in the README.md file for the 32Mbit-C1 and used the flash loader tool to download:
boot_v1.2.bin to 0x00000
user1.2048.new.5.bin  to 0x01000
esp_init_data_default_v05.bin to 0x3fc000
blank.bin to 0xfe000 & 0x3fe000

I have the download tool set for a 26MHz crystal, 40MHz SPI speed, QIO SPIO Mode, and a flash size of 32Mbit-C1.

I can erase the chip, and I can download the binaries with no issue.

However, when I reset the module, it will not communicate.   I get no reaction from the terminal.

After experimenting with different boot binaries (v1.6 & 1.7) and a different default init data binary (v08), I still could not get it to work.

On a hunch, I downloaded both the V1.5 and V1.62 binaries.

Using the same set up as noted above, but by replacing the links to the binary files of past versions, the module will load and work fine for both the V1.5 and V1.62.

It appears to me that there is an issue with the V1.7 binaries.

Quaqmire, 2018-08-29:
Additional info:
I also noted a discrepancy between the README.md file and Version 3.0 of the ESP8266 AT Instruction Set.

On page 3 of the ESP8266 AT Instruction Set, section 1.1.2 shows that the blank.bin should also be downloaded to 0x3fb000.    This is not shown in the README.md file.

I did try adding this, but it did not correct the issue.

Quaqmire, 2018-08-29:
Sorry...one more piece of information:  The Flash Download Tool is V3.6.4 if that makes any difference...

Quaqmire, 2018-08-29:
Some additional information:
When I flash the chip with the flash download tool, it lists the following in the DETECTED INFO window:
flash vendor:
A1h: FM
flash devID:
4016h
QUAD;32Mbit
crystal:
26Mhz

I just put the ESP8266 AT 1.6.2 Binaries into the module again using the 32MBit-C1 flash map, and it comes up working.

kriegste, 2018-08-29:
The user1.2048.new.5.bin you use is for flash map 5. The problem is (I guess) that the init data partition is no longer optional in the new SDK versions and it has to be at the correct address.

You can compile a new bin for flash map 6 or try to flash your module as if it only had 16MBit (I don't know if that works...).

Quaqmire, 2018-08-30:
@kriegste - Thanks again for taking the time to reply.

You seem very knowledgeable on this, and I appreciate your taking the time to answer my novice questions.

I tried flashing it as a 16Mbit-C1, and it did not work...I have read through the ESP8266 SDK Getting Started Guide, and I see what you mean about compiling a binary for flash map 6.  However, I have a few more questions if you don't mind:

1)  The ESP8266 AT Instructions (V3.0) show on page 3 to use the user1.2048.new.5.bin for both the 16Mbit-C1 and the 32Mbit-C1.   Is this just incorrect?

2)  I don't see anywhere that discusses the difference between the boot bin 1.2, 1.6, and 1.7 versions...can you point me to information on this?

3)  What is the difference between esp_init_data_default_v05. bin and esp_init_data_default_v08. bin?

I am thinking that it might just be easier to order the 2MB version of this module.   I really don't have a good time of compiling these binaries...I did it for the ESP32 AT, and it worked, but I had issues with the AT+CIPSEND commands not working in server mode, so I backed up and went the route of the ESP8266.

Any information you can provide is appreciated.

kriegste, 2018-08-30:
I know most of that only by trial and error. Here is some information ("changelog"):
https://github.com/espressif/ESP8266_NONOS_SDK/commits/master/bin
(expand by clicking "...")

I always use the latest versions for my projects, i.e. boot 1.7 and init no. 8 at the moment.

Quaqmire, 2018-08-30:
@kriegste - Thanks again!

Your suggestion to use the 16Mbit mapping and addresses ended up working...but I had to use the flash map from the AT Instruction Set pdf vs. the flash information in the README.md file included with the binaries.

If it helps anyone else, I used the 16Mbit flash map on page 3 of the ESP8266 AT Instruction Set V3.0 to flash the AT V1.7 onto an ESP-WROOM-02-4M  (4MB version) with the Flash Download Tool set on 16Mbit-C1 :

boot_v1.2.bin ->0x00000
user1.2048.new.5.bin ->0x01000
esp_init_data_default_v05.bin -> 0x1fc000
blank.bin -> 0xfe000
blank.bin ->0x1fe000
blank.bin ->0x1fb000

@Espressif - If anyone from Espressif is monitoring this, can you please put the binaries for flashing the 4MB chips in the binary zips?  Also, I think it would be helpful to make some corrections in the README files that are included with the binaries...and is the user1.2048.new.5.bin the correct file to call out on the 32Mbit-C1 flash map info on page 3 of the ESP8266 AT Instruction Set V3.0?

devanlai, 2018-09-01:
I've been trying to flash the version 1.7 bin onto an ESP-WROOM-02 module with 16Mbit flash, and I'm seeing the following error at boot:
```
mismatch map 5,spi_size_map 3
system_partition_table_regist fail
```

I'm using the following layout, uploaded via esptool:

| Address | Filename |
|----------|------------|
| 0x00000 | boot_v1.7.bin |
| 0x01000 | user1.2048.new.5.bin |
| 0xfe000 | blank.bin |
| 0x1fc000 | esp_init_data_default_v08.bin |
| 0x1fb000 | blank.bin |
| 0x1fe000 | blank.bin |

Quaqmire, 2018-09-01:
I think the problem maybe the esp_init_data_v08.bin...I used the v05 file, and it worked for me on the 4MB module.

xcguang, 2018-09-01:
@kriegste 
The flash map selected when you compiling must be the same as the flash map when flashing.

kriegste, 2018-09-01:
I know that. But it seems that might be @devanlai's problem.

xcguang, 2018-09-02:
I think you'd better specify flash size. Like
```
esptool.py --port /dev/tty.SLAB_USBtoUART --baud 921600 --chip esp8266 write_flash -z --flash_mode dio --flash_freq 40m --flash_size 2MB-c1 0x0 bin/boot_v1.6.bin 0x1000 bin/upgrade/user1.2048.new.5.bin 0xfc000 bin/esp_init_data_default.bin
```

devanlai, 2018-09-04:
@xcguang Specifying the flash size (`2MB-c1`) did the trick, thanks.

For the benefit of others, this is what my `esptool` parameters ended up looking like:

    python esptool.py --port COM8 --baud 1000000 --chip esp8266 write_flash --flash_size 2MB-c1 0x00000 boot_v1.7.bin 0x01000 user1.2048.new.5.bin 0xfe000 blank.bin 0x1fc000 esp_init_data_default_v08.bin 0x1fb000 blank.bin 0x1fe000 blank.bin

closed 2018-09-06 by FayeY

coisme, 2018-09-28:
@Quaqmire Thanks! This worked for me with following command.

```
esptool.py -p /dev/tty.usbserial \
--chip esp8266 \
write_flash \
--flash_size 2MB-c1 \
0x00000 ./bin/boot_v1.2.bin \
0x01000 ./bin/at/1024+1024/user1.2048.new.5.bin \
0x1fc000 ./bin/esp_init_data_default_v05.bin \
0xfe000 ./bin/blank.bin \
0x1fb000 ./bin/blank.bin \
0x1fe000 ./bin/blank.bin
```
> @kriegste - Thanks again!
>
> Your suggestion to use the 16Mbit mapping and addresses ended up working...but I had to use the flash map from the AT Instruction Set pdf vs. the flash information in the [...]

anudeepsekhar, 2018-09-29:
Hello guys i have followed every single instructions from this post but working and non working instructions but i still cant find a way to make my esp 01 work.. can someone help me out i have been on this for a week now.

The problem is after the download tool finish its download and i connect the esp-01 to putty i dont get any response its not even typing the at commands

kriegste, 2018-09-29:
ESP-01 has 1MB flash or even only .5MB (the very old ones). The new AT BIN files will not work on these. It might be possible to compile new ones using another flash layout and a few tricks, mentioned here, though:
https://github.com/espressif/ESP8266_NONOS_SDK/issues/162

hazee007, 2018-10-04:
@kriegste  can u give tell which AT BIN version will work with them... thanks

kriegste, 2018-10-04:
You can always try the previous version:
https://github.com/espressif/ESP8266_NONOS_SDK/tree/release/v2.2.x/bin/at/512%2B512

gustavosinbandera1, 2019-02-12:
here the best solution, work for me  https://unordered.org/timelines/5a0c6faccfc00000

joshkeys19, 2019-02-25:
Huge thanks! I was also able to flash my ESP-WROOM-02 by using the settings mentioned above. Specifically:

boot_v1.7.bin ->0x00000
user1.2048.new.5.bin ->0x01000
esp_init_data_default_v08.bin -> 0x1fc000
blank.bin -> 0xfe000
blank.bin ->0x1fe000
blank.bin ->0x1fb000

16Mbit-C1

[flashsettings](https://user-images.githubusercontent.com/19855744/53360503-0c48d000-390c-11e9-8460-841c5d7ec919.png)