espressif/ESP8266_NONOS_SDK #172: espconn_send data is const or not?

eriksl, 2018-09-14:
I cannot simply determine this from the espconn source in the SDK tree.

Currently espconn_send is prototyped as 

`sint8 espconn_send(struct espconn *espconn, uint8 *psent, uint16 length)`

Can this be changed to

`sint8 espconn_send(struct espconn *espconn, const uint8 *psent, uint16 length)`

in other words, the buffer supplied is it modified by espconn/lwip or not? I would expect not, but please explain. If the buffer is not modified, I can use the memory again, otherwise I'll have to malloc another buffer.

xcguang, 2018-09-21:
Hi,
espconn/lwip donot change the buffer.

eriksl, 2018-09-21:
Thanks.

Can I request then that the function prototype is changed to reflect that fact? Existing programs will not have issues when it's changed from non const to const.

xcguang, 2018-11-13:
There is no negative effect on your old code.

eriksl, 2018-11-13:
Yes there is, because it means I cannot make parameters in my own code "const" when they are passed to this function.

xcguang, 2018-11-14:
You can define the variable as `uint8 *`, for example `uint8 * data`, and call it like `espconn_send(espconn, (const uint8 *) data, length)`

eriksl, 2018-11-14:
Not that's not the issue. This is the issue.

void function_that_calls_espconn_send(**const** uint8_t *, unsigned int);

/* ... */

void function_that_calls_espconn_send(**const** uint8_t *data, unsigned int length)
{
    /* ... */
    espconn_send(espconn, data, length);
}

This will make the compiler abort compilation with an error, because a variable marked as const can never be made "unconst" (which would defy it's purpose).

This forces me to leave out the const qualifier for data in "function_that_calls_espconn_send" and all functions that in turn call this function, which has consequences for code clarity and for compilation efficienty. The compiler now can't trust the memory pointed to is not going to be changed by any of these functions.

eriksl, 2018-12-15:
Please change the header file.