espressif/ESP8266_NONOS_SDK #175: Sending UDP datagram through the STA interface when current mode is STA+SOFTAP

ThomArmax, 2018-09-25:
Hi,

I have implemented an UPnP client onto the ESP8266/85.
The first phase is the device discovery, which is made by sending a `MSEARCH ` message to the UPnP "_broadcast_" (`239.255.255.250:1900`).
If the ESP is in `STA` mode (`AT+CWMODE_CUR=1`) everything works well, the `MSEARCH` datagram is sent and I can see it by sniffing my network.
But if the ESP is in `STA+SOFTAP` mode, I look like the datagram is not sent (at least to the network on which the ESP station is connected to)

Here is my `MSEARCH` sending code : 
```
#define UDP_CONN_T espconn_t
[...]

void UPNP_FLASH_LOC
upnp_igd_send_msearch(void)
{
    IPADDR_T addr;
    webtool_create_ip(&addr, 239, 255, 255, 250);
    upnp_igd_discovery_state.broadcastConn = webtool_create_udp_connection(&addr, HTONS(SSDP_UDP_PORT));
    webtool_udp_send(upnp_igd_discovery_state.broadcastConn, &s_upnp_msearch_igd, sizeof(s_upnp_msearch_igd) - 1);
    upnp_igd_discovery_state.state = UPNP_IGD_STATE_MSEARCH_SENT;
}

[...]

void ICACHE_FLASH_ATTR
webtool_create_ip(IPADDR_T *addr, uint8_t addr1, uint8_t addr2, uint8_t addr3, uint8_t addr4)
{
    ip_addr_t a;
    IP4_ADDR(&a, addr1, addr2, addr3, addr4);
    *addr = a.addr;
}

[...]

UDP_CONN_T * ICACHE_FLASH_ATTR
webtool_create_udp_connection(IPADDR_T *addr, uint16_t port)
{
    UDP_CONN_T *udp_conn = (UDP_CONN_T*)os_malloc(sizeof(UDP_CONN_T));
    ets_memset(udp_conn, 0, sizeof(UDP_CONN_T));

    udp_conn->type = ESPCONN_UDP;
    udp_conn->proto.udp = (esp_udp *)os_zalloc(sizeof(esp_udp));
    udp_conn->proto.udp->local_port = HTONS(port);                  // Set an available port
    os_memcpy(udp_conn->proto.udp->remote_ip, addr, 4);             // ESP8266 udp remote IP
    udp_conn->proto.udp->remote_port = HTONS(port);                 // ESP8266 udp remote port
    //espconn_regist_recvcb(&udp_conn, upnp_client_udp_receive_cb); // register a udp packet receiving callback
    espconn_create(udp_conn);                                       // Create UDP connection

    return udp_conn;
}
```
Is there a way to specify the "interface" used to send the UDP datagram ?
Could it be a SDK issue/bug ?

Thanks by advance

kriegste, 2018-09-25:
Try
wifi_set_broadcast_if(wifi_get_opmode());

ThomArmax, 2018-09-25:
@kriegste Unfortunately using `wifi_set_broadcast_if(wifi_get_opmode())` does not solve the issue I have, maybe because the UPnP _broadcast_ address is not a standard broadcast address (`239.255.255.250`)

kriegste, 2018-09-25:
Maybe (completely untested):

struct ip_info info;
wifi_get_ip_info(STATION_IF, &info);
ip_addr_t mip;
IP4_ADDR(&mip, 239, 255, 255, 250);
espconn_igmp_join(&info.ip, &mip);

ThomArmax, 2018-09-25:
@kriegste I have quickly tested, I sounds good. Thank you so much :)