espressif/ESP8266_NONOS_SDK #178: Not enough IRAM

scargill, 2018-10-17:
I've just tried moving from SDK 2.1 to SDK 3.0

.text will not fit in region 'iram1_0_seg'

also

in function 'flash_data_check'
undefined reference  to user_pre_init'
(that is referred to twice in the console output).

HELP.

scargill, 2018-10-27:
That doen't help a lot. I've been using the SDK from very early on (pre 1.56) and moved some time ego to 2.0 then 2.1 but could not get beyond that due to not enough IRSM. I use Richard Burton's OTA at 0 and my code at 0x2000h, occupying around 700MB and hence need 4MB ESPs to do OTA. I moved the user area near the top of the 1 meg space, just below the Espressif info. This has been the basis oof my ESP-GO project for a long time now and don;'t want to change to Espressif OTA. I do not use Arduinoi code - just C with the SDK. I am tight for IRAM space despite putting all functions in FLASH. Moving to SDK 2.2 and up reports insufficient IRAM space and the MAKE file won't complete. What is needed to get enough space back from the SDK to use 3.0 ?

eriksl, 2018-10-27:
Did you read and implemented all of the required changes in your code to be using V3 at all? That would be the starting point.

Only after that, try to find out what functions end in IRAM and try to cut down on them. If you run the linker with a loadmap, you can spot which ones end up in IRAM and which ones in IROM.

V3 needs specific changes in the loader script, otherwise many functions that are not marked as "IRAM" end up there.

I did this. Also you need to use "-ffunction-sections -fdata-sections" as compiler options. And don't try to use -O2 or -O3, it will generate larger code but it's not faster. Use -Os and only that. Contrary to what the manual says, you can't use -O2 or -O3 together with -Os, the last used option will always win. So only use -Os.

```
diff --git a/loadscript-template b/loadscript-template
index 74f8a59..c91dbd0 100644
--- a/loadscript-template
+++ b/loadscript-template
@@ -151,6 +151,28 @@ SECTIONS
   } >dram0_0_seg :dram0_0_bss_phdr
 /* __stack = 0x3ffc8000; */
 
+  .irom0.text : ALIGN(4)
+  {
+    _irom0_text_start = ABSOLUTE(.);
+    *libmbedtls.a:(.literal.* .text.*)
+    *libat.a:(.literal.* .text.*)
+    *libcrypto.a:(.literal.* .text.*)
+    *libespnow.a:(.literal.* .text.*)
+    *libjson.a:(.literal.* .text.*)
+    *liblwip.a:(.literal.* .text.*)
+    *libmesh.a:(.literal.* .text.*)
+    *libnet80211.a:(.literal.* .text.*)
+    *libsmartconfig.a:(.literal.* .text.*)
+    *libssl.a:(.literal.* .text.*)
+    *libupgrade.a:(.literal.* .text.*)
+    *libwpa.a:(.literal.* .text.*)
+    *libwpa2.a:(.literal.* .text.*)
+    *libwps.a:(.literal.* .text.*)
+    *libm.a:(.literal .text .literal.* .text.*)
+    *(.irom0.literal .irom.literal .irom.text.literal .irom0.text .irom.text .flash.rodata)
+    _irom0_text_end = ABSOLUTE(.);
+  } >irom0_0_seg :irom0_0_phdr
+
   .text : ALIGN(4)
   {
     _stext = .;
@@ -198,14 +220,6 @@ SECTIONS
     *(.gnu.linkonce.lit4.*)
     _lit4_end = ABSOLUTE(.);
   } >iram1_0_seg :iram1_0_phdr
-
-  .irom0.text : ALIGN(4)
-  {
-    _irom0_text_start = ABSOLUTE(.);
-    *libmbedtls.a:(.literal .text .literal.* .text.*)
-    *(.irom0.literal .irom.literal .irom.text.literal .irom0.text .irom.text .flash.rodata)
-    _irom0_text_end = ABSOLUTE(.);
-  } >irom0_0_seg :irom0_0_phdr
 }
 
 /* get ROM code address */
```

kriegste, 2018-10-27:
And for the linker do not forget -Wl,-gc-sections

eriksl, 2018-10-27:
BTW Espressif made this change to actually get more free IRAM space, but if you don't follow the required steps, you'll end up with a lot less ;)

scargill, 2018-10-27:
I don't understand the MAKE part at all well (not a seasaoned Linux user - I use the unofficial Windows dev environment and have always just used the default options - so this is going to take time to get my head around this. Thanks.

eriksl, 2018-10-27:
Then I guess you'll have to rely on the one that maintains your environment to make it suitable for SDK V3.

scargill, 2018-10-27:
No-one else, just me. The project is open source on BitBucket.

xcguang, 2018-10-29:
Hi @scargill ,
you can try to add `user_iram_memory_is_enabled()` in your user_main.c like `https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/examples/at/user/user_main.c`

```
uint32 user_iram_memory_is_enabled(void)
{
    return 1;
}
```

eriksl, 2018-10-29:
That will very probably do exactly the opposite of what Scargill has issues with.

IIRC I already the exact opposite function of this, exactly for this sitation; also take part of IRAM and add it to the IRAM segment instead of flash cache.

On the other hand, it's always better if people learn to optimise for optimum IRAM cache usage. Even a large image like mine can do with the small amount of IRAM segment, one just has to be very careful.

scargill, 2018-10-29:
I manage just fine on SDK 2.1 – but attempts to move beyond that to 2.2 or 3.0 fail – it would help if I knew what I was doing with the makefile but that continues to escape me.

 

Pete

eriksl, 2018-10-29:
I don't know about V2.2, but for V3 you will definitely have to change the compilation/linking flags as described or you'll never have enough IRAM, not even if you're very careful.

Also for V3 you'll need to create a partition table.

All is described in updates to the SDK documentation.

scargill, 2018-10-29:
Thank you. The flags would be no problem but I have no idea how to make a partition table I’m afraid. I wonder if someone from Espressif can advise.

 

Pete

scargill, 2018-10-29:
> Espressif has made a special document in which it is decribed.

scargill, 2018-10-29:
Where? In the forum we are told to go to Github but the link is about RTOS_SDK – all very confusing.

 

Pete

eriksl, 2018-10-29:
https://github.com/espressif/ESP8266_NONOS_SDK/commit/5eb81b585ae2fdaedf67f8a717990e1543f99f68

https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/examples/peripheral_test/user/user_main.c

Both files can also be found in the ESP8266_NONOS_SDK repository, which you, I assume, have git cloned for building.

scargill, 2018-10-29:
I have only Os compilation flag and I have Wl already but not -gc section. No idea where to put that large table described above or where to put "-ffunction-sections -fdata-sections" 

This special document that Espresif put together - can anyone point me to it and does it include ALL the changes an SDK 2.1 user would need to use SDK 3.0 ?

eriksl, 2018-10-29:
Just click on the links? Two of them? Also the one that looks like a hex string.

scargill, 2018-10-29:
Thanks – that first link takes you to a file that is partly in Chinese… do I need to understand any of that file?

 

My user_main.c file looks NOTHING like that… I have rf_cal_sector_set function and 

 wifi_station_scan_done

 

Also in user_init function at the end I have 3 calls of my own – can I simply add these changes?

 

Regards

 

Pete

 

 <https://github.com/espressif/ESP8266_NONOS_SDK/commit/5eb81b585ae2fdaedf67f8a717990e1543f99f68> 5eb81b5

https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/examples/peripheral_test/user/user_main.c

eriksl, 2018-10-29:
The Chinese part is equal to the English part.

this one: rf_cal_sector_set() is no longer used.

You really need to read up on the new versions of the SDK documents, it's all in there!

hoekbrwr, 2018-10-30:
@eriksl 
I use the FW of Peter and occasionally compile and flash it from a clone locally.
I have been wading through a  lot of SDK docs and really can't find anything which does help me in changing from SDK 2.1 to SDK 3.0.
I do not understand the details of all commands, but the main structure I can follow.
So you present a part(?) of the loader file in one of your answers above. What should we do with this? What is the standard loader file, is it ld/eagle.app.v6? 
Maybe you can point out the changes we have to make to the SDK2.1 Makefile( 
[Makefile.zip](https://github.com/espressif/ESP8266_NONOS_SDK/files/2530635/Makefile.zip)
 ) please. 
You are also referring to quote Read the release notes for #3.0... unquote, but that points to another now closed forum item. Where can I find this doc?
Hopefully, you can direct us in the right direction to solve this matter.

eriksl, 2018-10-30:
Sorry I can't take you by the hand here, but you might some useful stuff in my project: https://github.com/eriksl/esp8266-universal-io-bridge

There are several commits there dedicated to transitions to newer SDK versions.

scargill, 2018-10-30:
Thank heavens its not just me.  I have real difficulty following makefiles etc, I’m from a Windows background and have adapted readily to programming C at the command line but the complexities of make files escape me. Any all in one boobs guide to moving from DSK 2.1 to 3.0 would REALLY be appreciated.. I can keep expanding my code (routines, I’m not short of normal RAM) for a while but would really like to move on.

 

Most of that partition stuff meant very little to me.

 

Pete

hoekbrwr, 2018-10-30:
Thanks anyway, we have to do it with that!

eriksl, 2018-10-30:
It's not the makefile itself that's complex, a properly made makefile is just dependencies and what to do to build the targets. Your problem is the C compiler and linker's command line + switches... Sorry no GUI there, just a manual page.

alxAgu, 2019-01-27:
@scargill, did you by any chance clone the repo and attempted to compile it? I was getting the same IRAM message until I realised I wasn't using the code from the official release:

https://github.com/espressif/ESP8266_NONOS_SDK/releases

A.

cnlohr, 2019-01-27:
Just making sure - this issue is fully resolved now, right?

cnlohr, 2019-01-27:
I am not sure if this is the right place to mention it but one of the tricks we use over at esp82xx is to specify many of the functions that already exist in IROM to be in IROM.  This frees up a _lot_ of space in IRAM for tons of algorithmic user stuff.

eriksl, 2019-01-27:
I guess you mean specify functions that are in IRAM in IROM. I do that frequently indeed, but far from all can be moved this way.

cnlohr, 2019-01-27:
Just checked.  Ooph.  Looks like they did get greedy.  The default esp82xx project with `STRIPPED_LIBGCC=YES` uses only 25644 bytes of IRAM... But if you upgrade to the brand newest NONOS_SDK, it's up to 27300 bytes.  Still a fair bit of space is free, and , but they are really on borrowed space.  But the operation of forcing the libgcc to use the mask ROM saves ~4kB.

eriksl, 2019-01-27:
See https://github.com/espressif/ESP8266_NONOS_SDK/issues/209

One of the last commit has eaten up > 2600 bytes of IRAM memory. I guess they forgot to add some "irom" section attributes or the compiler decided to inline some functions in the iram section.

I have traced back my submodule to before that commit, I hope they realise and fix this issue.

eriksl, 2019-01-27:
I did not know there is a version of libgcc in rom? But as long as you're using the version from the SDK, not the one that gcc itself provides, you're good, it's in irom.

cnlohr, 2019-01-27:
Hmm these look like some culprits:
```
40106314 000002f4 T wdev_hw_mac_reset
40106614 0000007a T wdev_print_mac
40106714 0000019b T wdev_dbg_reg
```

Too bad it's not possible to move symbols like those to different locations.

eriksl, 2019-01-27:
It is, but it's combersome and will "void warranty" ;-)

I guess they'll just have to fix it.

xcguang, 2019-02-21:
Please try it again using the latest master branch.

xcguang, 2019-03-14:
I will close this issue because no update for a long time, and you can reopen it if the problem persists.

closed 2019-03-14 by xcguang

eriksl, 2019-03-14:
I guess with NONOS SDK git HEAD we now have plenty of IRAM back :-)

cnlohr, 2019-03-15:
ARE YOU SERIOUS?

eriksl, 2019-03-15:
Yes I am serious. If you do not have loads of IRAM by now, you're definitely doing something wrong. See this and the other thread how maximise it.