espressif/ESP8266_NONOS_SDK #179: Flashing AT 1.7.0 binary firmware in 32m-c1 mode not working

c-w-e, 2018-10-17:
Hi dev team,

I tried flashing ESP8266 (32Mb) with latest AT firmware (1.7.0 binary distribution from espressif.com) using esptool.py in 32mb-c1 flash map mode using the flash adresses listet in readme.md / release notes in pdf.

Command used:
esptool.py --port /dev/ttyUSB0 --baud 115200 --chip esp8266 write_flash --flash_size 4MB-c1 0x00000 boot_v1.7.bin 0x01000 user1.2048.new.5.bin 0x3fc000 esp_init_data_default_v08.bin 0xfe000 blank.bin 0x3fe000 blank.bin

flash procedure runs through without an error but the ESP is not reacting for any serial communication afterwards using 
miniterm.py  --eol CRLF /dev/ttyUSB0 115200

Using the same procedure with AT v1.6.2 firmware (with adress information from related 1.6.2 version of readme.md in 32mb-c1 mode works just fine and communication with minitirm works as expected.

Any suggestions what's the issue? Is there an issue with 32mb-c1 mode in version 1.7.0?

Thanks and best regards,
c-w-e

VincentBreton, 2018-10-17:
I have the same trouble. 
**All is fine with AT v1.6.2 bin firmware from Espressif web site** but I met the same trouble as you get with **AT v1.7 bin firmware** from Espressif web site.
I Installed ESP8266 NONOS SDK V3.0.0 and I compiled "at" directory: I have the same trouble as I met with official AT v1.7 bin firmare from Espressif web site !

>  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
> 
> load 0x40100000, len 2408, room 16
> tail 8
> chksum 0xe5
> load 0x3ffe8000, len 776, room 0
> tail 8
> chksum 0x84
> load 0x3ffe8310, len 632, room 0
> tail 8
> chksum 0xd8
> csum 0xd8
> 
> 2nd boot version : 1.6
>   SPI Speed      : 40MHz
>   SPI Mode       : DOUT
>   SPI Flash Size & Map: 32Mbit(512KB+512KB)
> jump to run user2 @ 81000
> 
> mismatch map 5,spi_size_map 4
> system_partition_table_regist fail

Here my script to produce the error. It's work fine if I change the last four lines with AT v1.6.2 bin firmware :

#!/bin/bash
./esptool.py --port /dev/ttyUSB0 erase_flash
./esptool.py --port /dev/ttyUSB0 write_flash -fs 32m -fm dout 0x3FB000 ../../ESP8266_NONOS_SDK/bin/blank.bin

./esptool.py --port /dev/ttyUSB0 write_flash -fs 32m -fm dout 0x3FC000 ../../ESP8266_NONOS_SDK/bin/esp_init_data_default_v05.bin
./esptool.py --port /dev/ttyUSB0 write_flash -fs 32m -fm dout 0x7E000 ../../ESP8266_NONOS_SDK/bin/blank.bin
./esptool.py --port /dev/ttyUSB0 write_flash -fs 32m -fm dout 0x3FE000 ../../ESP8266_NONOS_SDK/bin/blank.bin
./esptool.py --port /dev/ttyUSB0 write_flash -fs 32m -fm dout 0x00000 ../../ESP8266_NONOS_SDK/bin/boot_v1.6.bin

./esptool.py --port /dev/ttyUSB0 write_flash -fs 32m -fm dout 0x01000 /home/vincent/ESP8266_AT_Bin_V1.7/bin/at/1024+1024/user1.2048.new.5.bin
./esptool.py --port /dev/ttyUSB0 write_flash -fs 32m -fm dout 0x81000 /home/vincent/ESP8266_AT_Bin_V1.7/bin/at/1024+1024/user2.2048.new.5.bin

I hope too an issue from Espressif. An appropriate simple script similar as below will be very convenient. A sufficient and simple procedure with essential steps to get an appropriate result with the  AT  Bin V1.7 seems to be necessary.

kriegste, 2018-10-17:
https://github.com/espressif/ESP8266_NONOS_SDK/issues/168#issuecomment-417360953

c-w-e, 2018-10-19:
Can confirm that using 16Mb-c1 memory map with with ESP8266 (32Mb) for the precompiled AT firmware version downloaded from espressif.com as a temp workaround (see comment from above)
https://github.com/espressif/ESP8266_NONOS_SDK/issues/179#issuecomment-430813613

But for my understanding this means a waste of 2MB of flash memory (correct me if I‘m wrong).

If 32Mb-c1 mode is not working/not supported ans more in AT firmware 1.7.0 I think this should be mentiomed in the release documentation on espressif.com and the readme.md in NONSDK/bin/at folder should get cleaned up.

Even better would be a fix from espressif with realesing new version if this is considdered to be a bug. :)

VincentBreton, 2018-10-19:
I'm using latest ESP8266_NONOS_SDK (Version 3.0.0). The bin file for AT commands in bin/1024+1024/ **user1.2048.new.5.bin** is exactly the same as the bin file provided by Espressif as the official binary 1.7 version .

You can't use it directly with an ESP8266 + 32Mbits but if I build it with  the SDK it generates another one in the ./bin/uprade directory with the new name user1.4096.new.6.bin. This final file is working fine with boot_v1.7 (and also with boot_v1.6 and probably others but I don't try them) and esp_init_data_default_v08.bin (and probably others versions but I don't try them).

**user1.2048.new.5.bin** is not working directly with an ESP8266 + 32Mbits so I infer that the ESP8266_NONOSK_SDK built an appropriate binarie file with it.

nimaltd, 2018-12-25:
> I'm using latest ESP8266_NONOS_SDK (Version 3.0.0). The bin file for AT commands in bin/1024+1024/ **user1.2048.new.5.bin** is exactly the same as the bin file provided by Espressif as the official [...]

Can you send your bin file please. And i need at_sdio version  . But i can compile it .

gsart100, 2019-02-07:
Had the same issue with ATMEGA2560 with integrated ESP8266 32M and confirmed below worked fine.

[image]

camp0s, 2019-03-11:
Same issue here, same memory addresses, SDK 3.0, same errors-

camp0s, 2019-03-31:
> 
> 
> Had the same issue with ATMEGA2560 with integrated ESP8266 32M and confirmed below worked fine.
> 
> [image]

Been back on the topic, tried th settings above and they worked fine! Tnx!

sadeqebrahimi, 2020-07-21:
> Had the same issue with ATMEGA2560 with integrated ESP8266 32M and confirmed below worked fine.
> 
> [image]

same issue. 
please share the solution if you solve it

leonty, 2020-08-03:
Same problem with D1 Mini 4Mb.

dfxhub, 2020-09-21:
Successfully flashed no-name esp-12e with 1.7.4:

esptool.py.exe -c esp8266 -p COM17 -b 115200 write_flash -ff 40m -fm qio -fs 4MB-c1 0x00000 boot_v1.7.bin 0x01000 at/1024+1024/user1.2048.new.5.bin 0x3fc000 esp_init_data_default_v08.bin 0xfe000 blank.bin 0x3fe000 blank.bin 0x3fb000 blank.bin

AT+GMR
AT version:1.7.4.0(May 11 2020 19:13:04)
SDK version:3.0.4(9532ceb)
compile time:May 27 2020 10:12:22
Bin version(Wroom 02):1.7.4
OK

Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
Manufacturer: 5e
Device: 4016
Detected flash size: 4MB

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : QIO
SPI Flash Size & Map: 32Mbit(1024KB+1024KB)

All addresses from page #3 of the manual - https://www.espressif.com/sites/default/files/documentation/4a-esp8266_at_instruction_set_en.pdf

xcguang, 2020-09-21:
Why not RTOS  AT?  https://docs.espressif.com/projects/esp-at/en/latest/

dfxhub, 2020-09-21:
> Why not RTOS AT?
> https://docs.espressif.com/projects/esp-at/en/latest/
> https://docs.espressif.com/projects/esp-at/en/latest/AT_Binary_Lists/ESP8266_AT_binaries.html
> https://github.com/espressif/esp-at

Sorry, I'm novice at ESP's world. )) Is that firmware better than "traditional" 1.x?

xcguang, 2020-09-21:
No feature will be added on v1.x. https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/README.md

Now we are put more focus on RTOS AT, and will add more features, eg,  IPv6, WPA3. And the commands is compatible with ESP32/ESP32S2 and later chips.