espressif/ESP8266_NONOS_SDK #180: error magic !

VincentBreton, 2018-10-18:
What does it means and how to solve that ?

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0
tail 8
chksum 0xd8
csum 0xd8

2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : DOUT
  SPI Flash Size & Map: 32Mbit(512KB+512KB)
jump to run user2 @ 81000

error magic!
backup boot failed.

user code done

```

ustccw, 2018-10-18:
@VincentBreton maybe OTA download a error bin which not beginning with `0xE9` or `0xEA,0x04`.

VincentBreton, 2018-10-18:
Thank you @ustccw but I verified my bin code with vi editor and my file starts with 0xEA.
```
vi my_file.bin
:%!xxd to display its content in hexa
00000000:  ea04 0002 0400 1040 0000 0000 c0a0 0500

```

VincentBreton, 2018-10-18:
I don't know what exactly "error magic!" message means. I received another error message after a new flash of the memory about size and map incompatibility but I solved the trouble:

1. I  entirely erased the Flash memory. In other words  I filled it with the value 0xFF.
2. I used the 1024KB+1024KB structure instead of 512KB+512KB structure with 32m-c1 value for flash size.  32m value is not for 1024KB+1024KB structure.  
3. I flashed esp_init_data_default at 0x3FC000 address; the boot sector at 0x00000 and the user1.4096.new.6.bin at 0x01000
4. I didn't flash user2.4096.new.6 but I think it would work too with it.

> AT+GMR
> 
> AT version:1.7.0.0(Aug 16 2018 00:57:04)
> SDK version:3.0.0(e27cf60)
> compile time:Oct 18 2018 23:44:15
> OK

closed 2018-10-18 by VincentBreton