espressif/ESP8266_NONOS_SDK #189: [SDK 2.0.0][AT 1.3.0.0] DHCP client release

ThomArmax, 2018-11-13:
Hi,

In my project, I need to be able to release the DHCP lease obtained by my ESP. I can't find any function in the SDK, nor in the AT commands set.

Does `wifi_station_dhcpc_stop()`or `AT+CWDHCP_CUR=2,0` will release the DHCP lease ?
If not, is there any nice way to do it ? I was thinking about implementing a custom AT command and send the appropriate datagram, but I not sure we can retrieve all the required informations like DHCP server ID (IP) & cie.

Do you have any clue ?

Thanks by advance.

xcguang, 2018-11-13:
You can try ```dhcp_release(eagle_lwip_getif(0x00)) ```

ThomArmax, 2018-11-13:
@xcguang Thanks, it works fine ! I would be nice to have a documentation about those "hidden" functions :)

closed 2018-11-13 by ThomArmax