espressif/ESP8266_NONOS_SDK #191: esp8266 sdk 2.2.1: system_deep_sleep causing reset

MopeZe, 2018-11-14:
Im using SDK 2.2.1

When running the following code:

```
system_deep_sleep_set_option(4);
system_deep_sleep(0);
```

ESP8266 immediately resets and doesnt enter deep sleep mode. 

Now what i really dont understand is if i run 

```
system_deep_sleep_set_option(4);
system_deep_sleep_instant(0);
```

everything works fine.

With SDK 2.0.0 system_deep_sleep() worked.

Can someone help? It is critical for us. 

Thanks and regards

MopeZe, 2018-11-23:
Update: The issue happens if the ESP8266 is connected to a station. 

Can you comfirm this behaviour?

xcguang, 2019-03-14:
@MopeZe ,
I  did test on the latest release/v2.2.x branch, it seems everything goes well. Could you try it again?
The following is my test code:
ESP8266 as SoftAP and one station connected ESP8266.
[image]

ESP8266 as Station and connected an AP
[image]

xcguang, 2019-03-18:
Hi,
I will close this issue, and you can reopen it if the problem persists.

closed 2019-03-18 by xcguang