espressif/ESP8266_NONOS_SDK #195: function definitions missing from ets_sys.h

davydnorris, 2018-11-20:
The following function definitions are missing from ets_sys.h, which causes some of the macro definitions to fail when compiling:

```
typedef void (* ets_isr_t)(void *);

void ets_intr_lock(void);
void ets_intr_unlock(void);
void ets_isr_attach(int i, ets_isr_t func, void *arg);
//the following are missing
void ets_isr_mask(unsigned intr);
void ets_isr_unmask(unsigned intr);
void ets_wdt_enable(void);  
void ets_wdt_disable(void);
```

davydnorris, 2018-11-20:
See pull request #135 - this issue adds a couple more functions to the list

xcguang, 2018-12-27:
Hi @davydnorris ,
`ets_isr_mask` and `ets_isr_unmask` should be added.
And `ets_wdt_xxx` is used for internal, we have `system_soft_wdt_xxx` apis for user, does they satisfy you?

xcguang, 2018-12-27:
Have added `ets_isr_mask` and `ets_isr_unmask` on branch release/v3.0.0 and branch release/v2.2.x

closed 2018-12-28 by davydnorris