espressif/ESP8266_NONOS_SDK #196: SYSTEM_PARTITION_OTA_SIZE in examples is not consistent

davydnorris, 2018-11-20:
I am trying to set up my partitions and noticed that the SYSTEM_PARTITION_OTA_SIZE in the examples is not consistent with the actual max size of the OTA partitions, which is confusing. 

In all the non-AT examples, the SYSTEM_PARTITION_OTA_SIZE is set to 0x6A000 regardless of how much space there actually is. However, in the AT examples, the SYSTEM_PARTITION_OTA_SIZE is set to 0xE0000.

Should these be set to different sizes depending on whether you have 512k or 1024k partitions, or should they be set to the actual size of the OTA code? I assumed that these should be set to the maximum possible size of a user1 or user2 partition, which should be either 488k (512-24 = 0x7A000) for a 512k partition, or 1000k (1024-24 = 0xFA000) for a 1024k partition, so:

```
#elif (SPI_FLASH_SIZE_MAP == 2)
#define SYSTEM_PARTITION_OTA_SIZE							0x7A000
.
.
.
#elif (SPI_FLASH_SIZE_MAP == 3)
#define SYSTEM_PARTITION_OTA_SIZE							0x7A000
.
.
.
#elif (SPI_FLASH_SIZE_MAP == 4)
#define SYSTEM_PARTITION_OTA_SIZE							0xFA000
.
.
.
#elif (SPI_FLASH_SIZE_MAP == 5)
#define SYSTEM_PARTITION_OTA_SIZE							0x7A000
.
.
.
#elif (SPI_FLASH_SIZE_MAP == 6)
#define SYSTEM_PARTITION_OTA_SIZE							0xFA000
.
.
.
```

kriegste, 2018-11-20:
As I understand it, this should be set to the maximum (allowed by your flash map). Otherwise you might not be able to update to a larger binary later on.

davydnorris, 2018-11-20:
That's exactly what I thought too - but none of the examples are set up that way. They have the same size set regardless of whether you choose a 512k or 1024k flash map. I think this is a 'bug' in the examples, or at least a bad example to follow.

closed 2019-03-14 by davydnorris