espressif/ESP8266_NONOS_SDK #198: espconn_gethostbyname returns ERR_MEM although there is more than enough free memory

kriegste, 2018-12-01:
How do I debug espconn_gethostbyname? I call it to retrieve the IP of "api.openweathermap.org". Sometimes (for a few hours in a row) it returns ERR_MEM although there should be more than enough memory available. I print the amount of free heap in the log and it is always around 49264 bytes more or less.

Has anyone experiences a similar behavior?

I suspect there is a fixed buffer inside the espconn_gethostbyname function which is too small for some requests.

kriegste, 2018-12-06:
Works.

someburner, 2018-12-06:
@kriegste - shouldn't you leave this open until it makes it into the SDK?

kriegste, 2018-12-08:
No problem. Don't let me stop you (or Espressif).

Looking at the new lwIP there are more bugs that need to be fixed.

gilpinheiro, 2019-03-04:
It appears the SDK is compiled with DNS_TABLE_SIZE 0, which makes the issue moot. I base this on the latest SDK liblwip.a not containing a dns_table at all.

It'd be nice if the lwipopt.h actually reflected what options the SDK is using.

kriegste, 2019-03-04:
The problem did not reoccur. All fine after the fix.

FayeY, 2019-03-13:
Is this issue solved? Could we close this issue now?

kriegste, 2019-03-13:
No, your code still has the bug.
https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/third_party/lwip/core/dns.c#L886

xcguang, 2019-03-14:
15993bf5181e04ed9f4f75e32dad4ea238f7147e & 89693b42639acd15611aea8767271f5c5d2bbb8a & 74358b75382c175cdfca194ad9fe8bec2f0634b5 fix it, please try it again.

closed 2019-03-14 by kriegste