espressif/ESP8266_NONOS_SDK #199: how to deal with REASON_AUTH_EXPIRE?

RockfordWei, 2018-12-02:
Hi, this is my code to set up a wifi connection from an esp8266 to a router:

``` c
	struct station_config * sta_conf = (struct station_config *)os_zalloc(sizeof(struct station_config));
	wifi_station_disconnect();
	system_restore();
	wifi_station_set_reconnect_policy(false);
	wifi_set_opmode_current(STATION_MODE);
	wifi_station_get_config(sta_conf);
	os_memset(sta_conf->ssid, 0, 32);
	os_memset(sta_conf->password, 0, 64);
	os_strcpy(sta_conf->ssid, "somessid");
	os_strcpy(sta_conf->password, "somepassword");
	sta_conf->bssid_set = 0;		//reset bssid_set
	wifi_station_set_config(sta_conf);
	wifi_station_set_config_current(sta_conf);
	wifi_station_connect();
```

It works well on regular WIFI routers, however, it always encounters `REASON_AUTH_EXPIRE` if there is a wifi mesh. What does it really mean and are there any document available to explain [all these reasons](https://github.com/espressif/ESP8266_NONOS_SDK/blob/3bcc1467723a0e8283322560ea86a8a4573725f7/include/user_interface.h#L469) and how to deal with it in detail?

xcguang, 2018-12-27:
Hi,
Could you set bssid when connecting router?

RockfordWei, 2018-12-27:
I am afraid my end users would not have a chance to know bssid... I could have a try somehow.

xcguang, 2019-03-13:
Hi, 
The issue is resolved? I will closed it, you can reopen it, if you have any question.

closed 2019-03-13 by xcguang