espressif/ESP8266_NONOS_SDK #201: Using UARTs for arbitrary byte output

eriksl, 2018-12-15:
Is it possible to disable the output of all of
- start bit
- stop bits
- parity bits
from the UARTs?

That would be very interesting, it would enable us to have a semi-automatic, buffered, parallel-to-serial converter at high speed.

My main goal would to be use it for a semi-software-semi-hardware PWM implementation (we now have to do most in software). But I could think of other uses as well.

I can now disable parity, set stop bits to "1" and start bits to "1" but not to "0".

Maybe otherwise the I2S interface can do something similar?

FayeY, 2018-12-27:
@xiongyumail , could you help with this issue?

eriksl, 2019-03-02:
Apparently not?