espressif/ESP8266_NONOS_SDK #202: AT problem

nimaltd, 2018-12-25:
I program like https://github.com/espressif/ESP8266_NONOS_SDK/tree/master/bin/at 
but don't answer  any at commands.
but when  i program ai thinker at version . its work.
i use esp-201 module with 32Mbit flash

xcguang, 2018-12-27:
Hi,
Can you provide any log?

nimaltd, 2018-12-27:
Hi.  Doesn't back any answer. 
I replace user1 with old version. Ad work fine

nimaltd, 2018-12-27:
But i program with this methode   its work fine
Sam:
esptool.py -p COM2 --chip esp8266 write_flash -fm dio -ff 26m --flash_size 2MB-c1 0x00000 ./bin/boot_v1.7.bin 0x01000 ./bin/at/1024+1024/user1.2048.new.5.bin 0x1fc000 ./bin/esp_init_data_default_v08.bin 0xfe000 ./bin/blank.bin 0x1fe000 ./bin/blank.bin 0x1fb000 ./bin/blank.bin

xcguang, 2018-12-27:
I guess I know the result about the issue. Since SDK3.0, we added check the flash_map, if the flash_map mismatch bin, it will fail. And we are considering removing this check.

nimaltd, 2018-12-27:
I want try at_sdio.
But i can not find any sample by stm32

xcguang, 2018-12-27:
https://github.com/espressif/ESP8266_NONOS_SDK/issues/203
this is a demo, maybe it is helpful.

xcguang, 2018-12-27:
Hi @nimaltd ,
I will close the issue, if you have any question, please reopen it or create a new issue. Thanks.

closed 2018-12-27 by xcguang