espressif/ESP8266_NONOS_SDK #204: [Documentation Request] need some clarification about system_rtc_clock_cali_proc() API for RTC time

IMAN4K, 2019-01-01:
Hi all,
As discussed in  [#5485](https://github.com/esp8266/Arduino/pull/5485), we are suppose to add RTC value read API to Arduino core, but first, we need to know what exact value stored in `system_rtc_clock_cali_proc()`?
for example, could you please tell us more about the rtc example in `ESP8266 Non-OS SDK API Reference, Appendix A.2`

xcguang, 2019-01-02:
Hi,
Suppose `cal1` is binary data `110 100111011000`, integral part is `0b110` and fractional part is `0b100111011000`, the real value is `2^2 + 2^1  + 2^-1 +2^-4 + 2^-5 + 2^-6 + 2^-8 + 2^-9` = `6.615234375`

`os_printf("cal1:%d.%d\r\n",((cal1*1000)>>12)/1000,((cal1*1000)>>12)%1000)`  is a rough calculation. `(cal1*1000)>>12` = `(0b11010011101100*1000)>>12` = `6615`, so `((cal1*1000)>>12)/1000` = `6615/1000` = `6`; `((cal1*1000)>>12)%1000)` = `6615%1000` = `615`, that is `os_printf("cal1:%d.%d\r\n",((cal1*1000)>>12)/1000,((cal1*1000)>>12)%1000)` outputs `6.615`.

xcguang, 2019-01-02:
Maybe `(cal1 * 1.0)/(2^12)` = `(0b100111011000 * 1.0)/(2^12)` = `6.615234375` is easiest to understand.

xcguang, 2019-01-02:
@devyte I think you are right. it seems the output is incorrect sometimes.
@FayeY Please check it.

davydnorris, 2019-01-02:
@devyte @xcguang - yes you would need to change the above to 

`os_printf("cal1:%0d.%03d\r\n", ((cal1*1000)>>12)/1000, ((cal1*1000)>>12) % 1000);`

where the digit in the format string %0_d matches the divisor - eg. for 6 decimal places you would divide by 1000000 and use %06d

devyte, 2019-01-04:
Thank you for the confirmation.
@xcguang I think that a fix to the document with the previous comment would address this.

IMAN4K, 2019-01-05:
@xcguang Thanks for the explanation!
It would be great if you add that explanation to `Non-OS SDK API` Docs  just for future reference

Regards.

closed 2019-01-05 by IMAN4K