espressif/ESP8266_NONOS_SDK #209: commit 0f81568ae9feee738c602b263e54b2d5000cda09 wastes 1600 bytes of IRAM

eriksl, 2019-01-26:
commit 0f81568ae9feee738c602b263e54b2d5000cda09
Merge: 5b293d5 eee7a81
Author: Xu Chun Guang <xuchunguang@espressif.com>
Date:   4 weeks ago

    Merge branch 'bugfix/disable_ampdu_packet' into 'master'
    
    fix(pp): Disable receiving ampdu packet
    
    See merge request sdk/ESP8266_NONOS_SDK!112

eriksl, 2019-01-26:
Also what does this mean:

> See merge request sdk/ESP8266_NONOS_SDK!112

Where can I find the merge request and/or the issue?

gilpinheiro, 2019-02-12:
There does seem to be something wrong here. Whether it is due to IRAM usage or not, I can't say, but it looks to memory management related. I get consistent crashes when I invoke espconn_secure_connect() (I've bisected the sdk, and this commit is the culprit).

If it is using the memory up, it is also worth noting that the crash occurs due to bug (null dereference?) inside the memory allocation/free routines. Without debug information it hard to say where exactly, but it seems to result in a crash (LoadProhibitedCause) in a unnamed function that is referenced by both pvPortMalloc and vPortFree.

Fatal exception 28(LoadProhibitedCause):
epc1=0x401010be, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

@xcguang

xcguang, 2019-02-12:
@gilpinheiro Could you provide your demo?

xcguang, 2019-02-12:
@eriksl,
The merge request is in our internal project, if it fix an issue or add a feature, we will describe it in commit information.

```
Merge branch 'bugfix/disable_ampdu_packet' into 'master'

fix(pp): Disable receiving ampdu packet

See merge request sdk/ESP8266_NONOS_SDK!112
```
Just disable receiving ampdu packet, because esp8266 cannot process ampdu packet.
Because we add some APIs in iram, so it will waste some iram.

eriksl, 2019-02-12:
"Some" IRAM? 1600 ****** bytes. That is about 1/4 of what's left after SDK has taken most of it.

IRAM is not something you can choose to use or not. Some of my functions MUST be in IRAM due to being interrupt handlers or similar. After this commit there is no way I can get my project working.

In the past, quite a bit of work has been done on Espressive's side to minimise IRAM use by the SDK code. I appreciate that. But now all that effort has been in vain, the memory space is gone again.

Then at least supply a way of choosing whether to include these functions, so those not using it, don't have their IRAM wasted. As source is closed.

eriksl, 2019-02-12:
To put this into perspective, for those interested: from 32 K usable IRAM, 27328 bytes are used by the SDK, leaving only **5440** bytes for your application. After this commit, that's only **3840** bytes. And that's with maximum IRAM hungry stuff disabled.

cnlohr, 2019-02-13:
I want to second the concern over IRAM usage.  It is intensely important for some products which use the ESP8266 for real-time tasks.

gilpinheiro, 2019-02-18:
@xcguang 

I think I know what is happening now: 

I don't think this particular commit contains the issue that I'm hitting - I think that because it uses more IRAM it triggers a pre-existing bug with the memory allocation system when the iram_memory_is_enabled function is true.

I can reliably reproduce a crash with the latest code by simply by enabling the use of excess iram as heap via:

```user_iram_memory_is_enabled() { return true; }"```

and attempting to start a secure ssl connection with a custom certificate i.e. using ```espconn_secure_ca_enabled(1, sector_id)``` and ```espconn_secure_cert_req_enable(1, sector_id)```.

The exact code works fine with ```user_iram_is_enabled() { return false; }```, I seem to get bad behaviour when heap fills the iram and the system starts allocating from dram.

The system should be falling back to using dram, when the allocation in iram fails, but something is going wrong. I've opened a second issue with the details of what I've found so far: https://github.com/espressif/ESP8266_NONOS_SDK/issues/211

davydnorris, 2019-02-20:
I will add my support for fixing this urgently - we need as much IRAM available as possible. This small fix should not have changed the amount by 1.6k. Something else must have been moved

xcguang, 2019-02-21:
Yes, some unneeded APIs are retained. `-ffunction-sections` doesn't seem to work for xt-xcc, I will remove these APIs later.

davydnorris, 2019-02-21:
@xcguang thank you!!

xcguang, 2019-02-21:
Hi all,
Please try it again using master branch.

eriksl, 2019-02-21:
Great work Espressif!

From 1600 bytes lost, I now gained 700 bytes, which will come in very useful!

Thanks!

davydnorris, 2019-02-21:
That's brilliant!!

someburner, 2019-02-21:
@xcguang - I'm assuming the release/v2.2.x branch could also benefit from these changes? It also has the "Disable receiving ampdu packet" commit referenced by OP

xcguang, 2019-02-22:
@someburner release/v2.2.x and release/v3.0.0 will be updated later.

someburner, 2019-02-22:
Great, thank you!

eriksl, 2019-02-22:
BTW I've found that the updated compiler optimisation flags are not the best related to code size (at least not for my code). I am now experimenting to pick the best (and still yield valid code). I will share my findinds when ready. In the meantime it looks like using the lto (link time optimisation) does a really good job and save another 400 bytes. I've not tested yet if it produces good code though ;-)

eriksl, 2019-02-23:
I've gained another 1300 bytes (!) with these optimisation options, use for compiling and linking stage and use "gcc" for linking (not "ld" etc.)
```
CFLAGS			:=	-Os -std=gnu11 -mlongcalls -mno-serialize-volatile \
						-fno-unroll-loops -fno-move-loop-invariants -freorder-blocks-and-partition \
						-fno-keep-static-consts -fuse-linker-plugin -fno-math-errno \
						-fno-tree-copy-prop -fno-tree-ccp -fno-tree-tail-merge \
						-flto=8 -flto-partition=none -flto-compression-level=0 \
						-ffunction-sections -fdata-sections
```

Then, add this #define for functions that do need full speed optimisation:
```
#define attr_speed __attribute__ ((optimize("O3", "unroll-loops","move-loop-invariants","tree-copy-prop","tree-ccp","tree-tail-merge")))
```

and prepend such function with "attr_speed".

eriksl, 2019-02-23:
@xcguang is it possible that Espressif releases lto enabled (-flto) versions of the libraries as well? That will gain probably even more precious IRAM space.

kriegste, 2019-02-23:
But this is flash size, not RAM.

In my project I use
```
CFLAGS += -Os
CFLAGS += -g
CFLAGS += -fpeel-loops
CFLAGS += -fno-cse-follow-jumps
CFLAGS += -fno-tree-fre
CFLAGS += -fno-ipa-sra
CFLAGS += -fno-signed-zeros -fno-trapping-math -fassociative-math
CFLAGS += -fno-caller-saves
CFLAGS += -fno-move-loop-invariants
CFLAGS += -freorder-blocks-and-partition
CFLAGS += -fno-tree-tail-merge
CFLAGS += -fno-guess-branch-probability
CFLAGS += -mno-serialize-volatile
CFLAGS += -fno-tree-copy-prop
```
Add/remove them one by one. This is highly code-dependent. -flto does not give me an advantage, for example. Maybe since I use -Wl,-gc-sections as a linker flag?

eriksl, 2019-02-23:
I am using that too (-Wl,-gc-sections) you need to have it active to make "-ffunction-sections -fdata-sections" effective.

I think you should not use -g for non-production builds, it turns off all kinds of optimisations. I would not be suprised that, when left out, -flto starts to actually work.

I have no floating point arithmetic in IRAM functions, so for me it's useless to do any further size optimisation there on fp options. I don't think "slow" code belong in IRAM and imho code using floats is exactly that.

I tried all -f options (including these) one by one and combined, and above is the list that left the most IRAM (!) space. I don't care about flash space, plenty of that.

And as said for lto to be effective, it needs to be passed to both compiling and linking stage, are you sure you're doing that?

kriegste, 2019-02-23:
-flto (added to compiling and linking) makes the binary bigger, actually. -g makes it smaller for me. And I have only 1k left, which is the only reason I care. Like I said, all these options are highly code-dependent...

eriksl, 2019-02-23:
Are you tight on flash or IRAM space?

Your gain with -g is probably because it turns off all kinds of optimisations (for speed). I'd rather not use -g but -Os and then find which optimisations cause the code to grow (which may be a kind of tedious work, been there, today...)

As said I have plenty of flash, it's only the IRAM I care about (hence this issue/topic).

eriksl, 2019-02-23:
I was curious and tried adding the -g, result
```
MEMORY USAGE
    iram:    available:  32 k, used:  32296, free:    472,  1 %
    dram:    available:  77 k, used:  73302, free:   5546,  7 %
    irom:    available: 408 k, used: 338084, free:  79708, 19 %
```
And without -g
```
MEMORY USAGE
    iram:    available:  32 k, used:  32296, free:    472,  1 %
    dram:    available:  77 k, used:  73302, free:   5546,  7 %
    irom:    available: 408 k, used: 338084, free:  79708, 19 %
```
Exactly the same. That's because all relevant optimisations are specified explicitly in my makefile. So in your case you'd best find the optimisation option that is trigger by -g (either enable or disable) so you can get rid of the -g. Maybe the -g also disables lto.

eriksl, 2019-02-23:
Now am thinking of it, -g also disables inlining. That may be the difference. I have -fno-inline to prevent that and select inlining explicitly using an attribute.

eriksl, 2019-02-23:
Without lto:
```MEMORY USAGE
    iram:    available:  32 k, used:  32380, free:    388,  1 %
    dram:    available:  77 k, used:  73378, free:   5470,  6 %
    irom:    available: 408 k, used: 339636, free:  78156, 18 %
```
Not such a big difference as earlier on, but I reshuffled a lot of code, between flash and IRAM and did quite some changes in inline/no-inline etc. It's still almost 100 bytes.

davydnorris, 2019-02-24:
This is a great discussion! I almost feel we should break it out and put it somewhere more formal - perhaps a thread on esp8266.com?

@eriksl what compiler/linker setting gives the section summaries above? I often use the map setting to get a detailed view of what is being pulled in by various bits of code but I always wanted a summary like that

I too have used flto and have found it makes a difference to me, and have also wished that the ESP supplied libraries were compiled with flto as well - it makes no difference if people aren't using flto switches themselves but could make a lot to those of us that do.

eriksl, 2019-02-24:
My linker doesn't know this option.

I am using this, in the makefile:
```
section_free	= $(Q) perl -e '\
						open($$fd, "xtensa-lx106-elf-size -A $(1) |"); \
						$$available = $(6) * 1024; \
						$$used = 0; \
						while(<$$fd>) \
						{ \
							chomp; \
							@_ = split; \
							if(($$_[0] eq "$(3)") || ($$_[0] eq "$(4)") || ($$_[0] eq "$(5)")) \
							{ \
								$$used += $$_[1]; \
							} \
						} \
						$$free = $$available - $$used; \
						printf("    %-8s available: %3u k, used: %6u, free: %6u, %2u %%\n", "$(2)" . ":", $$available / 1024, $$used, $$free, 100 * $$free / $$available); \
						close($$fd);'
```

Then do "make section_free" (or add section_free to your "all" target).

davydnorris, 2019-02-24:
Really? What are you using to build? I'm just using gcc 6.4.0 built from the esp_open_sdk base and latest crosstool

eriksl, 2019-02-24:
gcc 6.4.0, wow, it seems I have something to do here ;-)

I am using 4.8.5, which has been the standard for esp-open-sdk for years. I have requested newer versions several times, but there wasn't much interest, so I am kind of surprised they now suddenly have upgraded the toolchain.

eriksl, 2019-02-24:
I don't see a new gcc in http://github.com/pfalcon/esp-open-sdk, is there another origin for this repo now?

eriksl, 2019-02-24:
From a fresh build from esp-open-sdk, I still get gcc 4.8.5.

kriegste, 2019-02-24:
I use that old gcc 4.8.5, too.
Funny thing, I still do not know why -g gives me a smaller binary. Of course I tried the other options as well without -g. I ended up using some of the optimization switches in the linker run as well and this gave me 900 bytes more room in my flash.
But these options are always kind of voodoo. There are hundreds of them and some work together and some don't. But only with some code and not other. That's normal behavior. :o)

Another idea: You can try recompiling lwip and mbedtls, too, to minimize your iram. Maybe even remove some outdated ciphers and the ssl server stuff from mbedtls that is hardly usable anyway (it's all in the config file).

eriksl, 2019-02-24:
I am not using any TLS, so mbedtls is not linked in. Also it seems Espressif already made a lot of functions from lwip marked "flash".

Having said that, I have the plan to replace lwip with my own compiled one, some time, but it seems to be not a simple task.

I don't want TLS in my ESP8266. It takes up a huge amount of IRAM (which would prevent my image to work anyway) but at the same time doesn't give real security. If you want to do it really securely, insert a proxy or reverse proxy at the edge between the local network and internet.

eriksl, 2019-02-24:
I am also very curious where @davydnorris got his newer gcc version from.

kriegste, 2019-02-24:
Maybe
https://github.com/jcmvbkbc/gcc-xtensa/tree/master
(untested)

davydnorris, 2019-02-24:
Sorry - my comment was misleading.

I spent a lot of time trying to convince the esp-open-sdk guys to update the toolchain but i) they have moved onto the micropython project and ii) are of the opinion that if it's not broken then why change - which is their prerogative.

So I took esp-open-sdk, the latest pulls of crosstool, and jcmvbkbc/crosstool-NG (which hasn't changed since June 2017) and I patched crosstool so I could build the latest version of the toolchain. I see cherts had the same idea - his ESP environment is up to gcc 7.4.0!! I also use the toolchain with Cygwin64, not mingw.

I also changed the directory layout to be more like a real cross compiler environment - I now have a root file system at d:\xtools that has cross compilers for esp8266, ESP32, VoCore, RPi, C.H.I.P. and Texas Instruments DSPs all living together, with their SDKs all under /opt.

I'm very happy to share the toolchain, my current patched source tree, or the makefile patches and instructions if you want them. I documented my saga in a thread over on esp8266.com as well if you're interested.

eriksl, 2019-02-25:
@davydnorris I have been trying to understand the whole crosstool-NG thing for some time, but never really succeeded, now trying again. The amount of documentation to really get it working is absolutely minimal, please help me.

What I think I understand from it (googled)
- crosstool-NG makes compiler, binary tools and can even build a libc (which I don't use anyway)
- the official crosstool-NG does not have support for xtensa, at least not the lx106 variant used by the ESP8266
- @jcmvbkbc forked crosstool-NG and added patches so ESP2866 is supported. There are various branches of which is not completely clear which to use, xtensa-122.x is the default though, should one use that?
- this fork doesn't seem to be updated frequently and I don't see references to new gcc versions
- I also checked the crosstool-NG fork from @stilor, which I am building now, using the very generic instructions on the crosstool-NG website. The "samples" list only include two versions of xtensa, one using uclibc and one without, something with "fsf" which says nothing to me. The options include "use MMU" and for some things the config refers to repo's at Espressif including the term "ESP32" which lead me to believe this crosstool-NG fork is actually for ESP32 not ESP8266.

So any recommendations of what to do for own building of a newer gcc for ESP8266? On Linux please ;-)

someburner, 2019-02-25:
I use gcc 7.4.0 on my fork of esp-open-sdk: https://github.com/someburner/esp-open-sdk/tree/sdk2-lwip212

That branch has lwip 2.1.2 compiled in. You can just remove `lwip2` from the `all:` list and it should build a normal sdk. It also uses sdk 2.2.1, but works for 3.x too. You can see my work on other branches. One thing to be aware of is that later versions of crosstool-ng don't build into `sysroot`, and overriding crosstool-ng to do so by `cat`ing the `crosstool-config-overrides` file is "explicitly not supported" (maintainer is rather picky it sounds like, no reason given for why it is not supported), but for me it works.

eriksl, 2019-02-25:
In the meantime I found out that my supicion was true, the xtensa-fsf-elf-gcc from stilor is generic xtensa, not for ESP8266 and the byte ordering is the wrong way around to start with :-(

eriksl, 2019-02-25:
@someburner is sdk2-lwip212 the correct branch?

At the moment I only need the bare toolchain (no libs, no lwip, no sdk libs), but I am interested in replacing LWIP too, at least build it myself instead of using the prebuilt Espressif one. So this is a more recent version?

someburner, 2019-02-25:
Yes correct. That branch has the changes I made to build with GCC 7.4.0, but also includes building lwip2 based on this project: https://github.com/d-a-v/esp82xx-nonos-linklayer/pull/29

If you don't remove the lwip2 directive it will patch some of the SDK headers and build lwip2. So if you want to just try building the normal toolchain comment that out. You can always try building it later.

eriksl, 2019-02-25:
@someburner do you have somewhere some text on what you're working on and how it's meant to be used (e.g. the *linklayer* project)?

stilor, 2019-02-25:
JFYI, stilor/crosstool-ng is just a staging area before I integrate it back to "official" repo.

davydnorris, 2019-02-25:
I don't build into sysroot either, and agree with the crosstool folks - that's because I build crosstool for multiple architectures. Building into the toolchain sysroot means you have to clean everything to build again.

I use their option 3 - build into staging and install into a separate structure, then I use the Eclipse discovery to get my paths.

someburner, 2019-02-25:
@eriksl the PR discusses how it is used in my project. I am not the original author of it though, only making it work for esp-open-sdk. Essentially though you don't have to do much, the idea is that the SDK makes calls to lwip, and this works by wrapping those calls and acting as an interface between lwip1.4 and lwip2. You can `#include` espconn.h and use it like normal. But as mentioned in the PR you need to provide your own SNTP implementation, an example is given.

@davydnorris not saying it's not the "right" way to do it, but to maintain compatibility with esp-open-sdk as it is now, I don't know another way. It can get confusing having to know what directory to include and which not to. I have 1 architecture and don't mind 7 minute build times, but maybe I am missing out on some other benefit

eriksl, 2019-02-25:
Almost working, now I only have to fix this one it seems: "../../../../xtensa-lx106-elf/bin/ld:./loadscript:76 cannot move location counter backwards (from 000000003ffe8000 to 0000000000000004)"

eriksl, 2019-02-25:
@someburner whether it modifies SDK in any way or not is not relevant, I supply my own version in my project itself, using a git submodule construct. I don't use the SDK tree from the esp-open-sdk. I only need the toolchain :-)

eriksl, 2019-02-25:
@someburner apparently the semantics of the load script are changed slightly with the new gcc version. Can you share yours? I am getting really weird errors now.

someburner, 2019-02-25:
Mine is based on NodeMCU's. But that's strange you are getting issues- my project compiles the same with esp-open-sdk regardless of GCC version.

xcguang, 2019-02-26:
v5.2.0  toolchain can be download from https://github.com/espressif/ESP8266_RTOS_SDK/blob/master/README.md

eriksl, 2019-02-26:
RTOS! I don't want RTOS.

kriegste, 2019-02-26:
The links are under "Get toolchain". You can just swap your old gcc folder with that one.
I had to change xtensa-lx106-elf-ar to xtensa-lx106-elf-gcc-ar in my make file. Other than that it works perfectly. Thanks.

eriksl, 2019-02-26:
@someburner I have your toolchain working now, nice job! The error I was getting was from a suggested workaround for a misaligned exception vector, which I hope has been fixed now. Apparently the workaround was not 100% compliant.

For those interested, I see a huge shift in IRAM usage from certain optimisation options to other options. In general, though, it seems the newer gcc generates less compact code. I did not yet explore all options though. So I hope the code is at least more efficient... The LTO option still reduces the IRAM usage considerately, BTW.

kriegste, 2019-02-26:
It's the same with avr-gcc. Newer compilers are not always better.

But after playing around with the 5.2.0, it seems it is slightly better with my code.

BTW, I agree that the closed binaries should be compiled with -flto as well.

someburner, 2019-02-26:
@eriksl - Yes, for the ESP8266 I have found that the newer gcc results in slightly larger code. My reasons for using a later toolchain are mainly for new features- error/warn text coloring, more warn options, etc. Some of the new warning features are really handy. Functionally I have not seen any difference in the resulting app, but I have also not done any perf testing. Glad you got it working though!

@kriegste - I have found that newer gcc on avr (specifically atmega328p) gives me slightly smaller binaries, at least when using LTO. Judging by release notes, it looks like avr gets a lot of attention in gcc still.

eriksl, 2019-02-27:
@someburner here the same, extra features and more warnings.

Maybe we should attempt to get the patches/support into upstream crosstool-ng? Then everyone would profit.

eriksl, 2019-03-01:
I have found an interesting set of options for gcc 7 that generate same size or even smaller code than stock gcc 4.8.5, inclusing a very interesting but possily unsafe linker option called --size-opt.

That start with the latter, it is documented as "have the linker optimise for size not speed, so drop many alignments in the code where possible". I was under the impression that they were referring to just alignments. The ESP8266 has no handling in hardware for unaligment code/data, so one would need to write an exception handler (as explained by Cadence documentation) to do it in software (very inefficient, of course). The SDK does not have such a handler, so any unaligned access results in an exception -> reset.

Apparently this option is about something else. The instructions of Xtensa can be 1?, 2, 3 or 4 bytes long. So instructions are inherently mostly not aligned on 32 bits. The program counter, afaik, is required to always point to an aligned address, but apparently the instruction decoder / pipeline can handle unaligned instructions internally. So only parts of the code that are jumped to need to be aligned.

Also apparently, I am only guessing here, the xtensa cpu can execute certain / all instructions more efficiently when they're aligned. Maybe this is only relevant for cpu models that have cache (which the lx106 doesn't have).

Anyway, this option --size-opt (use -Wl,--size-opt on the gcc command line) reduces my IRAM usage by 316 bytes, 22% of my total IRAM usage. For so far, completely to my surprise, it just works.

Does anybody have experience with or knowledge of this option?

eriksl, 2019-03-01:
My optimisation options now are:

-Os -flto=8 -flto-compression-level=0 -fuse-linker-plugin -ffat-lto-objects -flto-partition=max -fno-tree-tail-merge -fno-guess-branch-probability -fno-tree-dominator-opts -fno-tree-forwprop -fno-tree-pta -fmerge-all-constants -frename-registers

For my timing relevant functions I use and attribute "optimise" that adds -O3 and -funroll-loops.

[10 more replies]

closed 2019-03-14 by eriksl