espressif/ESP8266_NONOS_SDK #210: How to use 4Bit devices?

jam3st, 2019-02-04:
I have a 4Mbit device. My partition table is thus:

``
    static const partition_item_t at_partition_table[] = { 
            { SYSTEM_PARTITION_BOOTLOADER,                      0x0,         0x1000},
            { SYSTEM_PARTITION_OTA_1,                           0x2000,     0x3A000},
            { SYSTEM_PARTITION_OTA_2,                           0x42000,    0x3A000},
            { SYSTEM_PARTITION_RF_CAL,                          0x7b000,     0x1000},  // 128 - 5 sectors of 4096
            { SYSTEM_PARTITION_PHY_DATA,                        0x7c000,     0x1000},
            { SYSTEM_PARTITION_SYSTEM_PARAMETER,                0x7d000,     0x3000}
``
The code is small and fits in this partition table and the RF_CAL data is in the usual place. How do I get this work with version 3 of the SDK?

jam3st, 2019-02-04:
Hey thanks for finding that. I changed the sizes to 0x39000 and I use 0 as the 3rd parameter of system_partition_table_regist call but I still get invalid partition table.

kriegste, 2019-02-04:
Not sure, but I think the first partition must begin at 0x01000.

jam3st, 2019-02-04:
I have read that too and changed that. I am using the rboot boot loader. With nodemcu I put 0x1000 in the OTA1 partition and flash at 0x2000 and don't have any problems.

I also read in issue https://github.com/espressif/esptool/issues/377 that the last paramater should be a 1. but if I do that I get a size mismatch error. Setting this to 0 makes the error go away.

In that issue it seems to suggest that OTA isn't support with 4Mbit flash. so I change the table to:
    static const partition_item_t at_partition_table[] = { 
            { 1 + SYSTEM_PARTITION_CUSTOMER_BEGIN,                      0x0,         0x1000},
            { 2 + SYSTEM_PARTITION_CUSTOMER_BEGIN,                           0x1000,     0x39000},
            { SYSTEM_PARTITION_RF_CAL,                          0x7b000,     0x1000},  // 128 - 5 sectors of 4096
            { SYSTEM_PARTITION_PHY_DATA,                        0x7c000,     0x1000},
            { SYSTEM_PARTITION_SYSTEM_PARAMETER,                0x7d000,     0x3000}
    };  

But I still get an error for system_partition_table_regist. If only the source was available...

jam3st, 2019-02-04:
Yes, that is 0. I think that I had seen those values when I was working on 32Mbit device last year. 
Anyhow, as the rf_cal data isn't read even the table without the OTA_1 values is being rejected. 
I don't understand why this is happening. In the previous versions you only needed to specify the rf_cal offset sector, usually flashed 5 sectors (each sector being 4096 bytes) from then end of the flash. If you don't use the OTA from the boot loader then I don't understand the reason the flash map is being rejected other than it is saying that it cannot handle this device. Yet the docs say that this should work. What is going on here?

kriegste, 2019-02-04:
Maybe try using the original bootloader. I never had any problems with it.

jam3st, 2019-02-05:
I gave up on using the partition table. I dumped the symbols for the app_partition.o file and I noticed that it still uses the globals system_rf_cal_sector so I used those directly.

closed 2019-02-05 by jam3st