espressif/ESP8266_NONOS_SDK #212: Upgrading to nonos SDK 3.10

scargill, 2019-02-21:
I've not upgraded since NONOS SDK 2.1 due to not enough IRAM in later versions. I use RBOOT by Richard Burton - virtually all my functions are in Flash - my code works perfectly in SDK 2.1 - . I've not kept up with the need to define a partition table. Does the latest version (today - 21/02/2019) free up more space and do the docs now simply explain partition table information?

fvpalha, 2019-02-21:
I'm using RBOOT, too.
I am using NON SDK 2.2.x commit 34f1abd (12/27/2018).
I'm still not sure on upgrading to the NONOS SDK 3.1.0

The project [esp_nano_httpd_fota](https://github.com/QB4-dev/esp_nano_httpd_fota) can be an alternative to RBOOT.

marcelstoer, 2019-02-21:
I haven't seen a v3.1 in the wild yet. Pointer?

scargill, 2019-02-21:
Github- it’s there and updated today.

marcelstoer, 2019-02-21:
Never mind, I thought you were referring to an official release. Seems you just mean the latest commit on master - which was first labeled as v3.1 on Jan 2nd.

scargill, 2019-02-21:
I don’t know the difference, I assumed if it’s on GIT it is official.

davydnorris, 2019-02-21:
I've been using latest since it came out - for me moving to the partition table model was only a minor hassle in terms of updating my code and my Makefile, but there are several things I have noticed that are quite cool:
 - my flash map is now explicit. I can say exactly how much room my user flash is allowed and can check at build and OTA update time
 - my SSL certificate locations are also explicit and I now use the partition table functions to locate them when needed
 - my user data partition is explicitly defined as well

Truth be told, these things were explicit before, but they were all defined solely in my makefile, and I had separate definitions for the various flash sizes I had to deal with. It's now laid out in my code, and automatically calculated from a single flash size define passed in during the build. I use additional macro directives in the code to output information about where to flash the various items, but these are also in my makefile

My code is now pretty much independent of the makefile settings

scargill, 2019-02-21:
That’s nice, Davy but in all this time I still have little clue how the make file works, never mind partition tables. What I have works all the way up to SDK 2.1 – I’m not a Linux programmer so some of that remains beyond me. I don’t know where to start changing the make file.

someburner, 2019-02-22:
While I think it's important to actually understand the partition table, which is really very simple- it's just a short list mapping addresses to various sectors of flash needed by the SDK- I can see it being a little confusing if you don't already have these locations defined in your project.

For a 4MB ESP8266 (eg ESP-12E) and rBoot with 512KB roms, all you have to do is, in `user_main.c`, make it look like this:

```c
#define SDK_RF_CAL_ADDR 0x3FB000
#define SDK_PHY_DATA_ADDR 0x3FC000
#define SDK_PARAM_ADDR 0x3FD000
#define SPI_FLASH_SIZE_MAP  FLASH_SIZE_32M_MAP_512_512 // 4
#define SDK_PRIV_PARAM_ADDR 0x7C000
#define SYSTEM_PARTITION_CUSTOMER_PRIV_PARAM SYSTEM_PARTITION_CUSTOMER_BEGIN

static const partition_item_t p_table[] = {
    { SYSTEM_PARTITION_BOOTLOADER,          0x0,                 0x1000 },
    { SYSTEM_PARTITION_RF_CAL,              SDK_RF_CAL_ADDR,     0x1000 },
    { SYSTEM_PARTITION_PHY_DATA,            SDK_PHY_DATA_ADDR,   0x1000 },
    { SYSTEM_PARTITION_SYSTEM_PARAMETER,    SDK_PARAM_ADDR,      0x3000 },
    { SYSTEM_PARTITION_CUSTOMER_PRIV_PARAM, SDK_PRIV_PARAM_ADDR, 0x1000 },
};

void user_pre_init(void)
{
    if(!system_partition_table_regist(p_table, sizeof(p_table)/sizeof(p_table[0]), SPI_FLASH_SIZE_MAP)) {
        os_printf("system_partition_table_regist fail\r\n");
        while(1);
    }
}
```

davydnorris, 2019-02-22:
Actually @someburner, there are a few things you would need to change:
 - if you're using rBoot, you need two sectors for the boot ROM so the first line needs to be changed to reflect the larger size:
`{ SYSTEM_PARTITION_BOOTLOADER,          0x0,                 0x2000 },`

and you also need to include the actual code partitions in this list, which for an OTA enabled 4MB 512k system would start at 0x002000 an 0x082000 (0x2000 higher than the base OTA due to rBoot), and would be 512k in size, minus 2k for rBoot and minus whatever size needed for your system or your own  parameters, whichever is larger. In this case let's assume the system parameters are largest size at 1+1+3=5k, so our max OTA partition size is 0x80000 - 0x002000 - 0x005000 = 0x079000

```
#define SYSTEM_PARTITION_OTA_SIZE							0x79000

{ SYSTEM_PARTITION_OTA_1,              0x002000,     SYSTEM_PARTITION_OTA_SIZE },
{ SYSTEM_PARTITION_OTA_2,              0x082000,     SYSTEM_PARTITION_OTA_SIZE },
```

someburner, 2019-02-22:
@davydnorris 

Good catch re: rBoot size. Although technically the rboot binary only takes a single sector, the 2nd sector stores configuration.

Regarding `SYSTEM_PARTITION_OTA_N` parameters, it was my understanding that is only required if using the Espressif OTA method, which I'm not even sure works for rBoot? The configuration I posted boots and OTAs using [rBoot-style OTA](https://github.com/raburton/rboot-sample/blob/master/rboot-ota.c).

davydnorris, 2019-02-22:
I think you may be right there regarding the rBoot OTA approach. I don't know if there's any way you can specify that info for rBoot (I seem to remember seeing something similar though).

I've found having the explicit OTA partition size has saved me once already when I accidentally set up an OTA flash built for a 1M partition for a set of devices that were configured as 512k

scargill, 2019-02-22:
Ok, now take it easy with me…. can I use that as below- assuming again 4MB ESP8266 and my code is somewhat OVER 512k…

scargill, 2019-02-22:
Which is the correct version - assuming 4MB ESP12 and roms that are in between 512k and 1 meg each... I think I'm up to 700k.??? Your feedback, guys, is MOST appresiated... so that's it? a  change to user_main.c?

someburner, 2019-02-22:
If you look in the SDK's `include/user_interface.h` you can find the `flash_size_map` enum that you need to look at to choose for the `#define SPI_FLASH_SIZE_MAP` definition. So if you have roms > 512KB, you will need to use the 1MB size, which would be `FLASH_SIZE_32M_MAP_1024_1024`, so the definition would be:

```c
#define SPI_FLASH_SIZE_MAP FLASH_SIZE_32M_MAP_1024_1024
```

scargill, 2019-02-23:
Chaps. thank you SO MUCH - can we just confim that below is correct now for 4 MB ESP12 with 1 meg roms using RBOOT. Looks right but clearly those writing in are better t this than I am.... I'm on SDK 2.1 - is this it -  add this to user_main.c for moving to the latest sdk?

#define SDK_RF_CAL_ADDR 0x3FB000
#define SDK_PHY_DATA_ADDR 0x3FC000
#define SDK_PARAM_ADDR 0x3FD000
#define SPI_FLASH_SIZE_MAP FLASH_SIZE_32M_MAP_1024_1024
#define SDK_PRIV_PARAM_ADDR 0xFC000
#define SYSTEM_PARTITION_CUSTOMER_PRIV_PARAM SYSTEM_PARTITION_CUSTOMER_BEGIN

static const partition_item_t p_table[] = {
    { SYSTEM_PARTITION_BOOTLOADER,          0x0,                 0x1000 },
    { SYSTEM_PARTITION_RF_CAL,              SDK_RF_CAL_ADDR,     0x1000 },
    { SYSTEM_PARTITION_PHY_DATA,            SDK_PHY_DATA_ADDR,   0x1000 },
    { SYSTEM_PARTITION_SYSTEM_PARAMETER,    SDK_PARAM_ADDR,      0x3000 },
    { SYSTEM_PARTITION_CUSTOMER_PRIV_PARAM, SDK_PRIV_PARAM_ADDR, 0x1000 },
};

void user_pre_init(void)
{
    if(!system_partition_table_regist(p_table, sizeof(p_table)/sizeof(p_table[0]), SPI_FLASH_SIZE_MAP)) {
        os_printf("system_partition_table_regist fail\r\n");
        while(1);
    }
}

scargill, 2019-02-23:
NICE - but I had a feeling it would not be THAT simple - so - I added the code above, adjusted for 1 meg roms - and upgraded to the SDK here in GIT. NOPE. 

In SDK 2.1 I get 
build/app_0.out:     file format elf32-xtensa-le

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .data         00000d38  3ffe8000  3ffe8000  000000e0  2**4
                  CONTENTS, ALLOC, LOAD, DATA
  1 .rodata       00002ea8  3ffe8d40  3ffe8d40  00000e20  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  2 .bss          0000b2e0  3ffebbe8  3ffebbe8  00003cc8  2**4
                  ALLOC
  3 .text         00007774  40100000  40100000  00003cc8  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  4 .irom0.text   0009d284  40202010  40202010  0000b440  2**4
                  CONTENTS, ALLOC, LOAD, READONLY, CODE

As you can see above compilation works in 2.1 and I still have .text room to go.

Using the new SDK 3.1 - with the additional info as advised above, I get all the way to the end of the compilation then the dreaded error - 

c:/espressif/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: build/app_0.out section `.text' will not fit in region `iram1_0_seg'
collect2.exe: error: ld returned 1 exit status
mingw32-make.exe: *** [build/app_0.out] Error 1

Just as I've had with previous attempts to upgrade the SDK. Ideas, anyone???

scargill, 2019-02-23:
One more change to the above - I changed the bootloader size to 0x2000 as davydnorris pointed out... made no difference - same ".text will nor fit into region iram1_0_seg" error.

davydnorris, 2019-02-24:
I am happy to take a look at your project @scargill and see if I can get it to fit?

scargill, 2019-02-24:
Here it is:  https://bitbucket.org/scargill/esp-go/src/master/ - this one is up to date -- works with SDK 2.1 - the new info in user_main.c is commented out as it prevents compiling for 2.1 - the main make file is in the outer directory - simply called MAKEFILE - th emain file which calls others is /user/petes_code.c

The project is written in C, not using Arduino.... and originally emanated from an MQTT demo and now conrtols a range of peripherals by serial of MQTT.

I'm in the process of adding more fonts.

Pete

davydnorris, 2019-02-24:
OK I've pulled the code and had a bit of a go - had to fiddle a lot with the Makefile, and pull down esptool2 and make that as well, and play with your python script to make Python 3.7 like it, and then I started building...

By default I turn on all warnings and there's a lot to fix - lots of implicit function declarations, which can make the code not optimise very well, and a bunch of other dodgy assignments to and from int32 and pointers. I'll have a bit more of a play but it's taken a chunk of my time already.

scargill, 2019-02-24:
Thank you so much for your efforts… sorry about esptool2. I use Python 2.7 – never thought about someone not having that.

 

The idea here is not to worry too much about making my code more efficient – but finding out why it fits into sdk 2.1 and nor 3.1 – Espressif are supposed to have done some optimisations, putting more of their library in FLASH. If I end up with less, not more room then there is no point in upgrading the SDK – I was hoping to get past the IRAM issue and get more space for expansion.

 

Regards

 

Pete

davydnorris, 2019-02-24:
I had to change it just to get it to compile - trying not to change anything at all if I can help it. One thing I am doing is pulling the espmissingincludes.h header as they have fixed all that up now. But the rest is just missing headers and definitions

fvpalha, 2019-02-28:
Hi.
Someone have a simple example with rboot in SKD 3.1.0 ?

scargill, 2019-02-28:
Even with new partition data and pre init I've still not managed a 
successful compile higher than sdk 2.1

fvpalha, 2019-02-28:
I compiled in SDK 3.1.0 after add the code:

```
#include "user_interface.h"

#define SDK_RF_CAL_ADDR 0x3FB000
#define SDK_PHY_DATA_ADDR 0x3FC000
#define SDK_PARAM_ADDR 0x3FD000
#define SPI_FLASH_SIZE_MAP  FLASH_SIZE_32M_MAP_512_512 // 4
#define SDK_PRIV_PARAM_ADDR 0x7C000
#define SYSTEM_PARTITION_CUSTOMER_PRIV_PARAM SYSTEM_PARTITION_CUSTOMER_BEGIN

static const partition_item_t p_table[] = {
    { SYSTEM_PARTITION_BOOTLOADER,          0x0,                 0x1000 },
    { SYSTEM_PARTITION_RF_CAL,              SDK_RF_CAL_ADDR,     0x1000 },
    { SYSTEM_PARTITION_PHY_DATA,            SDK_PHY_DATA_ADDR,   0x1000 },
    { SYSTEM_PARTITION_SYSTEM_PARAMETER,    SDK_PARAM_ADDR,      0x3000 },
    { SYSTEM_PARTITION_CUSTOMER_PRIV_PARAM, SDK_PRIV_PARAM_ADDR, 0x1000 },
};

void user_pre_init(void)
{
    if(!system_partition_table_regist(p_table, sizeof(p_table)/sizeof(p_table[0]), SPI_FLASH_SIZE_MAP)) {
        os_printf("system_partition_table_regist fail\r\n");
        while(1);
    }
}
```

But my code does not run after burn the ESP8266.

scargill, 2019-02-28:
Doesn’t work??? Not a lot of point in copying that, then? Mine is slightly different anyway as my code sdoesn’t fit in 521k, I’m up to about 800k…

 

If you get it to work, please do let me know.

fvpalha, 2019-03-01:
@davydnorris and @someburner , can you share the Makefile and the main file of the project than works with rBoot in SDK 3.1.0 ?

I do not understand why the rom does not run.

Thank you.

eriksl, 2019-03-02:
The latest (not officially released) commits from the SDK (master/HEAD) free up a LOT of IRAM. Definitely worth a try.

If you're using rboot, I don't think it's wise to mark it's flash regions as "well-known" SDK partition, like OTA_1 and OTA_2 etc. The SDK code will make assumptions about it, which may not hold. RBOOT is not the same as the SDK OTA mechanism.

This is my partition table (for use with either rboot or without), as an example for use with rboot:
```
static const partition_item_t partition_items[] =
{
	{	SYSTEM_PARTITION_RF_CAL, 				RFCAL_OFFSET,				RFCAL_SIZE,				},
	{	SYSTEM_PARTITION_PHY_DATA,				PHYDATA_OFFSET,				PHYDATA_SIZE,			},
	{	SYSTEM_PARTITION_SYSTEM_PARAMETER,		SYSTEM_CONFIG_OFFSET,		SYSTEM_CONFIG_SIZE,		},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 0,	USER_CONFIG_OFFSET,			USER_CONFIG_SIZE,		},
#if IMAGE_OTA == 0
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 1,	OFFSET_IRAM_PLAIN,			SIZE_IRAM_PLAIN,		},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 2,	OFFSET_IROM_PLAIN,			SIZE_IROM_PLAIN,		},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 3,	SEQUENCER_FLASH_OFFSET,		SEQUENCER_FLASH_SIZE,	},
#else
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 1,	OFFSET_OTA_BOOT,			SIZE_OTA_BOOT,			},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 2,	OFFSET_OTA_RBOOT_CFG,		SIZE_OTA_RBOOT_CFG,		},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 3,	OFFSET_OTA_IMG_0,			SIZE_OTA_IMG,			},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 4,	OFFSET_OTA_IMG_1,			SIZE_OTA_IMG,			},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 5,	SEQUENCER_FLASH_OFFSET_0,	SEQUENCER_FLASH_SIZE,	},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 6,	SEQUENCER_FLASH_OFFSET_1,	SEQUENCER_FLASH_SIZE,	},
#endif
};
```

scargill, 2019-03-03:
Eriksl - you have me confused - up to now I've ben working on the asumption that this was required and nothing else. Clealry below is not QUITE right as it won't compile - the usual "won't fit IRAM" message.

but now you've introduced something called partition_items[] - care to elaborate for non-Linux tpyes like me?

```
#define SDK_RF_CAL_ADDR 0x3FB000
#define SDK_PHY_DATA_ADDR 0x3FC000
#define SDK_PARAM_ADDR 0x3FD000
#define SPI_FLASH_SIZE_MAP FLASH_SIZE_32M_MAP_1024_1024
#define SDK_PRIV_PARAM_ADDR 0xFC000
#define SYSTEM_PARTITION_CUSTOMER_PRIV_PARAM SYSTEM_PARTITION_CUSTOMER_BEGIN

static const partition_item_t p_table[] = {
{ SYSTEM_PARTITION_BOOTLOADER, 0x0, 0x2000 },
{ SYSTEM_PARTITION_RF_CAL, SDK_RF_CAL_ADDR, 0x1000 },
{ SYSTEM_PARTITION_PHY_DATA, SDK_PHY_DATA_ADDR, 0x1000 },
{ SYSTEM_PARTITION_SYSTEM_PARAMETER, SDK_PARAM_ADDR, 0x3000 },
{ SYSTEM_PARTITION_CUSTOMER_PRIV_PARAM, SDK_PRIV_PARAM_ADDR, 0x1000 },
};

void user_pre_init(void)
{
if(!system_partition_table_regist(p_table, sizeof(p_table)/sizeof(p_table[0]), SPI_FLASH_SIZE_MAP)) {
os_printf("system_partition_table_regist fail\r\n");
while(1);
}
}
```

kriegste, 2019-03-03:
That's the name of the array!? In your example it is called p_table.

BTW, have you tried ditching rboot completely?

scargill, 2019-03-03:
I would not know whwre to start ditching RBOOT and the learing curve would detract from what I’m actually making with all of this. RBOOT has worked perfectly for a long time.

 

So do I take it you can call the array anything…. Indicating that pre-init is the only place that refers to it by name?

scargill, 2019-03-03:
No I understand – and your table is like the one I picked (an example further up) but mine has less in it – and still won’t compile in 3.1 (the whole point of this). A reminder that my code compiles and works every time in SDK 2.1 – any idea what changes are needed to make it work?

 

Pete

davydnorris, 2019-03-04:
still looking at it @scargill - just got my own delivery on right now

fvpalha, 2019-03-05:
Hi @eriksl 

I am trying follow your suggestions, but yet do not work.
The project compile, but the firmware (ROM) does not run in ESP8266.

I created an simple project to make an functional example using [rBoot](https://github.com/raburton/rboot) with [NONOS SDK 3.1.0](https://github.com/espressif/ESP8266_NONOS_SDK).

[test_rBoot_NonOS_SDK3.1.0](https://github.com/fvpalha/test_rBoot_NonOS_SDK3.1.0)

Please @raburton, @davydnorris, @someburner and someone who has an interest in the solution.

Thanks.

> The latest (not officially released) commits from the SDK (master/HEAD) free up a LOT of IRAM. Definitely worth a try.
> 
> If you're using rboot, I don't think it's wise to mark it's flash regions as "well-known" SDK partition, like OTA_1 and OTA_2 etc. The SDK code will make assumptions about it, which may not hold. RBOOT is not the same as the SDK OTA mechanism.
> 
> This is my partition table (for use with either rboot or without), as an example for use with rboot:
> 
> ```
> static const partition_item_t partition_items[] =
> {
> 	{	SYSTEM_PARTITION_RF_CAL, 				RFCAL_OFFSET,				RFCAL_SIZE,				},
> 	{	SYSTEM_PARTITION_PHY_DATA,				PHYDATA_OFFSET,				PHYDATA_SIZE,			},
> 	{	SYSTEM_PARTITION_SYSTEM_PARAMETER,		SYSTEM_CONFIG_OFFSET,		SYSTEM_CONFIG_SIZE,		},
> 	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 0,	USER_CONFIG_OFFSET,			USER_CONFIG_SIZE,		},
> #if IMAGE_OTA == 0
> 	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 1,	OFFSET_IRAM_PLAIN,			SIZE_IRAM_PLAIN,		},
> 	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 2,	OFFSET_IROM_PLAIN,			SIZE_IROM_PLAIN,		},
> 	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 3,	SEQUENCER_FLASH_OFFSET,		SEQUENCER_FLASH_SIZE,	},
> #else
> 	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 1,	OFFSET_OTA_BOOT,			SIZE_OTA_BOOT,			},
> 	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 2,	OFFSET_OTA_RBOOT_CFG,		SIZE_OTA_RBOOT_CFG,		},
> 	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 3,	OFFSET_OTA_IMG_0,			SIZE_OTA_IMG,			},
> 	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 4,	OFFSET_OTA_IMG_1,			SIZE_OTA_IMG,			},
> 	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 5,	SEQUENCER_FLASH_OFFSET_0,	SEQUENCER_FLASH_SIZE,	},
> 	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 6,	SEQUENCER_FLASH_OFFSET_1,	SEQUENCER_FLASH_SIZE,	},
> #endif
> };
> ```
> The actual values are supplied from the Makefile using -D...

davydnorris, 2019-03-05:
Hi @fvpalha - I don't actually use rBoot, I've just been using 3.1 for ages.

I had a look at your test project, and to be honest it's a bit of a mess. From what I can tell, you have dumped the rBoot source, the driver source supplied be Espressif, and a bunch of other stuff all into one directory. It's impossible to tell why your project isn't working - you need to make it really easy for us to help you!

Set up some subdirectories with the various standard pieces in, and then set up your user directory with your simple sample so that we know what file to look at

scargill, 2019-03-05:
DavyNorris - as I have similar issues, you're more than welcome to take a look at my code - which is NOT all in one folder ....https://bitbucket.org/scargill/esp-go/src/master/ 

There are lots of files inc support for a range of peripherals - the main MAKEFILE (I've never used the linux makefile) is in the outer directory - and the "main page" is in user/petescode.c - the /user/user_main.c page is set up for sdk 2.1 and earlier - from what I've seen in here, my commented out partition table and pre_init function are not a lillion miles out - but I cannot get compilation in 3.1 or anything past 2.1  ---- the code works perfectly in 2.1 - masically mqtt and serial communications (there's a web interface but I set everyting up (2 ssids, id and an mqtt broker) in serial usually.

Pete

davydnorris, 2019-03-05:
@scargill I am looking at yours but have my own code drop that's running a week late. As soon as I get that sorted I'll have a play.

As mentioned before, I tried to compile yours and got a ton of warnings. One of the things I've found is that undeclared functions can affect compiler optimisations, and you have a lot of those because your makefile suppressed the warnings.

scargill, 2019-03-05:
I wasn’t aware you were supposed to declare them and I’ve no idea how to turn warnings on and off, though I prefer the relevant silence of not having them on.. an example of that at the many scripts out there some of which I incorporate into “the script”. There are LOTS of warnings about which I can do nothing, I wish I could turn them off.

 

If you give me one concrete example of a function needing defining – and the simplest way to turn on the warnings reversibly, I’ll happily look at anything that could easily be cured in that respect and update the repository… but remember I’m primarily a Windows C-coder guy, much of this stuff does not come easy to me – I’d imagine that much is obvious by now.

 

And thanks for your interest and help.

 

Pete

fvpalha, 2019-03-06:
Hi.

I can see that @davydnorris knows how to evaluate the quality of third-party code. But does he know how to help?

I think the main files are the "Makefile" and the "main.c" for rBoot to work in SDK 3.1.0.

“It was pride that changed angels into devils; it is humility that makes men as angels.”
— St. Augustine

Regards.

scargill, 2019-03-06:
I only hope that Davynorris or someon can help with what SHOULD be an easy problem - my (large) project works a treat in SDK 2.1 and has worked in all previous versions - but I cannot move on - surewly the point of SDK updates is to make things better, not have people jump through hoops to upgrade.  I've been unable up to now to get anyone to give a simple, working answer. People say the new SDK has clawed back lots of room - FINE if you can actaully get a compilation to complete. I've not been able to - I'm not sure if my partition table is just incorrect or if it is something else. ake out the pre-init and partition table - go back to SDK 2.1 and it is just fine with a little iram room to spare. I was hoping to get even more by upgrading.

kriegste, 2019-03-06:
Have you tried removing parts of your project? Commenting out stuff. Putting a "return" as first line in a function. I mean, several functions. Just to make it smaller and see if the compiler still complains.

scargill, 2019-03-06:
Hi Kriegste – somewhat pointless – the point of upgrading is to fit MORE in, not less. Ta.

 

Pete

davydnorris, 2019-03-06:
> 
> Hi Kriegste – somewhat pointless – the point of upgrading is to fit MORE in, not less. Ta. 

Actually @kriegste is right - the point is to get something to compile at all, and *THEN* fit more in :-)

Have you at any point got your code to compile and load using 3.1 regardless of size? Let's get that happening first and then analyse where where the iram is going.

eriksl, 2019-03-06:
> Eriksl - you have me confused - up to now I've ben working on the asumption that this was required and nothing else. Clealry below is not QUITE right as it won't compile - the usual "won't fit IRAM" [...]

You keep insisting you're "not a linux guy", but almost anything I mention has nothing to do with Linux at all.

I never said creating a partition table would by miracle free up tons of IRAM space. So if you just add this and compile, it's to be expected that you still run out of IRAM space and the linker (not the compiler) will complain. Be happy it does that, otherwise you wouldn't now and your code would just crash at some point without a hint about the cause.

What I did say is that in recent 3.x commits a lot of IRAM space was freed up from the SDK code. And so your project is more likely to fit. And to be able to run 3.x you need to have the partition table.

One step at a time.

eriksl, 2019-03-06:
> Eriksl - you have me confused - up to now I've ben working on the asumption that this was required and nothing else. Clealry below is not QUITE right as it won't compile - the usual "won't fit IRAM" [...]

Now, about using rboot. I already explained this to you before. Rboot is a mechanism that does not use any of the SDK OTA features. So for the SDK you need to make it look like it's a "plain" image. The image, as seen from the SDK code, is the rboot code + your image, it cannot distinguish between the two. So don't confuse it.

So, using rboot, your partition table NEEDS to be something like this (add more "user" partitions at will, but these are the bare minimum):

```
static const partition_item_t partition_items[] =
{
	{	SYSTEM_PARTITION_RF_CAL, 				RFCAL_OFFSET,				RFCAL_SIZE,				},
	{	SYSTEM_PARTITION_PHY_DATA,				PHYDATA_OFFSET,				PHYDATA_SIZE,			},
	{	SYSTEM_PARTITION_SYSTEM_PARAMETER,		SYSTEM_CONFIG_OFFSET,		SYSTEM_CONFIG_SIZE,		},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 0,	USER_CONFIG_OFFSET,			USER_CONFIG_SIZE,		},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 1,	OFFSET_OTA_BOOT,			SIZE_OTA_BOOT,			},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 2,	OFFSET_OTA_RBOOT_CFG,		SIZE_OTA_RBOOT_CFG,		},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 3,	OFFSET_OTA_IMG_0,			SIZE_OTA_IMG,			},
	{	SYSTEM_PARTITION_CUSTOMER_BEGIN + 4,	OFFSET_OTA_IMG_1,			SIZE_OTA_IMG,			},
};
```

eriksl, 2019-03-06:
> I'm using RBOOT, too.
> I am using NON SDK 2.2.x commit [34f1abd](https://github.com/espressif/ESP8266_NONOS_SDK/commit/34f1abd9c4196e02180fe41c42a4027d2433fbb2) (12/27/2018).
> I'm still not sure on [...]

It can be done quite easily, see my partition table and it will work. I don't see any necessity to drop rboot.

eriksl, 2019-03-06:
> While I think it's important to actually understand the partition table, which is really very simple- it's just a short list mapping addresses to various sectors of flash needed by the SDK- I can see it being a little confusing if you don't already have these locations defined in your project.

I would like to add to that, that the partition table serves two purposes. One is to signal area's of the flash memory to SDK, so they can use it (and your code won't) and another purpose is to protect the flash area's (from your own project, data) from getting overwritten. I am not completely sure how that's achied, but at least it's good to have your flash layout documented (like you say).

> For a 4MB ESP8266 (eg ESP-12E) and rBoot with 512KB roms, all you have to do is, in `user_main.c`, make it look like this:

I never use rboot this way. It's a big hassle. Always having to generate two different images and make sure you're flashing the correct one. So I make sure my devices always have at least 2 Mb and make use of the "bigflash" feature from rboot, where you can use just one image and then rboot takes care to set up the propper address mapping. @scargill if you're not already the bigflash feature, I would highly recommend it.

eriksl, 2019-03-06:
> Actually @someburner, there are a few things you would need to change:
>
> * if you're using rBoot, you need two sectors for the boot ROM so the first line needs to be changed to reflect the larger [...]

I don't think you should do it this way. The SDK makes certain assumptions about the predefined partitions. And "OTA" partition is an "SDK OTA" partition and a "bootloader" partition is an "SDK bootloader" partition. I seem to remember I couldn't even get it to work this way and quickly switched to USER partitions (see my example) and then it works. The SDK code does not mind if bootloader or "OTA" partitions are missing. The SDK code just assumes a non-SDK image, which, as seen from the SDK code, it actually true. The SDK code starts the rboot loader + the image, it does not know about rboot. After rboot is started just like a normal image, reboot "trampolines" to your own image and the SDK code does not see that.

eriksl, 2019-03-06:
> c:/espressif/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: build/app_0.out section `.text' will not fit in region `iram1_0_seg'
> collect2.exe: error: [...]

That is because you still did not adapt your loadscript to the new compilation options that are used building the v3 SDK. If you don't change the load script, almost all functions will end up in IRAM, even though you did not specify them as such. But if you **do** adapt the load script, you'll see you'll end up actually with MORE free IRAM because many SDK functions that end up in IRAM, are no longer linked in, because they're not required by your project.

Als you need to add --gc-sections to your linker command line to make this work. If you have a line with LDFLAGS in your Makefile "somewhere", add this: "-Wl,--gc-sections". While you're at it, also add "-Wl,--size-opt". That also frees up additional IRAM (experimental). Don't type the quotes, but copy&paste the rest, it won't work if spaces are added or removed etc.

This is an example of a working load script. You will need to adjust the sections start addresses and sizes, probably the path to the include file at the bottom and maybe a few other things at all. Espressif has these changes in their supplied load scripts in the SDK as well, so it's not rocket science in itself ;-)

```
/* This linker script generated from xt-genldscripts.tpp for LSP . */
/* Linker Script for ld -N */
MEMORY
{
  dport0_0_seg :                        org = 0x3FF00000, len = 0x10
  dram0_0_seg :                         org = 0x3FFE8000, len = 0x14000
  iram1_0_seg :                         org = 0x40100000, len = 0x8000
  irom0_0_seg :                         org = 0x40202010, len = 0xf7ff0
}

PHDRS
{
  dport0_0_phdr PT_LOAD;
  dram0_0_phdr PT_LOAD;
  dram0_0_bss_phdr PT_LOAD;
  iram1_0_phdr PT_LOAD;
  irom0_0_phdr PT_LOAD;
}

/*  Default entry point:  */
ENTRY(call_user_start)
EXTERN(_DebugExceptionVector)
EXTERN(_DoubleExceptionVector)
EXTERN(_KernelExceptionVector)
EXTERN(_NMIExceptionVector)
EXTERN(_UserExceptionVector)
PROVIDE(_memmap_vecbase_reset = 0x40000000);
/* Various memory-map dependent cache attribute settings: */
_memmap_cacheattr_wb_base = 0x00000110;
_memmap_cacheattr_wt_base = 0x00000110;
_memmap_cacheattr_bp_base = 0x00000220;
_memmap_cacheattr_unused_mask = 0xFFFFF00F;
_memmap_cacheattr_wb_trapnull = 0x2222211F;
_memmap_cacheattr_wba_trapnull = 0x2222211F;
_memmap_cacheattr_wbna_trapnull = 0x2222211F;
_memmap_cacheattr_wt_trapnull = 0x2222211F;
_memmap_cacheattr_bp_trapnull = 0x2222222F;
_memmap_cacheattr_wb_strict = 0xFFFFF11F;
_memmap_cacheattr_wt_strict = 0xFFFFF11F;
_memmap_cacheattr_bp_strict = 0xFFFFF22F;
_memmap_cacheattr_wb_allvalid = 0x22222112;
_memmap_cacheattr_wt_allvalid = 0x22222112;
_memmap_cacheattr_bp_allvalid = 0x22222222;
PROVIDE(_memmap_cacheattr_reset = _memmap_cacheattr_wb_trapnull);

SECTIONS
{
  .dport0.rodata : ALIGN(4)
  {
    _dport0_rodata_start = ABSOLUTE(.);
    *(.dport0.rodata)
    *(.dport.rodata)
    _dport0_rodata_end = ABSOLUTE(.);
  } >dport0_0_seg :dport0_0_phdr

  .dport0.literal : ALIGN(4)
  {
    _dport0_literal_start = ABSOLUTE(.);
    *(.dport0.literal)
    *(.dport.literal)
[... 143 lines cut ...]
    *(.entry.text)
    *(.init.literal)
    *(.init)
    *(.literal .text .literal.* .text.* .stub .gnu.warning .gnu.linkonce.literal.* .gnu.linkonce.t.*.literal .gnu.linkonce.t.*)
    *(.fini.literal)
    *(.fini)
    *(.gnu.version)
    _text_end = ABSOLUTE(.);
    _etext = .;
  } >iram1_0_seg :iram1_0_phdr

  .lit4 : ALIGN(4)
  {
    _lit4_start = ABSOLUTE(.);
    *(*.lit4)
    *(.lit4.*)
    *(.gnu.linkonce.lit4.*)
    _lit4_end = ABSOLUTE(.);
  } >iram1_0_seg :iram1_0_phdr
}

/* get ROM code address */
INCLUDE "../ld/eagle.rom.addr.v6.ld"

```

eriksl, 2019-03-06:
> I am trying follow your suggestions, but yet do not work.
> The project compile, **but the firmware (ROM) does not run in ESP8266.**

What do you mean by that?

Maybe you're running into the same issue I ran into lately. When I generate an image for a 512 KB flash device, the image won't start. If I change it into an "1 MB 512-512" image, it can be flashed just fine and also runs fine. It seems the whole thong crashes before even anything of my code is called (tested and debugged) so my possibilities end there. It's also not rboot's fault, because it also happens with a "plain" image (no rboot involved).

eriksl, 2019-03-06:
> Hi @fvpalha - I don't actually use rBoot, I've just been using 3.1 for ages.
> 
> I had a look at your test project, and to be honest it's a bit of a mess. From what I can tell, you have dumped the rBoot source, the driver source supplied by Espressif, and a bunch of other stuff all into one directory. It's impossible to tell why your project isn't working - you need to make it really easy for us to help you!
> 
> Set up some subdirectories with the various standard pieces in, and then set up your user directory with your simple sample so that we know what file to look at

In itself it's fine, I also use rboot, the SDK and (now) the toolchain from within the projects directory but then make them git submodules, don't copy them.

eriksl, 2019-03-06:
> I wasn’t aware you were supposed to declare them

That's basic ANSI C, not Linux knowledge :-)

If you're going to use a function in another compilation unit (i.e. it's not declared static and it's also used in another .c file), you NEED to have the declaration of the function available in all files that use them. Also the file that defines the function, so the compiler can cross-check the correctness. You usually do that in an include file so don't have to do it many times.

Other functions (only used in the same compilation unit, and therefore declared as "static") don't need the declaration and the compiler won't bother you with it. Only if you leave the "static" out while you actually did mean such a function. Solution is simple then.

To have more robust and maintainable code, I would suggest to add these to your "CFLAGS": -Wall -Wextra. Then you'll see the compiler will spot a whole lot of potential (or actual) mistakes. That's normal, we're all human, after all. Start by resolving all these warnings until none is left.

> and I’ve no idea how to turn warnings on and off, though I prefer the relevant silence of not having them on

That is like wanting to shoot in your own foot.

> .. an example of that at the many scripts out there some of which I incorporate into “the script”. There are LOTS of warnings about which I can do nothing, I wish I could turn them off.

That's rubbish and nonsense. In my project I have enabled -Wall and -Wextra and a whopping number of 50 additional explicit warnings and there are no warning in the compilation process. I specified -Werror which means any warning found will abort the compilation.

Now I will give you some advice that you will like more I think. GCC has a nasty habit to automatically inline functions, if they're small or only called once. For most code that is beneficial, it will remove the code for the actuall calling of the function, thereby reducing the code size. BUT in our case it also means that the designation (IRAM/IROM) of the function is silently discarded. So what happens is, that you have an IRAM function that calls a function in IROM (because e.g. it's almost never called). The function gets inlined because GCC sees the benefit and you end up with all of the code from the IROM function in IRAM.

Hint: use -fno-inline and choose function to inline manually, where appropriate.

kriegste, 2019-03-06:
I recommend also
-Wmissing-prototypes -Wstrict-prototypes

scargill, 2019-03-06:
Eriksl - it gets more complicated by the moment, I've never touche d any load script - as far as I know, the linker is handled by the MAKEFILE - so now I know I need to put a partition table and pre-init function in user/user_main.c  - here's my MAKEFILE - the LDFLAGS LINE already has most of what you suggest..

LDFLAGS = -nostdlib -Wl,--no-check-sections -u Cache_Read_Enable_New, -u call_user_start -Wl,-static -Wl,-Map,foo.map  -Wl,-gc-sections 

I'll put the missing bits on  on the end as your suggestion....

LDFLAGS = -nostdlib -Wl,--no-check-sections -u Cache_Read_Enable_New, -u call_user_start -Wl,-static -Wl,-Map,foo.map  -Wl,-gc-sections  -Wl,--size-opt

With SDK 2.1 adding that tiny change - Wl,--size-opt" took me down from 7774 text size to 76f4 - so that was worthwhile in itself..

I'll now try the SDK 3.1 and the partition table and pre-init... 

#define SDK_RF_CAL_ADDR 0x3FB000
#define SDK_PHY_DATA_ADDR 0x3FC000
#define SDK_PARAM_ADDR 0x3FD000
#define SPI_FLASH_SIZE_MAP FLASH_SIZE_32M_MAP_1024_1024
#define SDK_PRIV_PARAM_ADDR 0xFC000
#define SYSTEM_PARTITION_CUSTOMER_PRIV_PARAM SYSTEM_PARTITION_CUSTOMER_BEGIN

static const partition_item_t p_table[] = {
{ SYSTEM_PARTITION_BOOTLOADER, 0x0, 0x2000 },
{ SYSTEM_PARTITION_RF_CAL, SDK_RF_CAL_ADDR, 0x1000 },
{ SYSTEM_PARTITION_PHY_DATA, SDK_PHY_DATA_ADDR, 0x1000 },
{ SYSTEM_PARTITION_SYSTEM_PARAMETER, SDK_PARAM_ADDR, 0x3000 },
{ SYSTEM_PARTITION_CUSTOMER_PRIV_PARAM, SDK_PRIV_PARAM_ADDR, 0x1000 },
};

void user_pre_init(void)
{
if(!system_partition_table_regist(p_table, sizeof(p_table)/sizeof(p_table[0]), SPI_FLASH_SIZE_MAP)) {
os_printf("system_partition_table_regist fail\r\n");
while(1);
}
}

Nope, won't fit... so I'm guessing the "load script" wherever that is, needs work AS WELL. I don't know where that is or what it is called I'm afraid.

eriksl, 2019-03-06:
> Actually @kriegste is right - the point is to get something to compile at all, and _THEN_ fit more in :-)
>
> Have you at any point got your code to compile and load using 3.1 regardless of size? Let's [...]

I would suggest that too. Working from a functioning point to a point what's desirable. Some functions might also be functional in IROM, just for testing. Also when you have a project that completely builds, you can inspect the linker map to find out which files have the functions that take up most IRAM and focus there.

eriksl, 2019-03-06:
> I recommend also
> -Wmissing-prototypes -Wstrict-prototypes

As far as I know they're in -Wall or at least in -Wextra (which I'd also recommend).

fvpalha, 2019-03-06:
> What do you mean by that?
>
> Maybe you're running into the same issue I ran into lately. When I generate an image for a 512 KB flash device, the image won't start. If I change it into an "1 MB [...]

Thank you very much @eriksl.

I will try to use of the "bigflash" feature from rboot.

eriksl, 2019-03-06:
@scargill you will have in your "Makefile" "MAKEFILE" or "Makefile" a line that specifies how the project is linked. There will be an option -T followed by a path to a load script (first lines should be similar to what I posted). It's either in the SDK tree and then you will have to make sure the script used is in the SDK v3 directory (and not maybe some leftover SDK v2 tree) or it's somewhere in your project.

If it's in the SDK (the correct version) it should be ok. If it's in your project then you'll have to change it. As far as I know you can't use the vanilla load scripts from the SDK when you're using rboot, BTW.

The load script tells the linker about the IRAM, IROM, DRAM segments and the sections within them. If it's incorrect, your image and data will end up at the wrong addresses (not so probable) or code will end up in the wrong segment (IRAM instead of IROM) and that's important.

Just search for a file that has the text PHDRS in it, if you found it, it most probably is the right file.

How do you have SDK integration BTW? Are you using esp-open-sdk? Is it in your project itself or outside? How do you change between various SDK versions?

eriksl, 2019-03-06:
Also more tips for sizing down on IRAM usage:

- make sure all your functions are "decorated" with a section designation. You can use the defines from Espressif, but I think they're ugly and very confusing. I made two defines myself simply called irom and iram. irom is __attribute__((section(".irom0.text"))) and iram is __attribute__((section(".text"))). If a function doesn't have such an attribute, it WILL end up in IRAM, which is counterintuitive but sadly true.
- make sure gcc doesn't inline irom functions into iram functions (see earlier)
- don't use -O3, it will make the code explose, use -Os
- don't use -O3 (or -O2) at the sime time as -Os. There are some text on the internet that say this is possible and sensible, but it's wrong. Whichever is specified last will win and earlier -O options are discarded.
- if you have a large function in IRAM, consider breaking it into smaller pieces and only make the parts IRAM that really need it
- use optimisation option -mno-target-align (if supported by the compiler version)
- use link time optimisation, this may or may not help (use: -flto=8 -flto-compression-level=0 -fuse-linker-plugin -ffat-lto-objects -flto-partition=max)

Last but not least:

Create a function (in iram) like this:

```
uint32_t user_iram_memory_is_enabled(void);
iram attr_const uint32_t user_iram_memory_is_enabled(void)
{
    return(0);
}

```
This will make sure IRAM is not used for the heap. I believe it's the default, but you never know.

eriksl, 2019-03-06:
@scargill what functions does your project have that need IRAM (and maybe actually not)? Only interrupt handlers, code that run very tightly timed loops and stuff that handles low-level flash need to be in IRAM. IRAM actually consists of two times 32 Kb and the other 32 Kb is used for caching flash memory. Most of the time the flash ("irom") code will already have been fetched into IRAM and will run exactly as IRAM code.

[116 more replies]