espressif/ESP8266_NONOS_SDK #215: 100ms Pause between +IPD

Boodorgad, 2019-03-04:
# 100ms Pause between +IPD  

I see a big difference on IPD-Speed between Server and Client.

Using a **TCP-Server(AT+CIPSERVER)**, the ESP will send out the Data really fast and as described in the AT Instruction Set V3: 

>active mode - ESP8266 will send all the received TCP data instantly to host MCU through
>UART with header “+IPD".                    
                
But if you make a **TCP-Client(AT+CIPSTART)**, the ESP inserts 100ms Pause between IPD blocks?(Snapshot attached) 

The Result is a massiv lower Transfer-Speed for the TCP-Clients.   

I think it's not intend to work like this?

Used Versions
- SDK 3.0.0(e27cf60) 
- AT: 1.7.0.0(Aug 16 2018 00:57:04)

[image]

xcguang, 2019-03-13:
Hi @Boodorgad ，
I did test, but it looks well.
[image]
[image]

xcguang, 2019-03-18:
Hi,
Does it work well?  I will close this issue, and you can reopen it if the problem persists.

closed 2019-03-18 by xcguang