espressif/ESP8266_NONOS_SDK #216: need 512kb Version from AT - Firmware

roliver75, 2019-03-10:
Hello, 

I still have old ESP8266 (ESP-01) with only 521kB Flash.
Could someone help me and build an AT firmware for the chip for flashing? 
Unfortunately I can not, sorry ... 

Thanks in advance greeting oliver

MaJerle, 2019-03-10:
Binary files are part of repository under `bin` folder.

roliver75, 2019-03-10:
Yes please, the chip has 512 kb size

roliver75, 2019-03-11:
I have a old PrezelBoard and I would like to take it for the children again in operation, I need again the AT firmware for a 512Kbit Flash type of the ESP-01 (ESP8266) can someone help me there? (256 + 256) ??

kriegste, 2019-03-11:
256 + 256 won't work with sdk version 3 (too big). Your only option is Non-FOTA, which involves changing the Makefile, ld file and the partition table.

roliver75, 2019-03-11:
and how can I proceed? or can someone prepare that for me?

kriegste, 2019-03-11:
I think you can use this for your partition table.

```
#define SYSTEM_PARTITION_RF_CAL_ADDR                  0x7b000
#define SYSTEM_PARTITION_PHY_DATA_ADDR                0x7c000
#define SYSTEM_PARTITION_SYSTEM_PARAMETER_ADDR        0x7d000
#define SYSTEM_PARTITION_AT_PARAMETER_ADDR            0x76000
#define SYSTEM_PARTITION_SSL_CLIENT_CERT_PRIVKEY_ADDR 0x79000
#define SYSTEM_PARTITION_SSL_CLIENT_CA_ADDR           0x7a000

static const partition_item_t at_partition_table[] = {
 { SYSTEM_PARTITION_CUSTOMER_BEGIN + 1,      0x00000,                                       0x10000},
 { SYSTEM_PARTITION_CUSTOMER_BEGIN + 2,      0x10000,                                       0x66000},
 { SYSTEM_PARTITION_RF_CAL,                  SYSTEM_PARTITION_RF_CAL_ADDR,                   0x1000},
 { SYSTEM_PARTITION_PHY_DATA,                SYSTEM_PARTITION_PHY_DATA_ADDR,                 0x1000},
 { SYSTEM_PARTITION_SYSTEM_PARAMETER,        SYSTEM_PARTITION_SYSTEM_PARAMETER_ADDR,         0x3000},
 { SYSTEM_PARTITION_AT_PARAMETER,            SYSTEM_PARTITION_AT_PARAMETER_ADDR,             0x3000},
 { SYSTEM_PARTITION_SSL_CLIENT_CERT_PRIVKEY, SYSTEM_PARTITION_SSL_CLIENT_CERT_PRIVKEY_ADDR,  0x1000},
 { SYSTEM_PARTITION_SSL_CLIENT_CA,           SYSTEM_PARTITION_SSL_CLIENT_CA_ADDR,            0x1000},
};
```

But I have no clue what the third parameter of system_partition_table_regist must be like for 512 kB NON-FOTA...
I mean, the possible values are

```
enum flash_size_map {
    FLASH_SIZE_4M_MAP_256_256 = 0,  /**<  Flash size : 4Mbits. Map : 256KBytes + 256KBytes */
    FLASH_SIZE_2M,                  /**<  Flash size : 2Mbits. Map : 256KBytes */
    FLASH_SIZE_8M_MAP_512_512,      /**<  Flash size : 8Mbits. Map : 512KBytes + 512KBytes */
    FLASH_SIZE_16M_MAP_512_512,     /**<  Flash size : 16Mbits. Map : 512KBytes + 512KBytes */
    FLASH_SIZE_32M_MAP_512_512,     /**<  Flash size : 32Mbits. Map : 512KBytes + 512KBytes */
    FLASH_SIZE_16M_MAP_1024_1024,   /**<  Flash size : 16Mbits. Map : 1024KBytes + 1024KBytes */
    FLASH_SIZE_32M_MAP_1024_1024,    /**<  Flash size : 32Mbits. Map : 1024KBytes + 1024KBytes */
    FLASH_SIZE_32M_MAP_2048_2048,    /**<  attention: don't support now ,just compatible for nodemcu;
                                           Flash size : 32Mbits. Map : 2048KBytes + 2048KBytes */
    FLASH_SIZE_64M_MAP_1024_1024,     /**<  Flash size : 64Mbits. Map : 1024KBytes + 1024KBytes */
    FLASH_SIZE_128M_MAP_1024_1024     /**<  Flash size : 128Mbits. Map : 1024KBytes + 1024KBytes */
};
```

Maybe Espressif can shed some light on this for you...

roliver75, 2019-03-11:
??

kriegste, 2019-03-12:
I've tried to compile an AT firmware version for 4M modules using boot map 2. It is a bit strange, but my test module boots ok (at least it says "ready"... I did not try to do more).

Files and addresses you need to flash them at:

0x00000
https://github.com/kriegste/ESP8266_NONOS_SDK/blob/master/bin/at/eagle.flash.bin?raw=true

0x10000
https://github.com/kriegste/ESP8266_NONOS_SDK/blob/master/bin/at/eagle.irom0text.bin?raw=true

0x7C000
https://github.com/kriegste/ESP8266_NONOS_SDK/blob/master/bin/esp_init_data_default_v08.bin?raw=true

xcguang, 2019-03-13:
@roliver75 , 512KB flash is too small to the latest AT firmware.

roliver75, 2019-03-13:
Have everibody a Oldenburg smal Version of This Firmware?

roliver75, 2019-03-13:
Oldenburg = old

davydnorris, 2019-03-13:
> 
> 
> Oldenburg = old

Just change back to older branches in this repo and it will be there