espressif/ESP8266_NONOS_SDK #218: Not getting STATION_WRONG_PASSWORD any more

kriegste, 2019-03-13:
I've tried by deliberately setting a wrong station password. The module tries to reconnect over and over again and wifi_station_get_connect_status() always reports back STATION_CONNECTING.

That is correct, however, in my opinion it makes more sense to return STATION_WRONG_PASSWORD.

Otherwise the user does not know why there is no connection established... And not using it would render the struct member STATION_WRONG_PASSWORD useless anyway.

xcguang, 2019-03-14:
Hi,
Maybe it is connecting when you call wifi_station_get_connect_status(). You can get more information when getting EVENT_STAMODE_DISCONNECTED. Or you can disable to reconnect by calling `wifi_station_set_reconnect_policy` and then get status.

kriegste, 2019-03-14:
I suspect that, too. But disabling reconnect only for this is a bit... crude. I want it to reconnect in all situations and I don't want to introduce new points of failure.

I've tried calling wifi_station_get_connect_status() in the event handler as well. Same result. In the event handler I can get event_info.disconnected.reason (== REASON_AUTH_EXPIRE) but that is not the same. Unfortunately there is no event_info.disconnected.status... that would be great.

kriegste, 2019-03-14:
If I set
wifi_station_set_reconnect_policy(false);
at start, and call wifi_station_get_connect_status() in the event handler, it still returns STATION_CONNECTING.

I know that it worked before and I got STATION_WRONG_PASSWORD in earlier versions of the SDK... even with automatic reconnect.

xcguang, 2019-03-14:
Could you query which version it works well?

kriegste, 2019-03-14:
I just tried to compile my project against the current 2.2. Same behavior. Unfortunately I cannot compile against the 2.1. Too much has changed.

kriegste, 2019-05-24:
By the way, we do not get STATION_NO_AP_FOUND returned, either.

Yes, wifi_station_set_reconnect_policy is set to false.

Kreetcha, 2019-11-26:
I''m also having this exact issue getting only STATION_CONNECTING when attempting to poll wifi status inside a timer callback.  I'm trying to setup a polling timer to check the status of my station connection and handle myself (without auto reconnect).

I am testing with wrong ssid/pwd and found no matter what i get the same returned, has any one found a solution to this? 

According to api docs, I set wifi_station_set_reconnect_policy(false)  after wifi is in station mode
inside user_init().
I also set wifi_set_event_handler_cb (docs say this is required after setting reconnect policy). 
I still get STATION_CONNECTING no matter how much time I add between os_timer polls.

kriegste, 2020-02-26:
It is really sad that the SDK does not behave like the documentation suggests. This makes WiFi connection error handling quite shaky... I'd love to see a fix here. I know that only "critical" bugs are to be corrected, but in my opinion WiFi connection problems are indeed critical ones in a WiFi SOC SDK...

Kreetcha, 2020-02-27:
I believe this issue definitely falls under the category of a critical bug.
This issue breaks the core functionality of the esp8266, and there is no workaround.

Fixing this issue so the api call (wifi_station_get_connect_status in station mode) acts as the documentation describes is not a new feature, as it is currently breaking the core functionality of the esp8266 (wifi).

st0ff3r, 2020-03-03:
still with latest master branch?

kriegste, 2020-03-03:
Computer says yes. Problem still exists...

kriegste, 2020-07-13:
Any news on this issue?

Kreetcha, 2020-07-30:
Pretty disheartening that it is taking well over year to fix such a critical issue :(
The extreme lack of communication doesn't help either.

kriegste, 2020-09-18:
Nothing?

I noticed that when I change the WiFi, the event handler gets called with EVENT_STAMODE_DISCONNECTED and the NEW WiFi SSID is in event_info.disconnected.ssid.
It was my understanding that event_info.disconnected.ssid holds the SSID of the OLD WiFi. The one before changing the config.
This may or may not have something to do with the problem I mentioned above. I guess the relevant info gets overwritten at some point before the event fires.