espressif/ESP8266_NONOS_SDK #219: My solution to mbedtls (https) problems

kriegste, 2019-03-14:
This is a hint for everyone having problems with mbedtls and just wants to use the client mode. I had problems, both with connecting to https servers (I just got handshake errors all the time) and memory usage (the lib was simply too big).
So I tried several options and ended up with this:

In third_party/include/mbedtls/config_esp.h

I uncommented (enabled)
#define MBEDTLS_DEPRECATED_REMOVED
#define MBEDTLS_CAMELLIA_SMALL_MEMORY
#define MBEDTLS_ENTROPY_FORCE_SHA256

and commented (disabled)
#define MBEDTLS_KEY_EXCHANGE_RSA_PSK_ENABLED
#define MBEDTLS_PK_PARSE_EC_EXTENDED
#define MBEDTLS_ERROR_STRERROR_DUMMY
#define MBEDTLS_GENPRIME
#define MBEDTLS_PK_RSA_ALT_SUPPORT
#define MBEDTLS_PKCS1_V21
#define MBEDTLS_SSL_ALL_ALERT_MESSAGES
#define MBEDTLS_SSL_CBC_RECORD_SPLITTING
#define MBEDTLS_SSL_PROTO_TLS1
#define MBEDTLS_SSL_PROTO_TLS1_1
#define MBEDTLS_SSL_SERVER_NAME_INDICATION
#define MBEDTLS_AESNI_C
#define MBEDTLS_MD5_C
#define MBEDTLS_PEM_PARSE_C
#define MBEDTLS_PEM_WRITE_C
#define MBEDTLS_SHA1_C
#define MBEDTLS_SHA512_C
#define MBEDTLS_SSL_SRV_C
#define MBEDTLS_VERSION_C
#define MBEDTLS_X509_CRL_PARSE_C
#define MBEDTLS_X509_CSR_PARSE_C
#define MBEDTLS_X509_CREATE_C
#define MBEDTLS_X509_CRT_WRITE_C
#define MBEDTLS_X509_CSR_WRITE_C
#define MBEDTLS_XTEA_C

This also disables the outdated and unsafe TLS 1.0 and 1.1 protocols.

jackinto, 2019-04-02:
Hi kriegste,

Do you know how can I make espconn_secure make use of mbedtls?

I am having memory and handshake issues and was hoping this could help.

kriegste, 2019-04-02:
If linked with libmbedtls.a, all espconn_secure* functions will use mbedtls.

jackinto, 2019-04-02:
Thanks, sorry I am a bit new on this, would you mind telling how I check whether it is being linked with libmbedtls.a?

loboris, 2019-04-23:
@kriegste
The changes you've made, breaks certificates verification.
You cannot connect to any SSL server with `AT+CIPSSLCCONF=2`.
If you don't need certificates, it is fine, the code size is ~20K smaller and it works.

kriegste, 2019-04-24:
That is certainly possible. I do not use this feature and just stripped everything down to a minimum working configuration. Also the SSL server part is gone.

maverickchongo, 2019-04-29:
@loboris 
Do you know the specific defines that break certificate verification?
I need to reduce the size of mbedtls to use TLS for my mqtt client but need to use certificates...

kriegste, 2019-04-29:
I wonder if it would be enough to just gererate a hash (with a built-in hash function) of the certificate and verify only that?

loboris, 2019-04-29:
@maverickchongo
I've just published on GitHub my new version of ESP8266 AT Firmware.
It has some changes to the mbedtls which can be used also for some other applications.
SSL/TLS now works without any issues, certificates can be used (I've even added support for loading CA certificates into RAM), downloads of couple of MB files works ...
I'll give some more informations in the next couple of days.

[ESP8266_AT_LoBo](https://github.com/loboris/ESP8266_AT_LoBo)

maverickchongo, 2019-04-29:
@loboris 
Thanks loboris.
One more question, I am using esp-open-sdk to build the sdk, but I need to find a way to rebuild it to include the changes on config_esp.h without it removing/cloning the whole esp sdk and overiding my changes...
I tried to find out by looking at the make file on esp-open-sdk but couldn't figure it out.
Would you know how to do it?
Unfortunately the project I am working on is setup to use the esp-open-sdk and I can't change it...
Thanks in advance

loboris, 2019-04-29:
@maverickchongo
The changes I've made are in _mbedtls_ and are based on the fact that the latest Non_OS SDK can leave quite a lot of free RAM for user application (~53KB).
I'm not shure if it is possible in _esp-open-sdk_, and, esp-open-sdk uses axTLS, not mbedtls, so, I'm afraid, I cant help you, sorry.

maverickchongo, 2019-04-29:
@loboris
Thanks
Correct me if I am wrong, but I think esp-open-sdk can use either, axTLS (a.c.a ssl) or mbedtls (it creates libssl.a and libmbedtls.a), when compiling the app you can choose whether to include the ssl or mbedtls library and then the esp_secure API will be based on that one.