espressif/ESP8266_NONOS_SDK #220: Please suggest any up-to-date toolchain on Windows

valkuc, 2019-03-14:
Hello. I'm currently using UDK from https://github.com/CHERTS/esp8266-devkit. It fully suits my needs and I'm actually don't have any problems working with it, but it looks quite outdated. Just interesting, is there any up-to-date toolchain on Windows to work with ESP8266 NON-OS SDK?

kriegste, 2019-03-14:
Have you tried
https://dl.espressif.com/dl/xtensa-lx106-elf-win32-1.22.0-92-g8facf4c-5.2.0.tar.gz

valkuc, 2019-03-15:
No, have not try yet. Thanks for the link, will try it.

valkuc, 2019-03-15:
I was able to build my project with this toolchain and latest NON_OS_SDK from release/v2.2.x branch. But I have faced an issue unpacking tar.gz file on Windows. The problem is in linux-style links in archive. When unpacking it on Windows (for example, using 7-zip archive manager) the linked files got broken (have size 0). The solution was to extract archive on Linux using `tar -h -zxf xtensa-lx106-elf-win32-1.22.0-92-g8facf4c-5.2.0.tar.gz` command and then copy extracted files to Windows. So, my suggestion is to pack Windows toolchain in ZIP archive.