espressif/ESP8266_NONOS_SDK #221: Getting sdk 3.0 to work with 1MB flash

MariusLer, 2019-03-14:
I am having trouble getting this version to work with a 1MB flash. It's my understanding that it should be possible. I'm trying to use the default AT example. I disabled smart config.

When compiling i get the following error;
`.output/eagle/release/image/eagle.app.v6.out section .irom0.text' will not fit in region irom0_0_seg'
`
At first i had these partition table settings as suggested in https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/documents/EN/%20Partition%20Table.md

```
#define SYSTEM_PARTITION_OTA_SIZE			                    0x6A000  // upper limit for 
#define SYSTEM_PARTITION_OTA_2_ADDR		                    0x81000
#define SYSTEM_PARTITION_RF_CAL_ADDR		                    0xfb000
#define SYSTEM_PARTITION_PHY_DATA_ADDR	                            0xfc000
#define SYSTEM_PARTITION_SYSTEM_PARAMETER_ADDR                0xfd000
#define SYSTEM_PARTITION_AT_PARAMETER_ADDR                      0x7d000
#define SYSTEM_PARTITION_SSL_CLIENT_CERT_PRIVKEY_ADDR           0x7c000
#define SYSTEM_PARTITION_SSL_CLIENT_CA_ADDR                     0x7b000
#define SYSTEM_PARTITION_WPA2_ENTERPRISE_CERT_PRIVKEY_ADDR      0x7a000
#define SYSTEM_PARTITION_WPA2_ENTERPRISE_CA_ADDR                0x79000

static const partition_item_t at_partition_table[] = {
    { SYSTEM_PARTITION_BOOTLOADER, 						0x0, 												0x1000},
    { SYSTEM_PARTITION_OTA_1,   						0x1000, 											SYSTEM_PARTITION_OTA_SIZE},
    { SYSTEM_PARTITION_OTA_2,   						SYSTEM_PARTITION_OTA_2_ADDR, 						SYSTEM_PARTITION_OTA_SIZE},
    { SYSTEM_PARTITION_RF_CAL,  						SYSTEM_PARTITION_RF_CAL_ADDR, 						0x1000},
    { SYSTEM_PARTITION_PHY_DATA, 						SYSTEM_PARTITION_PHY_DATA_ADDR, 					0x1000},
    { SYSTEM_PARTITION_SYSTEM_PARAMETER, 				SYSTEM_PARTITION_SYSTEM_PARAMETER_ADDR, 			0x3000},
    { SYSTEM_PARTITION_AT_PARAMETER, 					SYSTEM_PARTITION_AT_PARAMETER_ADDR, 				0x3000},
	{ SYSTEM_PARTITION_SSL_CLIENT_CERT_PRIVKEY, 		SYSTEM_PARTITION_SSL_CLIENT_CERT_PRIVKEY_ADDR, 		0x1000},
	{ SYSTEM_PARTITION_SSL_CLIENT_CA, 					SYSTEM_PARTITION_SSL_CLIENT_CA_ADDR, 				0x1000},
#ifdef CONFIG_AT_WPA2_ENTERPRISE_COMMAND_ENABLE
	{ SYSTEM_PARTITION_WPA2_ENTERPRISE_CERT_PRIVKEY, 	SYSTEM_PARTITION_WPA2_ENTERPRISE_CERT_PRIVKEY_ADDR,	0x1000},
    { SYSTEM_PARTITION_WPA2_ENTERPRISE_CA, 				SYSTEM_PARTITION_WPA2_ENTERPRISE_CA_ADDR, 			0x1000},
#endif
};
```

After this I also tried the following, but i was still getting the same error.

```
#define EAGLE_FLASH_BIN_ADDR                                    (SYSTEM_PARTITION_CUSTOMER_BEGIN+1)
#define EAGLE_IROM0TEXT_BIN_ADDR                                (SYSTEM_PARTITION_CUSTOMER_BEGIN+2)
#define SYSTEM_PARTITION_RF_CAL_ADDR		                    0xfb000
#define SYSTEM_PARTITION_PHY_DATA_ADDR	                        0xfc000
#define SYSTEM_PARTITION_SYSTEM_PARAMETER_ADDR                  0xfd000
#define SYSTEM_PARTITION_AT_PARAMETER_ADDR                      0x7d000
#define SYSTEM_PARTITION_SSL_CLIENT_CERT_PRIVKEY_ADDR           0x7c000
#define SYSTEM_PARTITION_SSL_CLIENT_CA_ADDR                     0x7b000
#define SYSTEM_PARTITION_WPA2_ENTERPRISE_CERT_PRIVKEY_ADDR      0x7a000
#define SYSTEM_PARTITION_WPA2_ENTERPRISE_CA_ADDR                0x79000

static const partition_item_t at_partition_table[] = {
    { EAGLE_FLASH_BIN_ADDR, 	                        0x00000,                                            0x10000},
    { EAGLE_IROM0TEXT_BIN_ADDR,                         0x10000,                                            0x60000},
    { SYSTEM_PARTITION_RF_CAL,  						SYSTEM_PARTITION_RF_CAL_ADDR, 						0x1000},
    { SYSTEM_PARTITION_PHY_DATA, 						SYSTEM_PARTITION_PHY_DATA_ADDR, 					0x1000},
    { SYSTEM_PARTITION_SYSTEM_PARAMETER, 				SYSTEM_PARTITION_SYSTEM_PARAMETER_ADDR, 			0x3000},
    { SYSTEM_PARTITION_AT_PARAMETER, 					SYSTEM_PARTITION_AT_PARAMETER_ADDR, 				0x3000},
    { SYSTEM_PARTITION_SSL_CLIENT_CERT_PRIVKEY, 		SYSTEM_PARTITION_SSL_CLIENT_CERT_PRIVKEY_ADDR, 		0x1000},
    { SYSTEM_PARTITION_SSL_CLIENT_CA, 					SYSTEM_PARTITION_SSL_CLIENT_CA_ADDR, 				0x1000},
#ifdef CONFIG_AT_WPA2_ENTERPRISE_COMMAND_ENABLE
	{ SYSTEM_PARTITION_WPA2_ENTERPRISE_CERT_PRIVKEY, 	SYSTEM_PARTITION_WPA2_ENTERPRISE_CERT_PRIVKEY_ADDR,	0x1000},
    { SYSTEM_PARTITION_WPA2_ENTERPRISE_CA, 				SYSTEM_PARTITION_WPA2_ENTERPRISE_CA_ADDR, 			0x1000},
#endif
};
```

kriegste, 2019-03-14:
Are you using the (bleeding edge) master branch of this SDK? Should be no problem.

MariusLer, 2019-03-15:
Thank you for your answer. I was using the SDK 3.0 release. I pulled the newest release from here and made the changes you suggested. It works perfectly now. I needed a newer SDK to be able to use TLS 1.2 which was not supported by SDK which was installed.

closed 2019-03-18 by MariusLer