espressif/ESP8266_NONOS_SDK #223: Allow AT+CIPSSLSIZE=8192

MariusLer, 2019-03-19:
I had trouble connecting the esp8266 to a server where client authentication was required, using AT commands.
I solved this problem by setting the SSL buffer size to 8192 by editing the default AT example and adding the following line to the user_init function.
``
espconn_secure_set_size(0x01, 8192);
``
After adding this I was able to connect to the server. I don't see any reason why this should not be possible to do with the AT+CIPSSLSIZE command. It currently only accepts the sizes 2048 and 4096, and returns error when trying to set it to 8192.

kriegste, 2019-03-19:
The size needs to be greater than the certificate's size. That is what I noticed. Many modern certificates are bigger.
https://www.ssllabs.com/ssltest/
also shows the size of the certificate.

xcguang, 2019-03-28:
Hi,
8192 is too larger for the common AT to stably run. If this requirement is necessary, please add a customer command to set it, and test it for all functions you use.

MariusLer, 2019-04-29:
Thank you for the response. I can confirm that the AT software is stable when using the ESP8266 as a TLS-client with the buffer size set to 8192. Running the AT+SYSRAM commands says that approximately 40kB is still free after the connection to the server has been established.

closed 2019-04-29 by MariusLer

aminismail, 2019-10-21:
Yes, I agree with the comments above. Please consider adding the capability to specify a SSLSIZE of 8192.  I can also confirm that it is stable. Without this it is impossible to use the ESP8266 as an MQTT client using TLS and the NONOS SDK.  Once of the important applications of the ESP8266 is in IoT and now most MQTT servers allow only TLS connections. The ESP8266 is a great product and with this capability it will be even better. Thank you.