espressif/ESP8266_NONOS_SDK #224: Commit "fix(wifi): Prohibit sending probe request when passive scan mode" causes crashing with frequent SPI flash writes

someburner, 2019-03-20:
Commit e2ac7e62f0ee74b40ddec347b1025cef083cda5c on `release/2.2.x` branch results in crashing when doing a large amount of SPI Flash writes. This was found on a project I'm working on and cannot provide a source. If I have to I can try writing a test project to produce the crash, but I was hoping someone could at least review that commit and see what changed with regards to SPI Flash and maybe spot a regression.

Essentially what is happening in the code is binary data is being sent over MQTT and written directly to an unused region of SPI Flash using `spi_flash_write`. Before this commit it does not crash, after this commit it crashes anywhere from just a few calls to `spi_flash_write` to around 60 calls. I am not 100% sure that it is SPI Flash triggering the crash, but am pretty certain because commenting out the call to `spi_flash_write` allows the transfer to complete without crashing (no data is written), so it seems unlikely to be a WiFi/MQTT issue.

davydnorris, 2019-03-21:
That almost sounds like some weird interaction between reading code in ROM and storing your data. Are you able to work out which MQTT function you're in and make that iRAM temporarily? Basically eliminate reading code while writing flash and see if that's the issue?

someburner, 2019-03-21:
Given that this code has been stable for over a year and works fine for all SDKs up until that commit, I am inclined to think it is a problem with the SDK. Of course it could be that they fixed a bug that was allowing my code to not crash this whole time. But then how am I supposed to know?

That is a fair suggestion and if no one at Espressif can help I will probably start with that. But I'm hoping they can go into their repo, see what changed in that commit, and at least give me pointers on troubleshooting. That would eliminate a lot of guesswork for me :)

kriegste, 2019-03-21:
Do you call spi_flash_write() 60 times in one go, without returning?

I do OTA firmware updates manually using spi_flash_write(), 1024 bytes in each go (then returning). This happens up to about 500 times (500 kB) total until the image is complete. No problems with the current SDK version.

someburner, 2019-03-21:
Yeah it goes like this:

```c
MQTT_OnData() {
    // malloc a struct, put data in struct
    system_os_post( *ota_data ); // post to event handler
}

// event handler
Task_Handler( e ) {
    // get ota_data pointer from e
    spi_flash_write(ota_data); // write
}
```

And I should say I don't actually use `spi_flash_write`, but rather `platform_flash_write` [from NodeMCU](https://github.com/nodemcu/nodemcu-firmware/blob/b41667b8374c2f27365cc5796a21a4a1fa78509a/app/platform/common.c#L88) which does things like feeding wdt and ensuring alignment.

someburner, 2019-03-23:
Taking the suggestion, I changed my .ld file from:

`*libpp.a:*(.literal .text)`

to

`*libpp.a:*(.literal .text .text.*)`

And no more crash. But unfortuantely that eats up about 15K of IRAM.

So it seems something in `libpp.a` is not mixing well with my code.

xcguang, 2019-03-23:
Thanks for your report, we will reproduce and fix it ASAP.

someburner, 2019-03-23:
Thanks. To summarize what I found:

- Before that commit, my project has `8244` left for iram, no crashing
- After that commit, changing nothing in my project, it jumps to `23568` left available for iram, but OTA crashes
- Putting all the `.text.*` methods back into iram by editing the `.ld` file and it goes back to `8468` left for iram, and this no longer crashes.

I'm using ld file from NodeMCU that defaults to everything in irom and puts stuff in iram [like this](https://github.com/nodemcu/nodemcu-firmware/blob/11592951b90707cdcb6d751876170bf4da82850d/ld/nodemcu.ld#L106).

kriegste, 2019-03-23:
What happens if you use the standard ld file?

If it works try adding the line for libc

*libc.a:(.literal .text .literal.* .text.*)

and see how much iram that leaves you.

xcguang, 2019-03-27:
Hi @someburner ,
`*libpp.a:*(.literal .text)` is your modification in ld? I can't find it in SDK ld.

someburner, 2019-03-27:
Yeah it's because I'm using the [NodeMCU .ld file](https://github.com/nodemcu/nodemcu-firmware/blob/master/ld/nodemcu.ld). But in any case, it appears some methods are no longer getting put into the text section. I'm wondering if that is intended or not? Since if I put all those methods into IRAM I no longer get crashes, but also don't benefit from decreased iram usage. This is okay but it just seems strange that I would have to move things into iram myself.

xcguang, 2019-03-28:
Hi @someburner ，
Could you refer to the ld file in SDK.
For smaller binary, we add `-ffunction-sections -fdata-sections` flags when compiling libs. The issue is https://github.com/espressif/ESP8266_NONOS_SDK/issues/209.