espressif/ESP8266_NONOS_SDK #226: osapi.h not using internal versions of strcat and strchr

davydnorris, 2019-03-24:
As I was checking my link map, I found that some of the code was using ets_strchr from libmain, but my code was using the libc strchr.

Investigating further I found that osapi.h declares both strcat and strchr to be the standard versions - is there a reason for this?
```
#define os_strcat strcat
#define os_strchr strchr
```

I have added the following to osapi.h and everything still compiles and appears to run fine

```
char *ets_strchr(const char *str, unsigned int c);
char *ets_strcat (char *to, const char *from);

#define os_strcat ets_strcat
#define os_strchr ets_strchr
```

kriegste, 2019-03-24:
Remember to recompile lwip to get rid of the standard function completely.

davydnorris, 2019-03-24:
Actually the current shipping version of lwIP is good - the only function it's using from libc is memchr(), which you would think should be easy to get rid of.

The instances I found were from my code, and were because of the osapi.h file definitions.

eriksl, 2019-05-15:
What do you expect from using the internal versions? These functions are just a few bytes, maybe one hundred. It might be interesting if it appears that access to (internal) ROM is much faster than access to FLASH, but as it's cached in (internal) IRAM, it doesn't matter really. And then, if it matters for performance, how many times in a row are they called?

I got rid of them all (the internal string functions), just using the gcc xtensa optimised versions, they work just as well. Flash memory is something I have plenty of, IRAM and DRAM not, so I tend to optimise to those instead of code size.