espressif/ESP8266_NONOS_SDK #231: wifi_station_scan() - err3, exceeded max time value

sdk156, 2019-04-13:
Hi,
 I am using the latest version of NON OS SDK downloaded from [here](https://github.com/espressif/ESP8266_NONOS_SDK). When wifi_station_scan() api is used to scan for APs in active scan mode, several lines of err3, exceeded max time value are printed on the screen. I have changed the min and max values of scan time per channel. But nothing helped me out. I am using the wifi scanning code from official espressif site from [here](https://www.espressif.com/en/hidden/support/explore/sample-codes) and ran active scan explicitly.
[image]

maverickchongo, 2019-05-29:
@sdk156 
I am experiencing the same issue, have you found the root cause?

R4gnarLothbrok, 2021-01-26:
I am having the same issue, did you find the cause?