espressif/ESP8266_NONOS_SDK #232: New libphy: Signal strength in scan results much lower; IRAM usage went up 928 bytes

kriegste, 2019-04-17:
After updating to the new phy lib, I noticed that the reported signal strength in the wifi scan results went down by 50. I had -30 before and now -80. Also, some of the weaker access points in my neighborhood disappeared from the list.

IRAM usage went up by 928 bytes. Is that really necessary or a mistake?

xcguang, 2019-04-18:
Hi,
1. The signal strength issue have be reproduced. 
2. Memory usage is occupied more 244 bytes.

Now RF engineers are dealing with them. I will update it here if there is any progress. Thank you very much for your report.

kriegste, 2019-04-18:
Thanks for analyzing. In the mean time I noticed that a module placed further away from my access point cannot connect at all.

Please also try connecting to the module in soft ap mode. There are problems, too.

closed 2019-04-23 by wujiangang