espressif/ESP8266_NONOS_SDK #235: Problem facing when Opening SSL Socket in ESP8266..

harikrishna483, 2019-04-23:
Hi this is Harikrishna..
I am facing problem in SSL connection with Azure IotHUb, while I am using AT Commands..I have successfully updated the AT Command Firmware in ESP8266..and that is latest one V1.7 i think..i followed this process
AT+CMUX=1
AT+CIPSSLSIZE=4096
AT+CIPSSLCCONF=2
AT+CIPSTART="TCP","www.google.com",443
I am getting response as CONNECT OK and the connection is fine..
And Then 
AT+CIPSTART="TCP","www.google.com",80
For this one also getting CONNECT OK response. 

But when open the SSL Connection I getting error

AT+CIPSTART="SSL","www.google.com",443
I am getting response as ERROR ,CLOSED
I have tried with different types of server address with TCP all are working but with SSL I am unable to connect any server..
Please any one helps this issue..

loboris, 2019-04-23:
If using `AT+CIPSSLCCONF=2` you must use the certificate file valid for the server you are connecting to.
You can use `AT+CIPSSLCCONF=0` to connect to the server without using certificate verification.
For _"www.google.com"_ you should use the certificate file with the following content:
```
-----BEGIN CERTIFICATE-----
MIIEXDCCA0SgAwIBAgINAeOpMBz8cgY4P5pTHTANBgkqhkiG9w0BAQsFADBMMSAw
HgYDVQQLExdHbG9iYWxTaWduIFJvb3QgQ0EgLSBSMjETMBEGA1UEChMKR2xvYmFs
U2lnbjETMBEGA1UEAxMKR2xvYmFsU2lnbjAeFw0xNzA2MTUwMDAwNDJaFw0yMTEy
MTUwMDAwNDJaMFQxCzAJBgNVBAYTAlVTMR4wHAYDVQQKExVHb29nbGUgVHJ1c3Qg
U2VydmljZXMxJTAjBgNVBAMTHEdvb2dsZSBJbnRlcm5ldCBBdXRob3JpdHkgRzMw
ggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQDKUkvqHv/OJGuo2nIYaNVW
XQ5IWi01CXZaz6TIHLGp/lOJ+600/4hbn7vn6AAB3DVzdQOts7G5pH0rJnnOFUAK
71G4nzKMfHCGUksW/mona+Y2emJQ2N+aicwJKetPKRSIgAuPOB6Aahh8Hb2XO3h9
RUk2T0HNouB2VzxoMXlkyW7XUR5mw6JkLHnA52XDVoRTWkNty5oCINLvGmnRsJ1z
ouAqYGVQMc/7sy+/EYhALrVJEA8KbtyX+r8snwU5C1hUrwaW6MWOARa8qBpNQcWT
kaIeoYvy/sGIJEmjR0vFEwHdp1cSaWIr6/4g72n7OqXwfinu7ZYW97EfoOSQJeAz
AgMBAAGjggEzMIIBLzAOBgNVHQ8BAf8EBAMCAYYwHQYDVR0lBBYwFAYIKwYBBQUH
AwEGCCsGAQUFBwMCMBIGA1UdEwEB/wQIMAYBAf8CAQAwHQYDVR0OBBYEFHfCuFCa
Z3Z2sS3ChtCDoH6mfrpLMB8GA1UdIwQYMBaAFJviB1dnHB7AagbeWbSaLd/cGYYu
MDUGCCsGAQUFBwEBBCkwJzAlBggrBgEFBQcwAYYZaHR0cDovL29jc3AucGtpLmdv
b2cvZ3NyMjAyBgNVHR8EKzApMCegJaAjhiFodHRwOi8vY3JsLnBraS5nb29nL2dz
cjIvZ3NyMi5jcmwwPwYDVR0gBDgwNjA0BgZngQwBAgIwKjAoBggrBgEFBQcCARYc
aHR0cHM6Ly9wa2kuZ29vZy9yZXBvc2l0b3J5LzANBgkqhkiG9w0BAQsFAAOCAQEA
HLeJluRT7bvs26gyAZ8so81trUISd7O45skDUmAge1cnxhG1P2cNmSxbWsoiCt2e
ux9LSD+PAj2LIYRFHW31/6xoic1k4tbWXkDCjir37xTTNqRAMPUyFRWSdvt+nlPq
wnb8Oa2I/maSJukcxDjNSfpDh/Bd1lZNgdd/8cLdsE3+wypufJ9uXO1iQpnh9zbu
FIwsIONGl1p3A8CgxkqI/UAih3JaGOqcpcdaCIzkBaR9uYQ1X4k2Vg5APRLouzVy
7a8IVk6wuy6pm+T7HT4LY8ibS5FEZlfAFLSW8NwsVz9SBK2Vqn1N0PIMn5xA6NZV
c7o835DLAFshEWfC7TIe3g==
-----END CERTIFICATE-----
```
You should convert it to the format used by ESP8266 using `makefile.sh` in `tools` directory, and flash to the CA certificate sector in Flash.

There are other issues with SSL, like problems with downloading larger files due to insuficient SSL buffer size.
I've modified the AT firmware a little, so that SSL buffer can be 16384 bytes large. The latest NonOS SDK version has enough free RAM to run with such a large buffer without issues.
With 16384 bytes SSL buffer I was able to download multi MB files using https with server certificate verification without any issue.

OzEdri, 2021-07-21:
@harikrishna483 Did you manage to solve this issue?
I'm having the same issue when trying to connect to Firebase (also by Google) with the AT+CIPSTART="SSL" command.
That said, I'm able to connect to other SSL sites with that command, including google.com