espressif/ESP8266_NONOS_SDK #237: [Vulnerability] found - Question

Matheus-Garbelini, 2019-04-26:
Hello, 

I've found a DoS Vulnerability and was wondering what is the correct procedure to follow (as I'm not sure if this would fit as a new CVE). 
Would I need to report everything in detail here or directly to Espressif site?([https://www.espressif.com/en/company/contact-extra/technical-inquiries-software](url))

The issue in question appears to affect all esp8266 using NONOS SDK up to the latest version as in this repository. This also includes the current SDK version used in the esp8266 arduino branch.

kriegste, 2019-04-26:
Try bugbounty@espressif.com

Ivyares, 2019-04-28:
The best way is to test whether the ESP32 has the same issue. :P  If it is, then maybe you can win the bug bounty of ESP32? Not sure if it is a "security" issue..
Anyway, I think you can try the email bugbounty@espressif.com.

Matheus-Garbelini, 2019-04-28:
@Ivyares Thank you, I'm not sure about ESP32 yet, but I'm doing tests with it. Rarely it crashes and as I can't manually reproduce the crash, I'm not even sure if its my code or not.

For ESP8266, I'll send espressif an email providing the results and the test code this week so they can reproduce it.

closed 2019-04-28 by Matheus-Garbelini