espressif/ESP8266_NONOS_SDK #245: ESP-12S and AT 1.7.0 will not start

WanaGo, 2019-06-13:
Hello,
I have been using AT 1.6.0 for some time, but I just tried to load 1.7.0 and I cannot get it to start up. I get:
mismatch map 5, spi_size_map 4
system_partition_table_regist fail

When using the new 1.7.0 files and the memory locations specified in the md file.

Can anyone spot what the issue is in the images below?

The md file does not explain what 32Mbit-C1 is vs 32Mbit, it also doesnt give info on what User1 vs User2 is, or what the esp_init_data_default files are v05 vs v08. Which one do I need to use for the ESP-12S?

This one works:
[image]

This one does not work:
[image]

Thanks

WanaGo, 2019-06-13:
The md file also talks about Boot mode vis non-boot mode, which it doesnt explain, and the files listed in non-boot mode do not even exist in the download. eagle.flash.bin and eagle.irom0text.bin
Incredibly confusing.

WanaGo, 2019-06-13:
I downloaded the 1.6.3 AT Command firmware instead, and in there is the 512+512 option, so I tried that, and that works successfully.... What am I doing wrong?

[image]

tsvehagen, 2019-09-07:
Have a look at https://github.com/espressif/ESP8266_NONOS_SDK/issues/179. That might solve your problem.