espressif/ESP8266_NONOS_SDK #246: When I use spi to use the master and slave to send data to each other, the data received is incorrect.

MrFanLee, 2019-06-27:
like this：
master send data：
[image]
slaver received data：
[image]
slaver send data：
[image]
master received data：
[image]
So why can't I send 0x35343332 and receive 0x35343332?

xcguang, 2019-10-19:
What's the length of the address and command in the master?

MaJerle, 2019-10-20:
Its called endianess issue. Master has different than slave. You should always send byte by byte when communicating between devices.