espressif/ESP8266_NONOS_SDK #251: Null-pointer exception in wifi libs

ilyakorostelev, 2019-07-16:
I'm trying to change WiFi channel by calling wifi_softap_set_config() and I get an exception in memcmp() function (especially in STA+AP mode):

```
state: 0 -> 2 (b0)
bcn 0
del if1
Fatal exception 28(LoadProhibitedCause):
epc1=0x4000deea, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000033, depc=0x00000000
```

I found, memcmp() called from hostap_input(): 

`memcmp(ptr1, ptr2, 6)`

ptr1 - I think, mac address
ptr2=0x33
len=6

What advise do you have?

eriksl, 2019-12-05:
Nope. Have a look at the definition of function wifi_softap_set_config and what it expects as it parameters.

kriegste, 2020-03-19:
In rare cases I get a very similar error when upgrading the firmware using system_upgrade_reboot(). The upgrade works, but instead of a clean reboot, this happens:

```
reboot to use1
state: 5 -> 0 (0)
rm 0
pm close 7
del if0
usl
Fatal exception 28(LoadProhibitedCause):
epc1=0x4000df2f, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000033, depc=0x00000000
```

This is at a slightly different address, but still inside memcmp. Also excvaddr is 0x00000033.

Any ideas?

kriegste, 2020-03-19:
I noticed other people having the same issues, e.g.
https://bbs.espressif.com/viewtopic.php?t=2934
https://bbs.espressif.com/viewtopic.php?t=5784
https://www.esp8266.com/viewtopic.php?p=83491
https://bbs.espressif.com/viewtopic.php?t=24449
It even happens in Arduino:
https://www.bountysource.com/issues/55412753-fail-to-reboot-esp8266
https://www.gitmemory.com/issue/esp8266/Arduino/6063/491462561
https://groups.google.com/forum/#!topic/sonoffusers/ETe2T-ZhF-U

kriegste, 2020-03-22:
Might have found the solution. I have a timer running which checks the connection status every minute.

Something like
if (wifi_station_get_connect_status() != STATION_GOT_IP) wifi_station_connect();

Now for some reason after calling system_upgrade_reboot(), timers are still active for fractions of a second. So they might be executed. However, the SDK crashes when calling wifi_station_connect() during this moment. It might happen with system_restart(), too. But I am not sure.

To properly solve the problem, the SDK needs a check in wifi_station_connect() so that no crash can happen. Should not be too hard or time consuming for Espressif.

A work around is to disarm the corresponding (or all) timers before calling system_upgrade_reboot() or system_restart().

st0ff3r, 2020-05-18:
after reading this: https://www.esp8266.com/viewtopic.php?p=74058 I tried calling the lwip function etharp_request() and got the same error. So maybe some part of lwip init is not called when interface is brought up?

Have to investigate more...

st0ff3r, 2020-05-18:
how does the sdk initialize/stop/start lwip? (esp-open-lwip)

st0ff3r, 2020-05-18:
think I fixed it in the lwip code https://github.com/nabovarme/esp-open-lwip/commit/dccc090844905aa138d0d039346c51ff35ce1ada