espressif/ESP8266_NONOS_SDK #260: Still OTA function some question!

imliubo, 2019-08-28:
Hi:
  I found another probelem,it's weird!
  I can upgrade by OTA(Over the Air) from  user1 to user2,also I can do it from user2 to user1.But when I upgrade success to user2,and I want do it again(user2 to user1) then~
```shell
[OXY] Device OTA upgrade begain
[OXY] HOST: xxxxxxxxx.com
[OXY] upgrade_version: 2.0.0 
[OXY] token: xxxxxxxxx
[OXY] ota server dns found XX.XX.XX.XX
[OXY] tcp client connect cb
Fatal exception 9
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 2592, room 16 
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8 
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4 
tail 0
chksum 0x22
csum 0x22
```
Serial will print error like this and ESP8266 will reset,when ESP8266 reset complete I send the OTA command again it can be work and no error will print from serial.
```SHELL
2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : DIO
SPI Flash Size & Map: 32Mbit(1024KB+1024KB)
jump to run user1 @ 1000

鼘{劕,	峦fc000勩?侼鞉躳愹n|?lll$`b?瀨r?d躱?僶?$`勨{抣?$?l`屻r沴鋖?l窰h鮝eu丼DK version:3.1.0-dev(f18b6d4)
[OXY] system_get_free_heap_size=45000

 user1 

[OXY] load ...
[OXY] gpio_name 4 
[OXY] system_free_size = 40672
mode : sta(b4:e6:2d:4a:96:c1)
add if0
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with CMCC-Master, channel 11
dhcp client start...
[OXY] wifi status is not connected
[OXY] PowerOnFlag: 1 
ip:192.168.1.103,mask:255.255.255.0,gw:192.168.1.1
[OXY] wifi status is Connected
[OXY] current time : Wed Aug 28 09:43:47 2019

client handshake start.
espconn_mbedtls.c 661, type[TLS.ca_x509.cer],length[704]
client handshake ok!
[OXY] MQTT: Connected
[OXY] GET OTA FLAG
pm open,type:0 0
[OXY] Receive topic: amq/topic/b4e62d4a96c1/upgrade, data: {
    "host": "xxxxxxxxxxxxxxxxxxxxxxxx",
    "upgrade_version": "2.0.0",
    "token": "xxxxxxxxx"
} 
[... 407 lines cut ...]
```
As you see it was success and now run at user1.But when do it again from user1 to user2 it was still print error like above log!
```shell
[OXY] Device OTA upgrade begain
[OXY] HOST: xxxxxxxxx.com
[OXY] upgrade_version: 2.0.0 
[OXY] token: xxxxxxxxx
[OXY] ota server dns found XX.XX.XX.XX
[OXY] tcp client connect cb
Fatal exception 9
 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 2592, room 16 
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8 
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4 
tail 0
chksum 0x22
csum 0x22
```
This means that I need to upgrade twice for each upgrade, and I don't know why the first upgrade will fail and the second upgrade will succeed.

xcguang, 2020-01-10:
[image]
it crashed

Dipendra77, 2020-03-04:
How do i perform  OTA ? Can you share an example code of OTA  with th ecurrent version of SDK?

imliubo, 2020-03-04:
> How do i perform OTA ? Can you share an example code of OTA with th ecurrent version of SDK?

I have a very basic code, you can refer to it.

[code Link](https://github.com/imliubo/makingfunxyz-esp8266/tree/master/makingfunxyz-esp8266-NONOS/15.FOTA)

Dipendra77, 2020-03-05:
Hi @imliubo I tried you'd code but it says undefined reference to system_upgrade_start when I compile it. Also I can't find a direct reference for this function system_upgrade_start in the SDK. What am I missing ?

imliubo, 2020-03-05:
Hi @Dipendra77 ,
I'm sorry to hear that,did you check your Makefile under app floder?
system_upgrade_start function is need a static lib(lupgrade),so maybe you need change your Makefile at line 55 - line 80, just add **-lupgrade\\**.Hope it can help you.

But seince the [ESP8266 NonOS SDK support policy changed](https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/README.md),I have n’t used it for a long time.

It is recommended that you also use the ESP8266 RTOS SDK for development.

Have a nice day(or night)!