espressif/ESP8266_NONOS_SDK #265: ESP8266 SSL SEND FAIL

zhaojinyun, 2019-09-30:
1. When using ESP8266 to connect to a specific server over SSL, the connection handshake is successful, but sending any data will be prompted for a send failure.
2. This is only a problem with mbedtls, not SSL.
3.  Esp8266_nonos_sdk-2.2.1 / esp8266_nonos_sdk-3.0.1 both have this problem.
4.The same mbedtls configuration worked fine when tested on a PC.And the connection handshake has been successful, so it should have nothing to do with configuration.

The following is the AT log and WIFI firmware log:

wifi_uart_send:20288(20),AT+CIPSSLSIZE=4096
wifi_uart_send:20300(45),AT+CIPSTART=1,"SSL","posapi.imipay.com",443
wifi_uart_recv1:20609,1,CONNECT
wifi_uart_send:20745(18),AT+CIPSEND=1,168
wifi_uart_send:20960(128)
wifi_uart_send:21005(40),
wifi_uart_recv1:21006,OK
wifi_uart_recv1:21025,> 
wifi_uart_recv1:21047,Recv 168 bytes
wifi_uart_recv1:21066,1,CLOSED
wifi_uart_recv1:21066,SEND FAIL

------------------------------------------------------------------

pm open,type:1 0
DNS found: 120.133.241.145
espconn_secure_get_size 6144
heap_size 33592 20480
lwIPSocket.c 518, 0 succeeded
lwip_connect(0) succeeded
client handshake start.
client handshake ok!
tcp client connect
pespconn 0x3fff25b0
ESPCONN_EVENT_RECV -30592
869 0xffff8880
client's data invalid protocol
Reason:[-0x7780]
disconnect
DNS found: 120.133.241.145
espconn_secure_get_size 6144
heap_size 33592 20480
lwIPSocket.c 518, 0 succeeded
lwip_connect(0) succeeded
client handshake start.
client handshake ok!
tcp client connect
pespconn 0x3fff25b0
ESPCONN_EVENT_RECV -30592
869 0xffff8880
client's data invalid protocol
Reason:[-0x7780]
disconnect

xcguang, 2019-10-25:
Could you try the latest commit?

zhaojinyun, 2019-11-26:
谢谢，按照补丁修改之后可以和这个服务器正常通讯了。但修改之后好像固件占用的内存变大了？

kriegste, 2019-11-26:
This is because ssl v3 was activated. Thus higher memory usage.

zhaojinyun, 2019-11-27:
屏蔽ESP8266_PLATFORM之后，mbedtls_ssl_setup中内存使用量增加了一倍，out_buf,in_buf都各自分配了一个缓存区，之前是复用的。这个有办法优化吗？

zhaojinyun, 2019-11-27:
> 
> 
> This is because ssl v3 was activated. Thus higher memory usage.

I did not turn on SSL v3.The memory usage is higher because mbedtls_ssl_setup has one more buffer allocated than before.

kriegste, 2019-11-27:
No, I meant it was turned on by Espressif by default to support older servers.

zhaojinyun, 2019-11-27:
> 
> 
> No, I meant it was turned on by Espressif by default to support older servers.

Yes, I did not merge the SSL V3 patch into my project.But after disable ESP8266_PLATFORM, the mbedtls_ssl_setup function allocates more memory. Is there any way to optimize this?thank you