espressif/ESP8266_NONOS_SDK #267: AT+CIPSERVER Bug?

GeraldR63, 2019-10-04:
Hello,

have problem with  ESP8266 HTTP server. I'm pretty sure that this is a firmware problem. using ESP8266 NONOS SDK

HTTP Server is configured this way:

AT+CIPSERVER=1,80

Between ESP8266 and Internet is a Router. Router is configured to listen at Port 5001 and move all requests to Port 80 at the ESP.

http://<my domain>:5001
In this case a local connect to ESP with FireFox show the expected Web Page but via router only the HTML source is shown. FireFox message: 

"The character encoding of the plain text document was not declared. The document will render with garbled text in some browser configurations if the document contains characters from outside the US-ASCII range. The character encoding of the file needs to be declared in the transfer protocol or file needs to use a byte order mark as an encoding signature."

If router is configured to use Port 80 instead of Port 5001 than it works. But router is not the problem. 

AT+CIPSERVER=1,5001

do  also not work for local/direct connection. Same problem. HTML code is shown.

MS IE works. Nevertheless I use Apache 2.0 and IIS here as reference. Everything works fine if doing same with Apache or IIS.  That's why it is most likely that there is a bug in the firmware.

regards
Gerald

In my opinion this is just a bug in the firmware. Your software developers should be able to reproduce ist.

MaJerle, 2019-10-04:
I confirm my usage of CIPSERVER works as expected. No bugs as explained here and I'm not sure how this can be related to AT commands.

GeraldR63, 2019-10-04:
[image]
[image]
[image]

GeraldR63, 2019-10-04:
Images above show what happened. This happens also if connecting via Internet to my domain or just local, direct. I have here Apache 2.0 at Port 80 and 5000, works fine and IIS at Port 81 and 4711. Works also fine. The ESP bought last week.  

That's why it should be a failure of the ESP/ BIOS. Don't know what's wrong if using a different port. 
If using MS IE instead FireFox it works but it's not a FireFiox bug.  Web Sniffer also do not work in this case.

Have a firmware update first?

Had the same Issue with ESP01S. Now it's a NodeMCU LUA Amica V2 from AZ-Delivery. Had an update to the latest NONOS SDK. 

Images above show clearly that this do not work. 
The other W-LAN HTTP server base upon Raspberry Pi 3 and 2 with EDIMAX Wi-Fi Nano Adapter. Doing same but works.

regards
Gerald

regards
Gerald