espressif/ESP8266_NONOS_SDK #270: WPS issue on SDK 3.0.1

esp-dev, 2019-10-22:
Hi,

I have issue with WPS on new SDK 3.0.1. WPS not working, and finally code is 23:
user_interface.h:    REASON_802_1X_AUTH_FAILED       = 23,  /* 11i */
log:
mode : sta(5c:cf:7f:XX:XX:XX)
add if0
wifi_wps_enable
wps scan
build public key start
build public key finish
wps discover [SSID]
scandone
WPS: neg start
mode: 0 -> 3
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt
….wiat
state: 5 -> 2 (17c0)
rm 0
Disconn from SSID;23

On SDK 3.0.0 WPS work correctly.
Any idea? What different? How to make it working?

hubciorz, 2019-10-25:
There seems to be a problem with this commit: 6580ca35e9f8f85d11100c1ad3d2ed27d4b26381
WPS stopped working on some APs

kriegste, 2019-11-08:
Which routers are affected? Is Espressif working on this?

Junhao-Espressif, 2019-11-18:
@hubciorz Could you try to connect to the AP directly with ssid and password to see if you can connect to it? 
Thanks!

esp-dev, 2019-11-18:
AP chipset is Broadcom, connect manually by SSID/password works.

Junhao-Espressif, 2019-11-19:
@esp-dev 

Could you provide more details about the AP? 
We will buy an AP for debugging.

Thank you!

FayeY, 2019-11-25:
@hubciorz, could you list "some AP" that will fail with WPS? Maybe we can find the same APs to test and debug with the issue. Thanks.

sjanek, 2019-11-25:
@Junhao-Espressif, @FayeY

hubciorz, 2019-11-25:
@FayeY as stated above - there is a problem with Internet-Box 2 from Swisscom.

Junhao-Espressif, 2019-11-26:
Hi @hubciorz 
Could you try with this lib on master version?

[libnet80211.zip](https://github.com/espressif/ESP8266_NONOS_SDK/files/3889738/libnet80211.zip)

hubciorz, 2019-11-27:
@esp-dev could you try this one?

esp-dev, 2019-11-28:
cloned commit: 4bf0aefee37ac1f272b793c2403c53a0c122b570 (branch v3.0.1, should I clone from master?)
replace libnet80211.a
**WPS working now.**

closed 2020-09-16 by esp-dev